linrunner/TLP #472: 1.3 configuration files location

helght, 2020-02-15:
Wouldn't it make more sense if  `/etc/tlp.d/*.conf` configs had precedence over `/etc/tlp.conf` base config? 

Consider the following: user installs tlp, gets distribution defaults in `/etc/tlp.conf` and then puts their platform-specific tweaks in `tlp.d`. A more natual way of doing conf things.

`/etc/tlp.d/0-disable-this.conf`
`/etc/tlp.d/9-enable-that.conf`
`/etc/tlp.d/99-my-custom-usb-blacklist.conf`
and so on. 

It makes it easier to track what was tweaked relative to the base config and to migrate configs from one system to another. Other than this - great feature

linrunner, 2020-02-17:
The original tlp.conf does *not* contain the defaults, rather all settings are commented out there, so that **/etc/tlp.d/*.conf** has priority. Your use case just works without any problems. Magic, isn't it?

helght, 2020-02-18:
Uhm, where are the defaults stored? Hardcoded in the source?

helght, 2020-02-18:
Somehow I thought that defaults are loaded from tlp.conf and the user has to tweak it.

linrunner, 2020-02-18:
The defaults are in the official documentation and also (as descriptions) inside **tlp.conf**.

closed 2020-02-18 by linrunner