linrunner/TLP #474: AHCI Port is shown as Bad in 'Powertop'

liny01-nbsa, 2020-02-22:
This is what shown in my powertop:

```
PowerTOP v2.11    Overview   Idle stats   Frequency stats   Device stats   Tunables   WakeUp                           
>> echo 'auto' > '/sys/bus/pci/devices/0000:00:17.0/ata2/power/control';

>> Good          Runtime PM for port ata2 of PCI device: Intel Corporation Q170/Q150/B150/H170/H110/Z170/CM236 Chipset 
   Bad           Runtime PM for port ata4 of PCI device: Intel Corporation Q170/Q150/B150/H170/H110/Z170/CM236 Chipset
   Bad           Runtime PM for port ata5 of PCI device: Intel Corporation Q170/Q150/B150/H170/H110/Z170/CM236 Chipset
   Bad           Runtime PM for port ata1 of PCI device: Intel Corporation Q170/Q150/B150/H170/H110/Z170/CM236 Chipset
   Bad           Runtime PM for port ata3 of PCI device: Intel Corporation Q170/Q150/B150/H170/H110/Z170/CM236 Chipset

```

And here is related tlp-stat output:

```
--- TLP 1.3.1 --------------------------------------------

+++ Configured Settings:

...
/etc/tlp.d/00-tlp.conf L0145: DISK_DEVICES="nvme0n1"
/etc/tlp.d/00-tlp.conf L0194: SATA_LINKPWR_ON_AC="min_power"
/etc/tlp.d/00-tlp.conf L0195: SATA_LINKPWR_ON_BAT="min_power"
/etc/tlp.d/00-tlp.conf L0212: AHCI_RUNTIME_PM_TIMEOUT="15"
/etc/tlp.d/00-tlp.conf L0206: AHCI_RUNTIME_PM_ON_AC="auto"
/etc/tlp.d/00-tlp.conf L0207: AHCI_RUNTIME_PM_ON_BAT="auto"

...

+++ File System
/proc/sys/vm/laptop_mode               =     0
/proc/sys/vm/dirty_writeback_centisecs =  6000
/proc/sys/vm/dirty_expire_centisecs    =  6000
/proc/sys/vm/dirty_ratio               =    50
/proc/sys/vm/dirty_background_ratio    =    20

...

+++ AHCI Link Power Management (ALPM)
/sys/class/scsi_host/host0/link_power_management_policy  = min_power
/sys/class/scsi_host/host1/link_power_management_policy  = min_power
/sys/class/scsi_host/host2/link_power_management_policy  = min_power
/sys/class/scsi_host/host3/link_power_management_policy  = min_power
/sys/class/scsi_host/host4/link_power_management_policy  = min_power

...

+++ AHCI Host Controller Runtime Power Management
/sys/bus/pci/devices/0000:00:17.0/ata1/power/control = on
/sys/bus/pci/devices/0000:00:17.0/ata2/power/control = on
/sys/bus/pci/devices/0000:00:17.0/ata3/power/control = on
/sys/bus/pci/devices/0000:00:17.0/ata4/power/control = on
/sys/bus/pci/devices/0000:00:17.0/ata5/power/control = on

...

```

Is there anything I can do to fix those ataX port configuration in TLP ?

Thanks!

liny01-nbsa, 2020-02-22:
Hi, @linrunner, I have attached more information, thanks!

liny01-nbsa, 2020-02-22:
It seems my tlp configuration wasn't rouching ataX devices at all.

linrunner, 2020-02-23:
Your NVMe disk doesn't have `autosuspend_delay_ms`, see output:

> Runtime PM: control = auto, **autosuspend_delay_ms = (not available)**

In this case TLP doesn't touch the AHCI hosts to prevent possible system lockups causing data loss. Before you ask: there is no way to override this in TLP.

https://github.com/linrunner/TLP/blob/1.3.1/func.d/15-tlp-func-disk#L351

liny01-nbsa, 2020-02-23:
@linrunner, thanks for your comment. I don't completely understand, my SSD is a nvme, those ataX port are on a different controller, and in my case I have no devices connected to it. Do you think it might make sense only when block device connected to ach ports has the issue you described, it would then make sense to apply the skip logic?

Thanks!

linrunner, 2020-02-23:
You're right, of course. After i wrote NVMe, i got the same thought.

For the assignment host – device i could not find an implementation at that time. I'll reconsider this for 1.4

linrunner, 2020-02-23:
ps. Did you actually measure the difference when you write "auto" to the ahci host sysfiles by hand?

Perform the measurement when *all* applications are closed and wait for > 5 minutes, otherwise you will not get a clean result.

liny01-nbsa, 2020-02-23:
Hi, @linrunner, please see the results below:

 - [ataX ports on](https://gist.github.com/liny01-nbsa/4e910a71bb51fcb447f0556651b3e4c9)
 - [ataX ports auto](https://gist.github.com/liny01-nbsa/a49f7ecce857a1569a23c66cf6ae3180)

For then it's on, it's quite difficult for me to capture the full screen, when I selected the full screen, the usage would jump to 15.6W+ or even 16.3W+; So in the screen I captured, is the value I see that waited for 5 mins, with other other applications open, only the following apps:

 - DWM, my window manager,
 - picom, the compositor
 - gnome-termial, and powertop

Thanks!

linrunner, 2020-02-24:
I didn't ask for powertop or "fullscreen", just for the difference in W.

I think your biggest problem/consumer is the 2nd graphics chip, you should look for a way to shut down the amdgpu (but please do not ask me).

liny01-nbsa, 2020-02-24:
My bad. Based on my observation, I think if the difference is visible, it would be less than 1W. more like 0.3W ~ 0.5W, However, it seems, the system would work at the low pc state longer.

And yes, I'm aware of my 2nd graphics card issue. As I still use it from time to time by DRI_PRIME, so I won't be able to completed turn it off. I think I might need to do more research on that.

Thanks, @linrunner .

liny01-nbsa, 2020-02-24:
And I did some quick google search, here are the results, hope it could be of use to you:

    ls -ld /sys/block/sd*/device

,

    cat /proc/scsi/scsi

,

    hdparm -i /dev/sdb

, or

    udevadm info -n /dev/sda -a

linrunner, 2020-05-05:
I have now solved your real problem, see commit.

Please test.

liny01-nbsa, 2020-05-05:
Thank you, @linrunner, I will do some testing this week.

liny01-nbsa, 2020-05-05:
I have tested commit 322d9f7 from master. It works like a charm. Thanks, @linrunner.

closed 2020-05-05 by liny01-nbsa

linrunner, 2020-05-06:
I added some more "vitamines", see commits. Could you please test once more with the latest master branch? Output of `tlp-stat -s -d` would be welcome.

liny01-nbsa, 2020-05-06:
Tested 52d0c7c. Works great too. And the output of [`tlp-stat -s -d`](https://pastebin.lsyoyom.xyz/?56a828714fe1863b#5UA8GZh4rFbmw9a3pviKw7Kbd5t7MaAYFKnibRCnaeY4)