linrunner/TLP #475: flock -w option not supported on busybox

nero, 2020-02-23:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)

**Describe the bug**

tlp-func-base uses the `-w` option of flock, which is not supported by busybox.

**To Reproduce**

Start tlp on Alpine Linux (without GNU coreutils as replacement for busybox). This spews the help text for the flock applet onto stderr, continuing without having acquired the lock.

Solution ideas:
- Declare an dependency on GNU coreutils
- Wrap around that line of code, checking whether -w is supported
- Downstream patch to remove that option completely

dywisor, 2020-02-23:
Note: proper busybox support requires careful inspection of the entire codebase as there will be more cases like [logger not accepting ``--id``](https://github.com/linrunner/TLP/blob/b2e9927b78cc635272955d2955182399bada3c45/tlp-func-base.in#L113).

linrunner, 2020-02-23:
I don't consider this an upstream problem. Please solve it downstream in the Alpine package.

I'll document the dependency on GNU coreutils. in developer docs. And to be clear: busybox is not supported as a replacement.

closed 2020-02-23 by nero

linrunner, 2020-04-29:
Btw: the dependency is _not GNU coreutils_ but **util-linux**: https://git.kernel.org/pub/scm/utils/util-linux/util-linux.git/tree/sys-utils/flock.c

Anyway i added a 'Busybox not supported' disclaimer: https://linrunner.de/tlp/developers/dependencies.html