linrunner/TLP #477: TLP configuration does not apply on startup

strahius, 2020-02-29:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)

**Describe the bug**

I am using Manjaro, on Dell XPS 15 7590 laptop, running on battery. I want the default TLP configuration to be applied when laptop reboots, but I have to explicitly start the service every time. When laptop stars, I see:
```
[stathis@laptop ~]$ sudo tlp-stat -s
[sudo] password for stathis: 
--- TLP 1.3.1 --------------------------------------------

+++ System Info
System         = Dell Inc.  XPS 15 7590
BIOS           = 1.3.3
Release        = "Manjaro Linux"
Kernel         = 5.5.6-1-MANJARO #1 SMP Mon Feb 24 09:24:51 UTC 2020 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-5.5-x86_64 root=UUID=b4f494fc-57b1-427f-85ad-a8303f21964b rw apparmor=1 mem_sleep_default=deep security=apparmor resume=UUID=1cc03051-990d-4117-89b9-c70ac9868be5 udev.log_priority=3
Init system    = systemd 
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = not installed
Last run       = 00:33:15,     61 sec(s) ago
Mode           = battery
Power source   = battery

[stathis@laptop ~]$ systemctl status tlp
● tlp.service - TLP system startup/shutdown
   Loaded: loaded (/usr/lib/systemd/system/tlp.service; enabled; vendor preset: disabled)
   Active: active (exited) since Sat 2020-02-29 00:33:15 GMT; 1min 7s ago
     Docs: https://linrunner.de/tlp
  Process: 995 ExecStart=/usr/bin/tlp init start (code=exited, status=0/SUCCESS)
 Main PID: 995 (code=exited, status=0/SUCCESS)

Feb 29 00:33:15 laptop systemd[1]: Starting TLP system startup/shutdown...
Feb 29 00:33:15 laptop tlp[995]: Applying power save settings...done.
Feb 29 00:33:15 laptop tlp[995]: Setting battery charge thresholds...done.
Feb 29 00:33:15 laptop systemd[1]: Started TLP system startup/shutdown.
```

`powertop` reports about 2 hours remaining time, and stays around there for 10 or minutes, after multiple tests.

Running `tlp start` or `systemctl restart tlp` brings the remaining time up to 10 hours.

**Expected behavior**

TLP configuration should be applied at startup, without explicit restart needed.

**To Reproduce**
1. Use default TLP configuration
2. Enable tlp service
3. Reboot
4. Notice that tlp service is actually on and mode is battery, but battery life hasn't actually increased.

strahius, 2020-02-29:
@linrunner could you please let me know what is incomplete in this issue?

linrunner, 2020-02-29:
You have been complaining that the settings don't apply. As written in the bug template I *always* need the complete output of `tlp-stat`. In your case (1) after the system start on battery and (2) after the manual `tlp start`.

strahius, 2020-02-29:
Sorry @linrunner, my bad, I thought those steps were an example of the template. Updated now.

strahius, 2020-02-29:
Just to add to this, I think it also happens when laptop suspends. This is the behaviour after resuming from suspend:

[image]

linrunner, 2020-03-01:
Comparing your outputs shows that settings are indeed applied on boot – with one exception concerning the autosuspend mode of your 2nd hybrid graphics chip.

On boot:
> /sys/bus/pci/devices/0000:01:00.0/power/**control = on** (0x030200, 3D controller, no driver)

After re-applying settings:
> /sys/bus/pci/devices/0000:01:00.0/power/control = auto (0x030200, 3D controller, no driver)

I'd guess the kernel blocks TLP's write request on boot or something else interferes – unloading the nvidia driver maybe? In contrast, it works at a later point in time using the same code. TLP is already running pretty much last in system start. Sorry, there's nothing I can do for you here.

In your case i would not rely solely on autosuspend but use one of the dedicated tools to reliably switch off nvidia.

linrunner, 2020-03-01:
ps. I know that fiddling with the tlp.service is very popular at the Manjaro forums although I have already explained to the supporters there that it is wrong.

Don't restart the service to apply new settings, just use `tlp start` as i told you above and wrote in the docs. Thanks.

strahius, 2020-03-01:
@linrunner thanks for looking at this and your suggestion. I am using `optimus-manager` for auto-switching to the Intel GPU on startup. Is that what you mean?

linrunner, 2020-03-01:
Maybe. I can't advise on hybrid graphics. Sorry.

strahius, 2020-03-01:
@linrunner thanks a lot, your comment actually helped me fix this! I applied [this](https://wiki.archlinux.org/index.php/Dell_XPS_15_9570#Use_Optimus_Manager_to_switch) `optimus-config` config, and now looks like it's working fine :smile:

closed 2020-03-01 by strahius

linrunner, 2020-03-02:
Thank you for the feedback.