linrunner/TLP #478: Add a TLP_CONFIG_NAME parameter to easily know which of our configuration file active

Eurbaix, 2020-03-02:
If we use different configuration files, to adjust the power/performance balance to our current use case, the only way currently to recognize which configuration file is "active" (i.e. has been copied as /etc/tlp.conf) is to read all parameters and try to remember which configuration file they come from.

A simple implementation would be to add a TLP_CONFIG_NAME or TLP_PROFILE_NAME parameter who would only be a place holder for a name, ideally at the beginning of the config file (just before TLP_ENABLE). It doesn't need any change in the code (since TLP doesn't care - apparently - if it reads an unrecognized parameter) but making a place holder like TLP_CONFIG_NAME "official" would allow other apps, like "TLP Switcher", to rely on this parameter in the future to read and show which configuration/profile is active.

P.S. /etc/tlp.d/*.conf file names do not really work for this use case because TLP apply all of them all the time. To change from one "profile" to another, /etc/tlp.conf is the best candidate, as it allows to change all the parameters at once, except for that problem that there is no way to know where the active /etc/tlp.conf come from.

linrunner, 2020-03-06:
I propose you call the variable something like `SWITCHER_*` and i'll keep TLP off that part of the namespace. No need for something "official".

I saw the installation instructions für the switcher contain something like `tlp-stat -c`. This won#t work anymore because the output changed. Since you'll have to filter the output anyway, i strongly recommend to remove the lines starting with `defaults.conf` completely and use only the rmaining lines containing the actual user changes.

Also please leave the pristine **/etc/tlp.conf** with the explanations of the variables untouched and write your configuration to **/etc/tlp.d/*.conf**.

Eurbaix, 2020-03-06:
Although I currently use TLP_switcher and modified one command to make it compatible with TLP1.3, I am not maintaining it (my programming days, and knowledge, are really far behind and I'd probably brake it instead of improving it).

The reason I thought it would be better if a profile name was implemented by TLP is to maintain compatibility between GUIs if someone develop a new profile switcher or if TLPUI implements the ability to switch between configuration files. If new variables are implemented by each GUI, it will soon become a mess of incompatibility.

As for using /etc/tlp.d/*.conf with only the modified variables, it's a good idea to make the profile files simpler, but there is still the problem that the active file will always have the same filename.

linrunner, 2020-03-12:
I saw your fork and had the impression that you might want to continue the development. But under the given circumstances i will not do anything further.

closed 2020-03-12 by linrunner