linrunner/TLP #479: Lenovo USB-C Gen2 Docking Station not Charging Thinkpad when TLP is Enabled

lvlanson, 2020-03-04:
Hey there,

as long as TLP is enabled the Docking Station does not charge my Thinkpad anymore. I cannot figure out which setting has to be altered to make it work. I checked quite a few settings. The docking station is connected via the thunderbolt 3 connector.

Also as soon as I plugged in the docking station, I need to "unplug" the battery via bios, otherwise I also cannot charge via my AC.

When TLP is disabled the docking station is charging my thinkpad and is working as expected.

What can I do?

lvlanson, 2020-03-04:
I want to add, disabling tlp alone does not help. Only when I rebooted after disabling TLP the charging works again.

linrunner, 2020-03-06:
Never heard of such a symptom before. 

Make shure all your Lenovo firmware is up to date. Specifically i remember critical thunderbolt FW updates by Lenovo recently.

Apart form that I'm afraid you're on your own to identify the cause, refer to the [troubleshooting guide](https://linrunner.de/en/tlp/docs/tlp-troubleshooting.html).

*Btw: no support whatsoever without all the outputs required by the bug template.**

lvlanson, 2020-03-09:
Blacklisting "mei_me" under PCIe fixed the problem.

Thanks for the heads up.

linrunner, 2020-03-12:
Thanks for your feedback. 

However for recent TLP versions mei_me is in RUNTIME_PM_DRIVER_BLACKLIST by default. Without the outputs I'm left in the dark about your system.

lvlanson, 2020-03-12:
Could you point me to the bug template and tell me which information you need?

After some testing it still works pretty unreliably.

linrunner, 2020-03-13:
https://github.com/linrunner/TLP/blob/master/.github/ISSUE_TEMPLATE/bug_report.md
https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md

It's just the full output of tlp-stat via https://gist.github.com/ please, AC suffices in your case.

You didn't comment about my suggestion about firmware updates and I don't very much like to ask for information repeatedly.

lvlanson, 2020-03-13:
I am really sorry not being clear so far. I will post the bugs asap.

Firmware for everything is up to date. The Thinkpad has **all** the latest Firmware updates and so has the docking station.

My apologies

lvlanson, 2020-03-14:
[x] I've read and accepted the Bug Reporting Howto

**Description**

Devices: 
- Lenovo Thinpad X1 Carbon 6th Gen
- Lenovo USB-C Gen2 Docking Station

When attaching the USB-C Docking Station to the Thinkpad, the thinkpad is **not charging**. When TLP is **disabled** the charging works as expected.

After the charging not working via the docking station, also the charging **via the AC does not work anymore**

The Docking Station is connected via a Thunderbolt3 cable at the regular charging connector of the Thinkpad. All firmware has been updated to the very latest versions prior to this.

All peripherals work so far, as far as I had been able to test them yet.

**Excpected Behaviour**

The Laptop should be charging, if the thunderbolt 3 cable is attached.

**To Reproduce**

The Problem does Occur when on AC.

1. Having TLP enabled
2. Attach the tb3 cable
2.a if its charging, try replugging it, soon it will fail

tlp-stat:
https://gist.github.com/lvlanson/58303db402276e8582da2ddc26f2baa6

If anything is missing or if you need anything additional, please let me know

linrunner, 2020-03-14:
(1) Your configuration differs from what you wrote above: RUNTIME_PM_DRIVER_BLACKLIST does *not* contain mei_me, which is probably irrelevant because on AC autosuspend is disabled (`on`) anyway. Or did you mean the opposite e.g. you removed mei_me from the list (seems more plausible)?

(2) You configured thresholds so the battery won't charge at the current SOC of 88% which is between 20 and 90 = works as designed by Lenovo.

    /sys/class/power_supply/BAT0/status                         = Unknown (threshold may prevent charging)
    /sys/class/power_supply/BAT0/charge_start_threshold         =     20 [%]
    /sys/class/power_supply/BAT0/charge_stop_threshold          =     90 [%]
    Charge                                                      =   87.9 [%]

Please comment the thresholds in the configuration and remove them with

    tlp fullcharge

Test again.

lvlanson, 2020-03-14:
I should have mentioned, that there is an LED indicating if the connector is receiving any power or not. If power is full, the LED is white, if its charging the LED is yellow. The LED when the error occurs is actually off.

I will drain the battery, comment out the threshholds and do tlp fullcharge and repost the tlp-stat command output.

linrunner, 2020-03-14:
I did *not* say to drain the battery, just remove the thresholds as described.

Afterwards undock, dock again and check if the symptom occurs. Only then show the output.

linrunner, 2020-03-14:
ps. I read about malfunctions where the LED didn't indicate the charging state correctly.

lvlanson, 2020-03-14:
Weirdly, the LED is now lit, but it is not charging, though the threshholds are off.

https://gist.github.com/lvlanson/1f57cd6664843d8c612b728ff8141fb3#file-gistfile1-txt

lvlanson, 2020-03-14:
After having rebooted a second time, the status LED is also not lit anymore.

linrunner, 2020-03-14:
Look at your output:

    /sys/class/power_supply/BAT0/charge_start_threshold         =     20 [%]
    /sys/class/power_supply/BAT0/charge_stop_threshold          =     90 [%]

It reveals the thresholds still effective. Either you missed the tlp fullcharge command or it failed because the embedded controller (EC) ignore the writing. Please show:

    tlp fullcharge
    fwupdmgr get-devices

lvlanson, 2020-03-14:
> sudo tlp fullcharge
Setting temporary charge thresholds for BAT0:
  stop  = 100
  start =  96
Charging starts now, keep AC connected.
> fwupdmgr get-devices
X1 Carbon Thunderbolt Controller
  DeviceId:             5f6ba1816efb8d8edc63b4e1aa7c166734c8ed17
  Guid:                 5ff9544e-fce2-5fd9-9cd2-671c95ac91dc
  Summary:              Unmatched performance for high-speed I/O
  Plugin:               thunderbolt
  Flags:                internal|updatable|require-ac|supported|registered
  Vendor:               Lenovo
  VendorId:             TBT:0x0109
  Version:              43.00
  VersionFormat:        pair
  Icon:                 computer
  Created:              2020-03-14

20KGS03800 System Firmware
  DeviceId:             1c53551e7da69d896138fac1ae131c83ad46d923
  Guid:                 a4b51dca-8f97-4310-8821-3330f83c9135
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.46
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 computer
  Created:              2020-03-14

UEFI Device Firmware
  DeviceId:             29749712e93b58730ecd1b8537a5503ff5b7fb27
  Guid:                 9c9d9769-32fa-4841-b550-ea998e754e99
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              184.70.3626
  VersionLowest:        0.0.1
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2020-03-14

UEFI Device Firmware
  DeviceId:             9698faabddf0d7b18925cfbbda95f8b0d0dacc53
  Guid:                 3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Version:              0.1.20
  VersionLowest:        0.1.20
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2020-03-14

UEFI Device Firmware
  DeviceId:             9e329270a7a68d289c82fe77d32d02208ddf0890
  Guid:                 cea87551-1701-43fb-afbc-6e8ce9728345
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.73.4
  VersionLowest:        0.73.4
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2020-03-14

WDC PC SN720 SDAQNTW-512G-1001
  DeviceId:             04e17fcf7d3de91da49a163ffe4907855c3648be
  Guid:                 ff2112dc-038c-596d-90ca-d43c5077c6ec
  Guid:                 63af1441-fb24-5368-b450-886fbc98610b
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
[... 135 lines cut ...]
  Plugin:               uefi
  Flags:                internal|updatable|require-ac|registered|needs-reboot
  Version:              0.73.4
  VersionLowest:        0.73.4
  VersionFormat:        triplet
  Icon:                 audio-card
  Created:              2020-03-14

WDC PC SN720 SDAQNTW-512G-1001
  DeviceId:             04e17fcf7d3de91da49a163ffe4907855c3648be
  Guid:                 ff2112dc-038c-596d-90ca-d43c5077c6ec
  Guid:                 63af1441-fb24-5368-b450-886fbc98610b
  Summary:              NVM Express Solid State Drive
  Plugin:               nvme
  Flags:                internal|updatable|require-ac|supported|registered|needs-reboot
  Vendor:               Sandisk Corp
  VendorId:             NVME:0x15B7
  Version:              10190101
  VersionFormat:        plain
  Icon:                 drive-harddisk
  Created:              2020-03-14

===========================================================

It is now charging

linrunner, 2020-03-14:
And the output of 

    tlp-stat -s -c 

please.

ps. Thunderbolt firmware 43.00 and EC 1.20 are newest indeed. I use a X1C6 with CS18 dock  (connected to the TB port) myself running kernel 5.5.8 and i observe *zero* charging issues.

lvlanson, 2020-03-14:
This is the output:

--- TLP 1.3.1 --------------------------------------------

+++ Configured Settings:
/etc/tlp.conf L0015: TLP_ENABLE="1"
/etc/tlp.conf L0025: TLP_PERSISTENT_DEFAULT="0"
/etc/tlp.conf L0030: DISK_IDLE_SECS_ON_AC="0"
/etc/tlp.conf L0031: DISK_IDLE_SECS_ON_BAT="2"
/etc/tlp.conf L0035: MAX_LOST_WORK_SECS_ON_AC="15"
/etc/tlp.conf L0036: MAX_LOST_WORK_SECS_ON_BAT="60"
defaults.conf L0010: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
defaults.conf L0011: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"
/etc/tlp.conf L0097: SCHED_POWERSAVE_ON_AC="0"
/etc/tlp.conf L0098: SCHED_POWERSAVE_ON_BAT="1"
/etc/tlp.conf L0103: NMI_WATCHDOG="0"
/etc/tlp.conf L0124: DISK_DEVICES="nvme0n1 sda"
defaults.conf L0016: DISK_APM_LEVEL_ON_AC="254 254"
defaults.conf L0017: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0018: DISK_IOSCHED="keep keep"
/etc/tlp.conf L0166: SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
/etc/tlp.conf L0167: SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
/etc/tlp.conf L0184: AHCI_RUNTIME_PM_TIMEOUT="15"
defaults.conf L0022: PCIE_ASPM_ON_AC="default"
defaults.conf L0023: PCIE_ASPM_ON_BAT="default"
defaults.conf L0024: RADEON_POWER_PROFILE_ON_AC="default"
defaults.conf L0025: RADEON_POWER_PROFILE_ON_BAT="default"
defaults.conf L0026: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
defaults.conf L0027: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
/etc/tlp.conf L0223: WIFI_PWR_ON_AC="off"
/etc/tlp.conf L0224: WIFI_PWR_ON_BAT="on"
/etc/tlp.conf L0228: WOL_DISABLE="Y"
/etc/tlp.conf L0233: SOUND_POWER_SAVE_ON_AC="0"
/etc/tlp.conf L0234: SOUND_POWER_SAVE_ON_BAT="1"
/etc/tlp.conf L0239: SOUND_POWER_SAVE_CONTROLLER="Y"
/etc/tlp.conf L0246: BAY_POWEROFF_ON_AC="0"
/etc/tlp.conf L0247: BAY_POWEROFF_ON_BAT="0"
/etc/tlp.conf L0250: BAY_DEVICE="sr0"
/etc/tlp.conf L0254: RUNTIME_PM_ON_AC="on"
/etc/tlp.conf L0255: RUNTIME_PM_ON_BAT="auto"
/etc/tlp.conf L0268: RUNTIME_PM_DRIVER_BLACKLIST="amdgpu nouveau nvidia pcieport radeon"
/etc/tlp.conf L0272: USB_AUTOSUSPEND="1"
/etc/tlp.conf L0283: USB_BLACKLIST_BTUSB="0"
/etc/tlp.conf L0288: USB_BLACKLIST_PHONE="0"
/etc/tlp.conf L0293: USB_BLACKLIST_PRINTER="1"
/etc/tlp.conf L0298: USB_BLACKLIST_WWAN="0"
defaults.conf L0045: USB_AUTOSUSPEND_DISABLE_ON_SHUTDOWN="0"
/etc/tlp.conf L0315: RESTORE_DEVICE_STATE_ON_STARTUP="0"
defaults.conf L0047: RESTORE_THRESHOLDS_ON_BAT="0"
/etc/tlp.conf L0368: NATACPI_ENABLE="1"
/etc/tlp.conf L0369: TPACPI_ENABLE="1"
/etc/tlp.conf L0370: TPSMAPI_ENABLE="1"
/etc/tlp.conf L0020: TLP_DEFAULT_MODE="AC"
/etc/tlp.conf L0072: CPU_HWP_ON_AC="balance_performance"
/etc/tlp.conf L0073: CPU_HWP_ON_BAT="balance_power"
/etc/tlp.conf L0080: CPU_MIN_PERF_ON_AC="0"
/etc/tlp.conf L0081: CPU_MAX_PERF_ON_AC="100"
/etc/tlp.conf L0082: CPU_MIN_PERF_ON_BAT="0"
/etc/tlp.conf L0083: CPU_MAX_PERF_ON_BAT="50"
/etc/tlp.conf L0091: CPU_BOOST_ON_AC="1"
/etc/tlp.conf L0092: CPU_BOOST_ON_BAT="0"
/etc/tlp.conf L0117: ENERGY_PERF_POLICY_ON_AC="performance"
/etc/tlp.conf L0118: ENERGY_PERF_POLICY_ON_BAT="power"
/etc/tlp.conf L0198: INTEL_GPU_MIN_FREQ_ON_BAT="0"
/etc/tlp.conf L0200: INTEL_GPU_MAX_FREQ_ON_BAT="800"
/etc/tlp.conf L0202: INTEL_GPU_BOOST_FREQ_ON_BAT="1000"

+++ System Info
System         = LENOVO ThinkPad X1 Carbon 6th 20KGS03800
BIOS           = N23ET71W (1.46 )
Release        = Pop!_OS 19.10
Kernel         = 5.5.9-050509-generic #202003120738 SMP Thu Mar 12 07:41:55 UTC 2020 x86_64
/proc/cmdline  = initrd=\EFI\Pop_OS-5ece54db-02d2-4fd3-8ad3-6e13212dcd96\initrd.img root=UUID=5ece54db-02d2-4fd3-8ad3-6e13212dcd96 ro quiet loglevel=0 systemd.show_status=false splash psmouse.synaptics_intertouch=0
Init system    = systemd v242 (242)
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 06:05:30 PM,      4 sec(s) ago
Mode           = AC
Power source   = AC

============================================================

Also I notice, normally the LED is orange when charging, now it is constantly white.

linrunner, 2020-03-14:
Sorry, I forgot

    tlp-stat -b

lvlanson, 2020-03-14:
--- TLP 1.3.1 --------------------------------------------

+++ Battery Features: Charge Thresholds and Recalibrate
natacpi    = active (data, thresholds)
tpacpi-bat = active (recalibrate)
tp-smapi   = inactive (ThinkPad not supported)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = LGC
/sys/class/power_supply/BAT0/model_name                     = 01AV494
/sys/class/power_supply/BAT0/cycle_count                    =    126
/sys/class/power_supply/BAT0/energy_full_design             =  57000 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  50580 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  48760 [mWh]
/sys/class/power_supply/BAT0/power_now                      =  11118 [mW]
/sys/class/power_supply/BAT0/status                         = Charging

/sys/class/power_supply/BAT0/charge_start_threshold         =     96 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold          =    100 [%]
tpacpi-bat.BAT0.forceDischarge                              =      0

Charge                                                      =   96.4 [%]
Capacity                                                    =   88.7 [%]

linrunner, 2020-03-15:
Now observe if the not-charging problem occurs again without charge thresholds.

I must say i can't make sense of your initial threshold values yet. start=20 means you may start travelling with 20% remaining capacity which is ~2h runtime max. on the X1C6. Did you look at the [FAQ "How to choose good battery charge thresholds"](https://linrunner.de/en/tlp/docs/tlp-faq.html#battery)?

lvlanson, 2020-03-15:
Actually I haven't yet looked at the charge threshholds.

So far it seems to work fine without the threshholds.

linrunner, 2020-03-29:
I assume that this was a problem with the handling of the charge thresholds. Feel free to reopen if you find other clues.

closed 2020-03-29 by linrunner

linrunner, 2020-04-15:
[Here](https://fwupd.org/lvfs/devices/com.lenovo.Gen2.ccgx.fw2.firmware) is an interesting firmware for the USB-C Gen2 Dock:

> Fix smart charging power mechanism issue on all P series Laptops(P52, P53, P72, and P73)

It's still in "testing" so won't show up in a plain `fwupdmgr update`yet.

lvlanson, 2020-04-15:
Thanks for the heads up! I will check it when available.

Normally I had to update the firmware of the Dock via Windows, because fwupdmgr couldn't really update it.