linrunner/TLP #480: Custom config files are not applied

validatedev, 2020-03-11:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)

**Describe the bug**
Cannot change the default settings with any of the `/etc/tlp.conf ` or `/etc/tlp.d/*.conf`. Interestingly, `tlp-stat` shows the tlp version is 1.2.2.
**Expected behavior**
User config should applied.

**To Reproduce**
Change `/etc/tlp.conf `
Reboot or run `sudo tlp start`
Run `sudo tlp-stat` for debugging
 
1. Does the problem occur on battery or AC or both? Both
2. Actions to reproduce the behaviour
3. Shell commands entered and their output

My user config:
```TLP_PS_IGNORE=BAT
PCIE_ASPM_ON_AC=default
PCIE_ASPM_ON_BAT=powersupersave
SOUND_POWER_SAVE_ON_AC=1
SOUND_POWER_SAVE_ON_BAT=1
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
RUNTIME_PM_DRIVER_BLACKLIST="iwlwifi"
RESTORE_DEVICE_STATE_ON_STARTUP=1
CPU_SCALING_GOVERNOR_ON_AC=powersave
CPU_SCALING_GOVERNOR_ON_BAT=powersave
CPU_ENERGY_PERF_POLICY_ON_AC=default
CPU_ENERGY_PERF_POLICY_ON_BAT=default
USB_AUTOSUSPEND=1
```
```
[validate@validatepc ~]$ uname -a
Linux validatepc 5.5.8-200.fc31.x86_64 #1 SMP Thu Mar 5 21:28:03 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

```

4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.**
https://gist.github.com/validatedev/33a0d5a8ed27bef65871b96fc67fa739

validatedev, 2020-03-11:
Oh, did you update the package for Fedora?

linrunner, 2020-03-12:
You won't get 1.3 features from a 1.2.2. Please check with Fedora / look into the Fedora repos.

Disclaimer: I do not make the packages for the official Fedora repo and I'm not accepting bug reports for them here (nor any package requests at all).

linrunner, 2020-03-12:
ps. for F32 1.3.1 is still in the `updates-testing` repo.

closed 2020-03-12 by linrunner