linrunner/TLP #481: Network freeze completly after few minute when using Wired Ethernet Thunderbolt dongle

Sushi21, 2020-03-13:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)

**Describe the bug**

In order to improve laptop battery life I installed TLP twice on different OS
Ubuntu 18.14 and Elementary OS. Both time same thing happen. After few minutes im loosing complete internet and the entire OS is behaving weird.
If I open a terminal I just get black terminal and can't enter any commands.

And Im loosing connection for all my tools even if im both on wifi and wired. 
Only way is to hard shutdown the OS.

**Expected behavior**

TLP should work with Ethernet Dongles

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Install TLP on a laptop that is using Thunderbolt dongles for Ethernet
2. Use your OS as normal
3. After few minutes everything is gonna freeze and connection will be gone

I have that issue twice. Once removing TLP everything is back to normal.

I dont experience any crash with TLP on being only on WIFI

linrunner, 2020-03-13:
> [x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)

Seems you didn't read careful enough, because you left out crucial information from the [bug reporting template](https://github.com/linrunner/TLP/blob/master/.github/ISSUE_TEMPLATE/bug_report.md):

 1) Does the problem occur on battery or AC or both?
 4) Full output of tlp-stat via https://gist.github.com/ for all matching cases of (1).

linrunner, 2020-04-18:
Feel free to reopen when you have the required evidence.

closed 2020-04-18 by linrunner