linrunner/TLP #482: Have option to disable wake-on-LAN only on battery

LinuxOnTheDesktop, 2020-04-09:
**Is your feature request related to a problem? Please describe.**

I wished to save power when on battery but I wish to use wake-on-LAN (henceforth 'WOL') when the device (a laptop) is plugged in.

**Describe the solution you'd like**

Give WOL a battery and an AC option, just like most of the other options.

**Describe alternatives you've considered**

Leaving WOL on all the time. This is not a proper alternative, because, when on battery, I want the powersaving.

**Additional context**

```
$ tlp --version
Error: unknown command "--version".
Usage: tlp start|true|bat|false|ac|usb|bayoff|chargeonce|discharge|setcharge|fullcharge|recalibrate|diskid
```
:)

linrunner, 2020-05-07:
Hi, 

you're aware this can only work under a special boundary condition? Enabling or disabling depending on the power source must happen *before* the laptop enters suspend or shutdown, because _ethtool_ must be invoked as long as the OS runs. When the power source changes after that, the OS and TLP won't notice.

https://github.com/linrunner/TLP/blob/master/func.d/05-tlp-func-pm#L216

LinuxOnTheDesktop, 2020-05-07:
Thanks, but I do not understand. When you say that 'this can work . .', what do mean by 'this'? Note that all I requesting is this: that tlp's default configuration file give some account of how a computer's BIOS might affect TLP's wake-on-lan setting.

linrunner, 2020-05-07:
I read this as a feature request to have something like this:

    # Disable wake on LAN: Y/N.
    # Default: Y
    WOL_DISABLE_ON_AC=Y
    WOL_DISABLE_ON_BAT=N

Correct?

Regarding your last post: please forgive me that I don't want to have a "ThinkPad howto" in **tlp.conf**.

LinuxOnTheDesktop, 2020-05-07:
Well, forgive _me_: I have equivocated between (1) my original request and (2) a different request that I came to make later. Let me specify those requests here.

My original request - **1** - was indeed for what you have just presented, namely, distinct toggles within `tlp.conf` for (1a) disabling wake-on-lan on mains power, (1b) disabling wake-on-lan on battery power.

My subsequent request was (**2**) not to have such functionality, but, instead, to indicate, within `tlp.conf`, that some computers will have a BIOS option to allow wake-on-lan only on battery.

Your statement that you 'don't want to have a "ThinkPad howto" in tlp.conf' is a response, I take it, to **2**. Yet, as I have said in a previous post, I suspect that the BIOS option in question is not limited to ThinkPads (though I note also that tlp does seem to be targetted especially at ThinkPads; for, tlp has options that work only on ThinkPads, and no such manufacturer options for other computers).

I do not really understand your response to **1**. However, I take it that the thrust of that response is this: technical limitations make the proposed toggles pointless.

linrunner, 2020-05-07:
OK, about (2):

Would it be acceptable for you, if I add something like 

> How to disable WOL only on battery?

to the FAQ instead?

LinuxOnTheDesktop, 2020-05-07:
If it is really the case that option **1** does not make sense, then yes. Thanks.

linrunner, 2020-05-07:
Now about (1): 

If it makes sense or not depends on your exact use case. I'll try with an example:

1. Your laptop running and plugged into AC and LAN
2. You close the display cover, triggering a suspend to RAM
3. Because AC was connected WOL will *not* be disabled when TLP is invoked on suspend
4. You carry your laptop to another location an plug it into LAN only
Result: WOL is still enabled despite on battery

I prefer "bullet proof" features and want to spare users unexpected results.

LinuxOnTheDesktop, 2020-05-07:
Thanks. Perhaps I still misunderstand[1] but could not the following be made to happen?

5. The laptop resumes from suspend and starts using battery power. tlp notices the resume, checks the battery status, sees that the battery is being used, checks `WOL_DISABLE_ON_BAT`, finds the value there to be `Y`, and so disables wake-on-lan. I suppose that this would leave it that case that, _before_ the laptop resumes, it can be woken by the LAN - and that may not be what the user expects. Hmm.

[1] What I may misunderstand - or just not understand - is this phrase of yours: 'ethtool must be invoked as long as the OS runs'.

linrunner, 2020-05-08:
That's right, only after resume on battery WOL would be disabled.

ethtool is the utility TLP uses to switch WOL off --> https://github.com/linrunner/TLP/blob/master/func.d/05-tlp-func-pm#L216

linrunner, 2020-05-08:
I've added a note to the settings docs: https://linrunner.de/tlp/settings/network.html#wol-disable

LinuxOnTheDesktop, 2020-05-08:
Good. We can close this issue now - unless you want to close it later.

linrunner, 2020-05-08:
Thanks for your suggestion.

closed 2020-05-08 by linrunner