linrunner/TLP #483: tlp 1.3.1 + Linux 5.6.x unable recognize using BLK-MQ I/O Scheduler anymore.

FadeMind, 2020-04-10:
@linrunner hello very long time ago in https://github.com/linrunner/TLP/issues/187   (2016) I asked you for adding support for recognize I/O Scheduler named BLK-MQ. Now, after lots of major changes in kernel code this is not works as should anymore. In previous solution tlp mapping `cat /sys/class/block/sd*/queue/scheduler` but now they should check one more thing.
Checking of 
```
cat /sys/module/dm_mod/parameters/use_blk_mq
```
RETURN y = using

If BLK-MQ is in use properly, then in tree 
`tree /sys/block/sda/mq`
is available something like:
```
tree /sys/block/sda/mq
/sys/block/sda/mq
└── 0
    ├── cpu0
    ├── cpu1
    ├── cpu2
    ├── cpu3
    ├── cpu4
    ├── cpu5
    ├── cpu6
    ├── cpu7
    ├── cpu_list
    ├── nr_reserved_tags
    └── nr_tags

9 directories, 3 files
```
Otherwise BLK-MQ is not in use properly. 

Correct flags for enabling this tweaks are:
```
scsi_mod.use_blk_mq=1 dm_mod.use_blk_mq=y
```
First is for SCSI devices, second for Device Mapper devices (mostly memory cards). 

Regards 

See: https://yarondar.wordpress.com/2018/07/29/have-you-tried-blk-mq/

Regards 

Tomasz

linrunner, 2020-04-13:
Hi, thanks for your thoughts on the subject. 

1. BLK-MQ is not a scheduler but summarizes the class of multi-queue schedulers, but that's  terminology only [1]

2. Kernel 5.0 removed all the single-queue (or non-multiqueue) scheduler code, so your tweaks are irrelevant for kernels >= 5.0 [2]

3. For kernels < 5.0 it depends on what the distro's kernel package maintainer intended and configured and of course if the user added only scsi_mod.use_blk_mq=1 or also  dm_mod.use_blk_mq=y as boot option(s). However, i do not consider it a task of tlp-stat to point this out. tlp-stat correctly tells if multi-queue is enabled for the physical device and that's it

Conclusion: tlp-stat works as intended and there's no point to change it.

[1] https://wiki.ubuntu.com/Kernel/Reference/IOSchedulers
[2] https://kernelnewbies.org/Linux_5.0#Block_layer

FadeMind, 2020-04-13:
@linrunner I was wrong about WHAT bfq-mq. Apologize. Regards.

closed 2020-04-13 by FadeMind