linrunner/TLP #484: RPM packages/repositories for Fedora 32 (now in final freeze, soon to be released)

kqr, 2020-04-17:
**Is your feature request related to a problem? Please describe.**

I am running Fedora 32 beta, since it will likely be released 11 days from now. I installed TLP from the Fedora 31 TLP repositories, which seems to have worked, but installation of `akmod-acpi_call` which I desire for my ThinkPad fails with the following error:

```
Errors during downloading metadata for repository 'tlp-updates':
  - Status code: 404 for https://repo.linrunner.de/fedora/tlp/repos/updates/32/x86_64/os/repodata/repomd.xml (IP: 85.13.161.171)
  - Status code: 404 for https://repo.warpnine.de/fedora/tlp/repos/updates/32/x86_64/os/repodata/repomd.xml (IP: 92.60.39.128)
Error: Failed to download metadata for repo 'tlp-updates': Cannot download repomd.xml: Cannot download repodata/repomd.xml: All mirrors were tried
tlp RPM packages                                                                                                                                                                   279  B/s | 162  B     00:00    
Errors during downloading metadata for repository 'tlp':
  - Status code: 404 for https://repo.linrunner.de/fedora/tlp/repos/releases/32/x86_64/os/repodata/repomd.xml (IP: 85.13.161.171)
  - Status code: 404 for https://repo.warpnine.de/fedora/tlp/repos/releases/32/x86_64/os/repodata/repomd.xml (IP: 92.60.39.128)
Error: Failed to download metadata for repo 'tlp': Cannot download repomd.xml: Cannot download repodata/repomd.xml: All mirrors were tried
Ignoring repositories: tlp-updates, tlp
Last metadata expiration check: 0:19:54 ago on Fri 17 Apr 2020 09:19:03 AM CEST.
No match for argument: akmod-acpi_call
Error: Unable to find a match: akmod-acpi_call
```

I suspect this is because something auto-detects that I'm running Fedora 32 and attempts to use the corresponding repositories, which do not exist.

**Describe the solution you'd like**

I would be happy if packages/repositories were to be created for Fedora 32, since it is now unlikely anything about Fedora 32 will change before it is released.

**Describe alternatives you've considered**

I have considered manually downloading the packages and their dependencies and installing that way, but it seems overkill when Fedora 32 is soon to be released anyway.

linrunner, 2020-04-18:
The maintainer (not me) for the akmod packages will provide them when they're ready. Please be patient and restrain from opening issues for this.

closed 2020-04-18 by linrunner

kqr, 2020-04-19:
I assumed from the URL "https://repo.linrunner.de/fedora/tlp/repos/" that it was this project's area. I was clearly mistaken. Sorry!

btannous, 2020-04-28:
+1, Fedora 32 officially released today: https://fedoramagazine.org/announcing-fedora-32/