linrunner/TLP #485: Misleading error message, if battery does not discharge to exactly 0%

flat235, 2020-04-17:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)

**Describe the bug**
while recalibrating tlp reports
> Error: battery BAT0 was not discharged completely. Check your hardware.

after my battery stopped discharging further than 4%. Maybe the battery is broken, maybe it isn't and this is just some protective limit imposed by the controller. Whatever the cause, as far as I can tell, the rest of the process seems to work just fine (discharging from "full" to 4%, then charging again to full).

**Expected behavior**

The message should probably just be less harsh, i.e. "the battery did not discharge to 0%, if it stopped discharging immediately or at high state of charge, it may be broken". Other possibilities include checking for some arbitrary limit higher than 0% or finding out the limits of certain controllers (if this is the cause). In any case, the message should probably somehow convey, that the process might not have failed completely.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?
Only AC (as recalbration is only possible on AC)
2. Actions to reproduce the behaviour
`tlp recalibrate`
3. Shell commands entered and their output
`tlp recalibrate`
discharging screen until 4%, then
> Error: battery BAT0 was not discharged completely. Check your hardware.

4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.**
omitted for privacy, because probably not relevant. Can be provided if really needed.

**Additional context**
relevant issue https://github.com/linrunner/TLP/issues/438 was fixed via https://github.com/linrunner/TLP/commit/2240e90b08c4395c8045a976a536cac364c9db7a but set the boundary to exactly 0%, which I think is too strict (or at least the message is)

linrunner, 2020-04-18:
Hi, 

I can tell from more than 10 years experience with the development of TLP, with ThinkPads having the discharge feature and with related user feedback that the controller limit for all models is exactly 0%.

It also follows from the fact that *recalibration doesn't work at all* without fully discharging the battery to tell the controller (the one in the battery pack in this case) where the actual 0% is.

Therefore TLP checks for the 0% since the recalibration feature was added. The issue/change you referred to, was necessary to properly terminate the command for the special case where the embedded controller terminates the process at > 0%. 

Regarding your complaint about the message being too "harsh". I respect your opinion but i prefer messages to be short and to the point. The battery is not the only possible hardware cause. If, after checking, the user chooses to ignore the hardware issue for whatever reason, that's fine with me. 

And i certainly will *never* speculate like "battery ... may be broken" because i don't want to trigger false conclusions like "TLP broke my hardware".

What i can give you is to record the SOC percentage at termination in the message.

linrunner, 2020-04-18:
ps. just to be clear: if 0% are not reached the recalibration failed completely (see my 2nd paragraph).

flat235, 2020-04-18:
Thank you for your explanation, I respect your knowledge, it was just unclear *to me* from the issue/commit.
The message being too "harsh" was poor wording on my part, sorry for that.
I suggest including the fact, that recalibration doesn't work unless exactly 0% is reached in docs/faq.
Anyways, sorry for bothering you and thank you for your patience.

linrunner, 2020-04-18:
I indeed plan to add the explanation to the FAQ. Thanks for your input.

linrunner, 2020-04-18:
Remark @self: make shure that recalibrate properly terminates when AC is unplugged.

linrunner, 2020-04-28:
Added to the FAQ: https://linrunner.de/tlp/faq/battery.html#tlp-recalibrate-terminates-with-error-battery-was-not-discharged-completely-check-your-hardware

closed 2020-04-28 by linrunner