linrunner/TLP #487: Fedora 32 Repositories are missing.

srinathgs, 2020-04-29:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs (see below)

**Describe the bug**

I recently upgraded my OS to Fedora 32 and started seeing these errors crop up while running any `dnf` subcommands.

**Expected behavior**

Should not be getting 404s while running DNF commands

**To Reproduce**

Steps to reproduce the unexpected behavior:
Upgrade to Fedora 32 & run `sudo dnf update --refresh`

```    
tlp RPM packages Updates                                                                                                                                        69  B/s | 196  B     00:02    
Errors during downloading metadata for repository 'tlp-updates':
  - Status code: 404 for https://repo.warpnine.de/fedora/tlp/repos/updates/32/x86_64/os/repodata/repomd.xml (IP: 92.60.39.128)
  - Status code: 404 for https://repo.linrunner.de/fedora/tlp/repos/updates/32/x86_64/os/repodata/repomd.xml (IP: 85.13.161.171)
Error: Failed to download metadata for repo 'tlp-updates': Cannot download repomd.xml: Cannot download repodata/repomd.xml: All mirrors were tried
tlp RPM packages                                                                                                                                                69  B/s | 196  B     00:02    
Errors during downloading metadata for repository 'tlp':
  - Status code: 404 for https://repo.warpnine.de/fedora/tlp/repos/releases/32/x86_64/os/repodata/repomd.xml (IP: 92.60.39.128)
  - Status code: 404 for https://repo.linrunner.de/fedora/tlp/repos/releases/32/x86_64/os/repodata/repomd.xml (IP: 85.13.161.171)
Error: Failed to download metadata for repo 'tlp': Cannot download repomd.xml: Cannot download repodata/repomd.xml: All mirrors were tried
```

AwlsomeAlex, 2020-04-29:
@srinathgs I was able to install `tlp` from the RPMFusion repos in FC32, but it looks like tlp's Fedora 32 repos haven't been created yet, which is unfortunate because my ThinkPad needs akmod-acpi_call. But Fedora 32 was released like, yesterday. So all we can do is give the devs some time :) (or build from source).

linrunner, 2020-04-29:
@srinathgs : you confirm that you have read the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md), which states
> What not to report
> - Missing packages (including tp-smapi and acpi_call)

So why this issue? Are we in the *kindergarten* here?

Please read my response at https://github.com/linrunner/TLP/issues/484

linrunner, 2020-04-29:
@AwlsomeAlex : did you even bother to check if you need `acpi_call` at all? Fedora has current kernels that support `natacpi` for thresholds at least. Please take the time to read the [FAQ](https://localhost/tlp-doc/faq/battery.html#which-kernel-module-do-i-need-for-my-hardware).

linrunner, 2020-04-29:
I've added a note to the [Fedora installation instructions](https://linrunner.de/tlp/installation/fedora.html#thinkpads-only).

Thorashh, 2020-04-29:
Repository for F32 is alive.

For now, the Packages can be found in tlp-updates-testing

AwlsomeAlex, 2020-04-29:
@linrunner Using a ThinkPad T440s, so I can assume from reading the FAQ I need acpi_call. Will wait for akmod to release the module. Thank's for the hard work.

Thorashh, 2020-04-29:
**Update:** All Packages for F32 Build an available in tlp-Repo

srinathgs, 2020-04-30:
Thanks @Thorashh : A few suggestions as it didn't seem appropriate to create a ticket in this repository for the F32 issue.

1) The default name that comes when we run `dnf upgrade` is "tlp RPM packages". So, it is not clear whether RPM fusion is responsible for the TLP package or the tlp RPM packages are responsible for the TLP package. As an end user, it is not easy for me to figure out what is what. Can you please consider renaming the repo to "tlp-thinkpad-extras" or something like that? 

2) If there can be a separate repo/place to notify about these issues, it would be great so that the main TLP repo need not be flooded with issues like this. It would also be good if the links are placed in the fedora installation documentation. 

Thank you for all the hardwork!

linrunner, 2020-05-01:
@srinathgs : thanks for your suggestion, i've changed the [docs](https://linrunner.de/tlp/installation/fedora.html) to "ThinkPad Extras repo" and clarified the support status. 

In fact in there have been only two kinds of "issues" concerning the repo

* Pushing for missing release
* Handling problems installing them

I see no point in having a separate place for that, because i consider them non-issues.

linrunner, 2020-05-01:
ps. @Thorashh will change the repo subject for F33.

closed 2020-05-01 by linrunner