linrunner/TLP #488: Idle temperature 20 °C higher on battery on Ubuntu 20.04 compared to 19.10

pflanzenandi, 2020-05-07:
After reinstalling Ubuntu 20.04 I installed TLP and noticed that the temperature in idle mode is higher than on 19.10. For comparison I installed laptop-mode-tools to see if it is an Ubuntu problem. With laptop-mode-tools the temperature is on the same level as 19.10.
If you need any outputs from me, I will be happy to add them.

Dell XPS 9570

linrunner, 2020-05-07:
Hi.

I'm afraid you'll have to find out yourself, because i don't have your hardware. Compare the kernel settings which TLP and LMT produce, i'd use Powertop's "Tunables" tab and look at the "bad" ones.

The required information and outputs are described in detail in the [bug template](https://github.com/linrunner/TLP/blob/master/.github/ISSUE_TEMPLATE/bug_report.md), which you deliberately choose to ignore ...

linrunner, 2020-05-08:
Btw: i would like to know the result in order to adjust TLP's defaults eventually.

pflanzenandi, 2020-05-09:
[tlp-stat-bat](https://gist.github.com/pflanzenandi/45bf96544c9a4807c0838720e15cfdcc)

laptop-tools-tunables-bat
[image]

tlp-tunables-bat
[image]

pflanzenandi, 2020-05-10:
These settings, changed via powertop, seem to work fine.
It also removed the bug where the fans are always on while charging.

[image]

linrunner, 2020-05-10:
The usual suspect, Nvidia Optimus. I hope you have a little bit of time to get to the root of this.

> /sys/bus/pci/devices/0000:01:00.0/power/control = on   (0x030200, 3D controller, no driver)

I wonder why Runtime PM is off (control=on). There is no driver loaded, this should rule out RUNTIME_PM_DRIVER_BLACKLIST as the cause. You didn't employ RUNTIME_PM_BLACKLIST on the device either. 

My working hypothesis is, that the driver is unloaded somewhere during boot. Questions:

1. Is the proprietary driver *nvidia* installed?
2. Did you configure something like this (to prevent the driver from loading):

**/etc/modprobe.d/blacklist-nvidia-nouveau.conf**

    blacklist nouveau
    options nouveau modeset=0 

3. Did you install some tool to switch the Nvidia discrete graphics off? Which one?
4. Do you use the Nvidia at all (in Linux)?

ps. please _never_ abbreviate the tlp-stat outputs. I'm missing version no. and the configuration.

pflanzenandi, 2020-05-10:
[updated-tlp-stat-bat](https://gist.github.com/pflanzenandi/45bf96544c9a4807c0838720e15cfdcc)

1. Yes, the nvidia-driver-440 (proprietary, tested) ships with a fresh Ubuntu 20.04 installation. It is also activated automatically after first boot of Ubuntu.

[image]

2. No, I haven't changed anything regarding GPU drivers. Just the fresh 20.04 installation. But I deactivated the nvidia gpu via X server settings. I also have no blacklist-nvidia-nouveau.conf in modprobe.d. I also checked all other existing blacklist configs and none of them has a nouveau value set.

[image]

3. No, my set up is the default ubuntu 20.04.
4. No, only the intel gpu.

linrunner, 2020-05-10:
Well. Seems to be a logic inclued into the (Ubuntu) proprietary driver package.

1. Please show in the actual state of things:

      cat /sys/kernel/debug/vgaswitcheroo/switch

2. Remove only the `nvidia` driver from the blacklist:

       RUNTIME_PM_DRIVER_BLACKLIST="amdgpu mei_me nouveau pcieport radeon"

3. Reboot and show:

      tlp-stat -e
      cat /sys/kernel/debug/vgaswitcheroo/switch

pflanzenandi, 2020-05-10:
```
#
# Configuration file for Laptop Mode Tools module vgaswitcheroo
#
# For more information, consult the laptop-mode.conf(8) manual page.
#

###############################################################################
# vgaswitcheroo control settings
# -----------------------------
#
#__COMMENT vga_switcheroo is the Linux subsystem for laptop hybrid graphics.
#__COMMENT For hybrid graphics machines, the discrete graphics chip usually is
#__COMMENT idle most of the time.
#__COMMENT Enable this module to switch off the unused graphics card, when not in use
#__COMMENT Note: You need to ensure debugfs is enabled/mounted on your system
#
###############################################################################

# Enable debug mode for this module
# Set to 1 if you want to debug this module
DEBUG=0

# Control video output settings?
CONTROL_VGASWITCHEROO=0

# Modes of operation for Hybrid Graphics
# The default settings will switch off discrete graphics, when on battery
# and not in use
BATT_ENABLE_VGASWITCHEROO=1
LM_AC_ENABLE_VGASWITCHEROO=0
NOLM_AC_ENABLE_VGASWITCHEROO=0
VGASWITCHEROO_FILE="/sys/kernel/debug/vgaswitcheroo/switch"
```

pflanzenandi, 2020-05-10:
```
$ tlp-stat -e
--- TLP 1.3.1 --------------------------------------------

+++ PCIe Active State Power Management
/sys/module/pcie_aspm/parameters/policy = [default] performance powersave powersupersave (using BIOS preferences)

+++ Runtime Power Management
Device blacklist = (not configured)
Driver blacklist = amdgpu mei_me nouveau nvidia pcieport radeon

/sys/bus/pci/devices/0000:00:00.0/power/control = on   (0x060000, Host bridge, skl_uncore)
/sys/bus/pci/devices/0000:00:01.0/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:02.0/power/control = on   (0x030000, VGA compatible controller, i915)
/sys/bus/pci/devices/0000:00:04.0/power/control = on   (0x118000, Signal processing controller, proc_thermal)
/sys/bus/pci/devices/0000:00:08.0/power/control = on   (0x088000, System peripheral, no driver)
/sys/bus/pci/devices/0000:00:12.0/power/control = on   (0x118000, Signal processing controller, intel_pch_thermal)
/sys/bus/pci/devices/0000:00:14.0/power/control = on   (0x0c0330, USB controller, xhci_hcd)
/sys/bus/pci/devices/0000:00:14.2/power/control = on   (0x050000, RAM memory, no driver)
/sys/bus/pci/devices/0000:00:15.0/power/control = on   (0x0c8000, Serial bus controller [0c80], intel-lpss)
/sys/bus/pci/devices/0000:00:15.1/power/control = on   (0x0c8000, Serial bus controller [0c80], intel-lpss)
/sys/bus/pci/devices/0000:00:16.0/power/control = auto (0x078000, Communication controller, mei_me)
/sys/bus/pci/devices/0000:00:17.0/power/control = on   (0x010601, SATA controller, ahci)
/sys/bus/pci/devices/0000:00:1b.0/power/control = on   (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1c.0/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1d.0/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1f.0/power/control = on   (0x060100, ISA bridge, no driver)
/sys/bus/pci/devices/0000:00:1f.3/power/control = on   (0x040380, Audio device, snd_hda_intel)
/sys/bus/pci/devices/0000:00:1f.4/power/control = on   (0x0c0500, SMBus, i801_smbus)
/sys/bus/pci/devices/0000:00:1f.5/power/control = on   (0x0c8000, Serial bus controller [0c80], no driver)
/sys/bus/pci/devices/0000:01:00.0/power/control = on   (0x030200, 3D controller, no driver)
/sys/bus/pci/devices/0000:02:00.0/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:03:00.0/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:03:01.0/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:03:02.0/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:04:00.0/power/control = auto (0x088000, System peripheral, thunderbolt)
/sys/bus/pci/devices/0000:3a:00.0/power/control = auto (0x0c0330, USB controller, xhci_hcd)
/sys/bus/pci/devices/0000:3b:00.0/power/control = on   (0x028000, Network controller, ath10k_pci)
/sys/bus/pci/devices/0000:3c:00.0/power/control = on   (0x010802, Non-Volatile memory controller, nvme)
```

pflanzenandi, 2020-05-10:
Sorry, I think this is the wrong file. I do not have /sys/kernel/debug/vgaswitcheroo/switch on my system.

linrunner, 2020-05-10:
> # Configuration file for Laptop Mode Tools module vgaswitcheroo
That's obviously the contents of **/sys/kernel/debug/vgaswitcheroo/switch** but something from LMT's configuration files.

Did you execute the 

    cat /sys/kernel/debug/vgaswitcheroo/switch

command as root?

Also you didn't change TLP's configuration, because `nvidia` is still in the blacklist:
> Driver blacklist = amdgpu mei_me nouveau nvidia pcieport radeon

Last but not least: did you boot on BAT? My impression was the first `tlp-stat -e` output you later changed was on AC.

pflanzenandi, 2020-05-10:
Yes, I removed the blacklist:
[tlp-config](https://gist.github.com/pflanzenandi/fa67d9e4f531836bec7f03335ed51721)

```
me@myself:~$ sudo cat /sys/kernel/debug/vgaswitcheroo/switch
[sudo] password for me: 
cat: /sys/kernel/debug/vgaswitcheroo/switch: No such file or directory
```

linrunner, 2020-05-10:
Removing the whole line doesn't work at all, because the intrinsic defaults kicks in. 

Anyway I don't want *all* drivers removed from the list at once. Just (re)enter the following line into **tlp.conf**:

     RUNTIME_PM_DRIVER_BLACKLIST="amdgpu mei_me nouveau pcieport radeon"

(In the original file the leading *#* has to be removed to activate the change)

pflanzenandi, 2020-05-10:
Aye, I will change it later and update you.

linrunner, 2020-05-10:
Take your time.

pflanzenandi, 2020-05-10:
I removed the #, restarted and ran tlp-stat -e. But the output looks identical.

[image]

```
me@myself:~$ tlp-stat -e
--- TLP 1.3.1 --------------------------------------------

+++ PCIe Active State Power Management
/sys/module/pcie_aspm/parameters/policy = [default] performance powersave powersupersave (using BIOS preferences)

+++ Runtime Power Management
Device blacklist = (not configured)
Driver blacklist = amdgpu mei_me nouveau nvidia pcieport radeon

/sys/bus/pci/devices/0000:00:00.0/power/control = auto (0x060000, Host bridge, skl_uncore)
/sys/bus/pci/devices/0000:00:01.0/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:02.0/power/control = auto (0x030000, VGA compatible controller, i915)
/sys/bus/pci/devices/0000:00:04.0/power/control = auto (0x118000, Signal processing controller, proc_thermal)
/sys/bus/pci/devices/0000:00:08.0/power/control = auto (0x088000, System peripheral, no driver)
/sys/bus/pci/devices/0000:00:12.0/power/control = auto (0x118000, Signal processing controller, intel_pch_thermal)
/sys/bus/pci/devices/0000:00:14.0/power/control = auto (0x0c0330, USB controller, xhci_hcd)
/sys/bus/pci/devices/0000:00:14.2/power/control = auto (0x050000, RAM memory, no driver)
/sys/bus/pci/devices/0000:00:15.0/power/control = auto (0x0c8000, Serial bus controller [0c80], intel-lpss)
/sys/bus/pci/devices/0000:00:15.1/power/control = auto (0x0c8000, Serial bus controller [0c80], intel-lpss)
/sys/bus/pci/devices/0000:00:16.0/power/control = auto (0x078000, Communication controller, mei_me)
/sys/bus/pci/devices/0000:00:17.0/power/control = auto (0x010601, SATA controller, ahci)
/sys/bus/pci/devices/0000:00:1b.0/power/control = on   (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1c.0/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1d.0/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1f.0/power/control = auto (0x060100, ISA bridge, no driver)
/sys/bus/pci/devices/0000:00:1f.3/power/control = auto (0x040380, Audio device, snd_hda_intel)
/sys/bus/pci/devices/0000:00:1f.4/power/control = auto (0x0c0500, SMBus, i801_smbus)
/sys/bus/pci/devices/0000:00:1f.5/power/control = auto (0x0c8000, Serial bus controller [0c80], no driver)
/sys/bus/pci/devices/0000:01:00.0/power/control = on   (0x030200, 3D controller, no driver)
/sys/bus/pci/devices/0000:3b:00.0/power/control = auto (0x028000, Network controller, ath10k_pci)
/sys/bus/pci/devices/0000:3c:00.0/power/control = auto (0x010802, Non-Volatile memory controller, nvme)
```

linrunner, 2020-05-10:
Compare your line to my [post](https://github.com/linrunner/TLP/issues/488#issuecomment-626316464). You didn't remove the 'nvidia' from the string.

pflanzenandi, 2020-05-10:
```
me@myself:~$ tlp-stat -e
--- TLP 1.3.1 --------------------------------------------

+++ PCIe Active State Power Management
/sys/module/pcie_aspm/parameters/policy = [default] performance powersave powersupersave (using BIOS preferences)

+++ Runtime Power Management
Device blacklist = (not configured)
Driver blacklist = amdgpu mei_me nouveau pcieport radeon

/sys/bus/pci/devices/0000:00:00.0/power/control = auto (0x060000, Host bridge, skl_uncore)
/sys/bus/pci/devices/0000:00:01.0/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:02.0/power/control = auto (0x030000, VGA compatible controller, i915)
/sys/bus/pci/devices/0000:00:04.0/power/control = auto (0x118000, Signal processing controller, proc_thermal)
/sys/bus/pci/devices/0000:00:08.0/power/control = auto (0x088000, System peripheral, no driver)
/sys/bus/pci/devices/0000:00:12.0/power/control = auto (0x118000, Signal processing controller, intel_pch_thermal)
/sys/bus/pci/devices/0000:00:14.0/power/control = auto (0x0c0330, USB controller, xhci_hcd)
/sys/bus/pci/devices/0000:00:14.2/power/control = auto (0x050000, RAM memory, no driver)
/sys/bus/pci/devices/0000:00:15.0/power/control = auto (0x0c8000, Serial bus controller [0c80], intel-lpss)
/sys/bus/pci/devices/0000:00:15.1/power/control = auto (0x0c8000, Serial bus controller [0c80], intel-lpss)
/sys/bus/pci/devices/0000:00:16.0/power/control = auto (0x078000, Communication controller, mei_me)
/sys/bus/pci/devices/0000:00:17.0/power/control = auto (0x010601, SATA controller, ahci)
/sys/bus/pci/devices/0000:00:1b.0/power/control = on   (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1c.0/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1d.0/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1f.0/power/control = auto (0x060100, ISA bridge, no driver)
/sys/bus/pci/devices/0000:00:1f.3/power/control = auto (0x040380, Audio device, snd_hda_intel)
/sys/bus/pci/devices/0000:00:1f.4/power/control = auto (0x0c0500, SMBus, i801_smbus)
/sys/bus/pci/devices/0000:00:1f.5/power/control = auto (0x0c8000, Serial bus controller [0c80], no driver)
/sys/bus/pci/devices/0000:01:00.0/power/control = on   (0x030200, 3D controller, no driver)
/sys/bus/pci/devices/0000:3b:00.0/power/control = auto (0x028000, Network controller, ath10k_pci)
/sys/bus/pci/devices/0000:3c:00.0/power/control = auto (0x010802, Non-Volatile memory controller, nvme)

```

linrunner, 2020-05-11:
> /sys/bus/pci/devices/0000:01:00.0/power/control = on   (0x030200, 3D controller, no driver)

The device is still "on", setting didn't apply.

Please try

    sudo tlp start

and show 

    tlp-stat -e 

again.

pflanzenandi, 2020-05-11:
```
me@myself:~$ sudo tlp start
[sudo] password for me: 
TLP started in battery mode (auto).
me@myself:~$ tlp-stat -s -e 
--- TLP 1.3.1 --------------------------------------------

+++ System Info
System         = Dell Inc.  XPS 15 9570
BIOS           = 1.15.0
Release        = Ubuntu 20.04 LTS
Kernel         = 5.4.0-29-generic #33-Ubuntu SMP Wed Apr 29 14:32:27 UTC 2020 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-5.4.0-29-generic root=UUID=19bfa1aa-2a3a-48aa-8116-0a13e16a02af ro quiet splash vt.handoff=7
Init system    = systemd v245 (245.4-4ubuntu3)
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 15:27:53,      8 sec(s) ago
Mode           = battery
Power source   = battery

+++ PCIe Active State Power Management
/sys/module/pcie_aspm/parameters/policy = [default] performance powersave powersupersave (using BIOS preferences)

+++ Runtime Power Management
Device blacklist = (not configured)
Driver blacklist = amdgpu mei_me nouveau pcieport radeon

/sys/bus/pci/devices/0000:00:00.0/power/control = auto (0x060000, Host bridge, skl_uncore)
/sys/bus/pci/devices/0000:00:01.0/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:02.0/power/control = auto (0x030000, VGA compatible controller, i915)
/sys/bus/pci/devices/0000:00:04.0/power/control = auto (0x118000, Signal processing controller, proc_thermal)
/sys/bus/pci/devices/0000:00:08.0/power/control = auto (0x088000, System peripheral, no driver)
/sys/bus/pci/devices/0000:00:12.0/power/control = auto (0x118000, Signal processing controller, intel_pch_thermal)
/sys/bus/pci/devices/0000:00:14.0/power/control = auto (0x0c0330, USB controller, xhci_hcd)
/sys/bus/pci/devices/0000:00:14.2/power/control = auto (0x050000, RAM memory, no driver)
/sys/bus/pci/devices/0000:00:15.0/power/control = auto (0x0c8000, Serial bus controller [0c80], intel-lpss)
/sys/bus/pci/devices/0000:00:15.1/power/control = auto (0x0c8000, Serial bus controller [0c80], intel-lpss)
/sys/bus/pci/devices/0000:00:16.0/power/control = auto (0x078000, Communication controller, mei_me)
/sys/bus/pci/devices/0000:00:17.0/power/control = auto (0x010601, SATA controller, ahci)
/sys/bus/pci/devices/0000:00:1b.0/power/control = on   (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1c.0/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1d.0/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1f.0/power/control = auto (0x060100, ISA bridge, no driver)
/sys/bus/pci/devices/0000:00:1f.3/power/control = auto (0x040380, Audio device, snd_hda_intel)
/sys/bus/pci/devices/0000:00:1f.4/power/control = auto (0x0c0500, SMBus, i801_smbus)
/sys/bus/pci/devices/0000:00:1f.5/power/control = auto (0x0c8000, Serial bus controller [0c80], no driver)
/sys/bus/pci/devices/0000:01:00.0/power/control = on   (0x030200, 3D controller, no driver)
/sys/bus/pci/devices/0000:3b:00.0/power/control = auto (0x028000, Network controller, ath10k_pci)
/sys/bus/pci/devices/0000:3c:00.0/power/control = auto (0x010802, Non-Volatile memory controller, nvme)
```

linrunner, 2020-05-11:
I was wrong, sorry. You have to remove both nvidia and nouveau so that the line looks like

    RUNTIME_PM_DRIVER_BLACKLIST="amdgpu mei_me pcieport radeon"

After reboot show:

    tlp-stat -s -e

pflanzenandi, 2020-05-11:
```
me@myself:~$ tlp-stat -s -e
--- TLP 1.3.1 --------------------------------------------

+++ System Info
System         = Dell Inc.  XPS 15 9570
BIOS           = 1.15.0
Release        = Ubuntu 20.04 LTS
Kernel         = 5.4.0-29-generic #33-Ubuntu SMP Wed Apr 29 14:32:27 UTC 2020 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-5.4.0-29-generic root=UUID=19bfa1aa-2a3a-48aa-8116-0a13e16a02af ro quiet splash vt.handoff=7
Init system    = systemd v245 (245.4-4ubuntu3)
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 18:45:19,     32 sec(s) ago
Mode           = battery
Power source   = battery

+++ PCIe Active State Power Management
/sys/module/pcie_aspm/parameters/policy = [default] performance powersave powersupersave (using BIOS preferences)

+++ Runtime Power Management
Device blacklist = (not configured)
Driver blacklist = amdgpu mei_me pcieport radeon

/sys/bus/pci/devices/0000:00:00.0/power/control = auto (0x060000, Host bridge, skl_uncore)
/sys/bus/pci/devices/0000:00:01.0/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:02.0/power/control = auto (0x030000, VGA compatible controller, i915)
/sys/bus/pci/devices/0000:00:04.0/power/control = auto (0x118000, Signal processing controller, proc_thermal)
/sys/bus/pci/devices/0000:00:08.0/power/control = auto (0x088000, System peripheral, no driver)
/sys/bus/pci/devices/0000:00:12.0/power/control = auto (0x118000, Signal processing controller, intel_pch_thermal)
/sys/bus/pci/devices/0000:00:14.0/power/control = auto (0x0c0330, USB controller, xhci_hcd)
/sys/bus/pci/devices/0000:00:14.2/power/control = auto (0x050000, RAM memory, no driver)
/sys/bus/pci/devices/0000:00:15.0/power/control = auto (0x0c8000, Serial bus controller [0c80], intel-lpss)
/sys/bus/pci/devices/0000:00:15.1/power/control = auto (0x0c8000, Serial bus controller [0c80], intel-lpss)
/sys/bus/pci/devices/0000:00:16.0/power/control = auto (0x078000, Communication controller, mei_me)
/sys/bus/pci/devices/0000:00:17.0/power/control = auto (0x010601, SATA controller, ahci)
/sys/bus/pci/devices/0000:00:1b.0/power/control = on   (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1c.0/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1d.0/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1f.0/power/control = auto (0x060100, ISA bridge, no driver)
/sys/bus/pci/devices/0000:00:1f.3/power/control = auto (0x040380, Audio device, snd_hda_intel)
/sys/bus/pci/devices/0000:00:1f.4/power/control = auto (0x0c0500, SMBus, i801_smbus)
/sys/bus/pci/devices/0000:00:1f.5/power/control = auto (0x0c8000, Serial bus controller [0c80], no driver)
/sys/bus/pci/devices/0000:01:00.0/power/control = auto (0x030200, 3D controller, no driver)
/sys/bus/pci/devices/0000:3b:00.0/power/control = auto (0x028000, Network controller, ath10k_pci)
/sys/bus/pci/devices/0000:3c:00.0/power/control = auto (0x010802, Non-Volatile memory controller, nvme)

```

pflanzenandi, 2020-05-11:
The temps are at 47 now on battery and idle, compared to 60 before. Tunables look like this:

[image]
:

linrunner, 2020-05-11:
Very well. 

The "bad" for the "RuntimePM PCI Express Root Port #17" can be removed by un-blacklisting "pcieport" 
i.e. 

    RUNTIME_PM_DRIVER_BLACKLIST="amdgpu mei_me radeon"

and "RuntimePM port ata 1 SATA AHCI Controller" is adressed by uncommenting the lines

    AHCI_RUNTIME_PM_ON_AC=on
    AHCI_RUNTIME_PM_ON_BAT=auto

Note the change of the 2nd line to "auto".

Activate the changes with

    sudo tlp start

and show

    sudo tlp-stat -s -d -e

Check with Powertop as well.

Note @self: remove nvidia nouveau from defaults.conf and add this case to the FAQ.

pflanzenandi, 2020-05-11:
```
--- TLP 1.3.1 --------------------------------------------

+++ System Info
System         = Dell Inc.  XPS 15 9570
BIOS           = 1.15.0
Release        = Ubuntu 20.04 LTS
Kernel         = 5.4.0-29-generic #33-Ubuntu SMP Wed Apr 29 14:32:27 UTC 2020 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-5.4.0-29-generic root=UUID=19bfa1aa-2a3a-48aa-8116-0a13e16a02af ro quiet splash vt.handoff=7
Init system    = systemd v245 (245.4-4ubuntu3)
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 20:30:45,      3 sec(s) ago
Mode           = battery
Power source   = battery

+++ Storage Devices
Devices = nvme0n1 sda

/dev/nvme0n1:
  Type      = NVMe
  Model     = KXG50ZNV256G NVMe TOSHIBA 256GB         
  Firmware  = AADA4107
  Scheduler = [none] mq-deadline (multi queue)

  Runtime PM: control = auto, autosuspend_delay_ms = (not available)

  SMART info:
    Critical Warning:                   0x00
    Temperature:                        29 Celsius
    Available Spare:                    100%
    Available Spare Threshold:          10%
    Percentage Used:                    14%
    Data Units Written:                 8.439.570 [4,32 TB]
    Power Cycles:                       1.255
    Power On Hours:                     3.750
    Unsafe Shutdowns:                   375
    Media and Data Integrity Errors:    0

/dev/sda: not present.

+++ AHCI Link Power Management (ALPM)
/sys/class/scsi_host/host0/link_power_management_policy  = med_power_with_dipm

+++ AHCI Host Controller Runtime Power Management
/sys/bus/pci/devices/0000:00:17.0/ata1/power/control = on

+++ PCIe Active State Power Management
/sys/module/pcie_aspm/parameters/policy = [default] performance powersave powersupersave (using BIOS preferences)

+++ Runtime Power Management
Device blacklist = (not configured)
Driver blacklist = amdgpu mei_me radeon

/sys/bus/pci/devices/0000:00:00.0/power/control = auto (0x060000, Host bridge, skl_uncore)
/sys/bus/pci/devices/0000:00:01.0/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:02.0/power/control = auto (0x030000, VGA compatible controller, i915)
/sys/bus/pci/devices/0000:00:04.0/power/control = auto (0x118000, Signal processing controller, proc_thermal)
/sys/bus/pci/devices/0000:00:08.0/power/control = auto (0x088000, System peripheral, no driver)
/sys/bus/pci/devices/0000:00:12.0/power/control = auto (0x118000, Signal processing controller, intel_pch_thermal)
/sys/bus/pci/devices/0000:00:14.0/power/control = auto (0x0c0330, USB controller, xhci_hcd)
/sys/bus/pci/devices/0000:00:14.2/power/control = auto (0x050000, RAM memory, no driver)
/sys/bus/pci/devices/0000:00:15.0/power/control = auto (0x0c8000, Serial bus controller [0c80], intel-lpss)
/sys/bus/pci/devices/0000:00:15.1/power/control = auto (0x0c8000, Serial bus controller [0c80], intel-lpss)
/sys/bus/pci/devices/0000:00:16.0/power/control = auto (0x078000, Communication controller, mei_me)
/sys/bus/pci/devices/0000:00:17.0/power/control = auto (0x010601, SATA controller, ahci)
/sys/bus/pci/devices/0000:00:1b.0/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1c.0/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1d.0/power/control = auto (0x060400, PCI bridge, pcieport)
/sys/bus/pci/devices/0000:00:1f.0/power/control = auto (0x060100, ISA bridge, no driver)
/sys/bus/pci/devices/0000:00:1f.3/power/control = auto (0x040380, Audio device, snd_hda_intel)
/sys/bus/pci/devices/0000:00:1f.4/power/control = auto (0x0c0500, SMBus, i801_smbus)
/sys/bus/pci/devices/0000:00:1f.5/power/control = auto (0x0c8000, Serial bus controller [0c80], no driver)
/sys/bus/pci/devices/0000:01:00.0/power/control = auto (0x030200, 3D controller, no driver)
/sys/bus/pci/devices/0000:3b:00.0/power/control = auto (0x028000, Network controller, ath10k_pci)
/sys/bus/pci/devices/0000:3c:00.0/power/control = auto (0x010802, Non-Volatile memory controller, nvme)``

pflanzenandi, 2020-05-11:
[image]

linrunner, 2020-05-12:
Correct, this will only work with 1.4: https://github.com/linrunner/TLP/commit/16c3fb5f616aaf088032d5238e6da5b01142df5a

I hope that the measures taken so far are sufficient for you.

Thanks for reporting this. Please leave the issue open.

pflanzenandi, 2020-05-12:
This was fun, glad I could help.

pflanzenandi, 2020-05-14:
@linrunner 
How can I blacklist 'nouveau & nvidia' automatically on AC?
Because 'RUNTIME_PM_DRIVER_BLACKLIST="amdgpu mei_me radeon"' is only activated on BAT.

linrunner, 2020-05-14:
The blacklists do not allow differentiation between AC and BAT. But you may use

    RUNTIME_PM_ON_AC=auto

pflanzenandi, 2020-05-14:
This setting worked, but my fans are still spinning. I think it has to to with when tlp starts while booting. But this has nothing to do with tlp. Fuck dell, never again :)

linrunner, 2020-05-14:
There will always be users complaining about fans and for all laptop brands. But TLP is in the power saving business not in the fan control business  ... ;)

pflanzenandi, 2020-05-14:
Aye, keep up the work!

linrunner, 2020-06-14:
Blacklist adjusted, new settings to permanently enable/disable based on address, FAQ upgraded. I think that's enough for now :).

closed 2020-06-14 by linrunner