linrunner/TLP #489: acpi-call-dkms is not working on kernel 5.6.11

LinuxOnTheDesktop, 2020-05-13:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs (see below)

**Describe the bug**

 `tlp-stat -b` recommends I install the 'acpi_call kernel module'. I determined that on my system package 'acpi-call-dkms' provides that module. That package is installed already. Reinstalling it throws an error.

**Expected behavior**

The installed package should work or at least reinstallation should work. If the kernel is the problem - I suspect it is - it would be nice if tlp told one the most recent kernel that will work.

**To Reproduce**

Have Linux kernel 5.6.11-050611-generic. Install tlp and 'acpi-call-dkms' (or - if this bug is cross-distro - whatever package for one's distro provides the module at issue).

```
$ sudo tlp-stat -b
--- TLP 1.3.1 --------------------------------------------

+++ Battery Features: Charge Thresholds and Recalibrate
natacpi    = active (data, thresholds)
tpacpi-bat = inactive (kernel module 'acpi_call' not installed)
tp-smapi   = inactive (ThinkPad not supported)

+++ ThinkPad Battery Status: BAT0
/sys/class/power_supply/BAT0/manufacturer                   = LGC
/sys/class/power_supply/BAT0/model_name                     = 01AV494
/sys/class/power_supply/BAT0/cycle_count                    =    145
/sys/class/power_supply/BAT0/energy_full_design             =  57000 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  52460 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  40400 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Unknown

/sys/class/power_supply/BAT0/charge_start_threshold         =     65 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold          =     93 [%]

Charge                                                      =   77.0 [%]
Capacity                                                    =   92.0 [%]

+++ Recommendations
* Install acpi_call kernel module for ThinkPad battery recalibration
```

**Additional context**

```
$ sudo apt install acpi-call-dkms --reinstall
[sudo] password for nicholas: 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
0 to upgrade, 0 to newly install, 1 reinstalled, 0 to remove and 0 not to upgrade.
Need to get 13.2 kB of archives.
After this operation, 0 B of additional disk space will be used.
Get:1 http://archive.ubuntu.com/ubuntu bionic/universe amd64 acpi-call-dkms all 1.1.0-4 [13.2 kB]
Fetched 13.2 kB in 1s (14.7 kB/s)         
(Reading database ... 468443 files and directories currently installed.)
Preparing to unpack .../acpi-call-dkms_1.1.0-4_all.deb ...

-------- Uninstall Beginning --------
Module:  acpi-call
Version: 1.1.0
Kernel:  4.15.0-96-generic (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

acpi_call.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.15.0-96-generic/extra/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod...

DKMS: uninstall completed.

-------- Uninstall Beginning --------
Module:  acpi-call
Version: 1.1.0
Kernel:  4.15.0-99-generic (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

acpi_call.ko:
 - Uninstallation
   - Deleting from: /lib/modules/4.15.0-99-generic/extra/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod...

DKMS: uninstall completed.

-------- Uninstall Beginning --------
Module:  acpi-call
Version: 1.1.0
Kernel:  5.3.0-51-generic (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

acpi_call.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.3.0-51-generic/extra/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod...

DKMS: uninstall completed.

-------- Uninstall Beginning --------
Module:  acpi-call
Version: 1.1.0
Kernel:  5.5.12-050512-generic (x86_64)
-------------------------------------

Status: Before uninstall, this module version was ACTIVE on this kernel.

acpi_call.ko:
 - Uninstallation
   - Deleting from: /lib/modules/5.5.12-050512-generic/extra/
 - Original module
   - No original module was found for this module on this kernel.
   - Use the dkms install command to reinstall any previous module version.

depmod...

DKMS: uninstall completed.

------------------------------
Deleting module version: 1.1.0
completely from the DKMS tree.
------------------------------
Done.
Unpacking acpi-call-dkms (1.1.0-4) over (1.1.0-4) ...
Setting up acpi-call-dkms (1.1.0-4) ...
Loading new acpi-call-1.1.0 DKMS files...
Building for 5.6.11-050611-generic
Building initial module for 5.6.11-050611-generic
Error! Bad return status for module build on kernel: 5.6.11-050611-generic (x86_64)
Consult /var/lib/dkms/acpi-call/1.1.0/build/make.log for more information.
```

```
KMS make.log for acpi-call-1.1.0 for kernel 5.6.11-050611-generic (x86_64)
Wed 13 May 02:42:39 BST 2020
make: Entering directory '/usr/src/linux-headers-5.6.11-050611-generic'
  AR      /var/lib/dkms/acpi-call/1.1.0/build/built-in.a
  CC [M]  /var/lib/dkms/acpi-call/1.1.0/build/acpi_call.o
/var/lib/dkms/acpi-call/1.1.0/build/acpi_call.c: In function ‘init_acpi_call’:
/var/lib/dkms/acpi-call/1.1.0/build/acpi_call.c:355:53: error: passing argument 4 of ‘proc_create’ from incompatible pointer type [-Werror=incompatible-pointer-types]
  355 |                                                     &proc_acpi_operations);
      |                                                     ^~~~~~~~~~~~~~~~~~~~~
      |                                                     |
      |                                                     struct file_operations *
In file included from /var/lib/dkms/acpi-call/1.1.0/build/acpi_call.c:6:
./include/linux/proc_fs.h:64:24: note: expected ‘const struct proc_ops *’ but argument is of type ‘struct file_operations *’
   64 | struct proc_dir_entry *proc_create(const char *name, umode_t mode, struct proc_dir_entry *parent, const struct proc_ops *proc_ops);
      |                        ^~~~~~~~~~~
cc1: some warnings being treated as errors
scripts/Makefile.build:267: recipe for target '/var/lib/dkms/acpi-call/1.1.0/build/acpi_call.o' failed
make[1]: *** [/var/lib/dkms/acpi-call/1.1.0/build/acpi_call.o] Error 1
Makefile:1683: recipe for target '/var/lib/dkms/acpi-call/1.1.0/build' failed
make: *** [/var/lib/dkms/acpi-call/1.1.0/build] Error 2
make: Leaving directory '/usr/src/linux-headers-5.6.11-050611-generic'
```

linrunner, 2020-05-13:
Hi, 

acpi_call is not provided by the TLP project and neither are the distribution packages for it. 
Btw the acpi_call upstream project was abandoned years ago. 

You installed **acpi-call-dkms** from the official Ubuntu repos, so please file your issue there.
Thanks for your understanding.

That said, I can give you the hint that the Debian maintainer fixed this in unstable a week ago, might as well been propagated to Ubuntu Groovy already.

closed 2020-05-13 by linrunner

LinuxOnTheDesktop, 2020-05-13:
Thanks. Perhaps I should have said I am on Mint 19 (which is based on Ubuntu Bionic). I have amended the above to say that. I do not fully understand your comments but I have filed a bug with Ubuntu [here](https://bugs.launchpad.net/ubuntu/+source/acpi-call/+bug/1878396). Possibly the problem will be solved with Mint 20, which is due out soon and has a new Ubuntu base. In the meantime: _in the absence of this package/module, will battery thresholds work on my Thinkpad (X1 Carbon 6th Generation)?_