linrunner/TLP #490: Intel_pstate: x86_energy_perf_policy vs TLP

DiegoGuidaF, 2020-05-13:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs (see below)

**Describe the bug**
When setting CPU_ENERGY_PERF_POLICY via TLP, values shown in x86_energy_perf_policy defer from setting it directly via x86_energy_perf_policy.

When TLP sets balance-power (I have trimmed all the CPUs and will only show one CPU):
```
$ sudo x86_energy_perf_policy
cpu0: EPB 6
cpu0: HWP_REQ: min 4 max 46 des 0 epp 192 window 0x0 (0*10^0us) use_pkg 0
cpu0: HWP_CAP: low 1 eff 8 guar 18 high 46
```

When setting balance-power directly via:
`sudo x86_energy_perf_policy --all "balance-power"`:
```
$ sudo x86_energy_perf_policy
cpu0: EPB 8
cpu0: HWP_REQ: min 0 max 255 des 0 epp 192 window 0x0 (0*10^0us) use_pkg 0
cpu0: HWP_CAP: low 1 eff 8 guar 18 high 46
```

**Expected behavior**
If I have understood it correctly, which I might have not, both TLP and x86_energy_perf_policy should produce the same output for `sudo x86_energy_perf_policy` command with the exception of EPB value (which is ignored since my CPU is using EPP).

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?
Both
2. Actions to reproduce the behaviour

- Plug laptop to battery
- Check output of `$ sudo x86_energy_perf_policy`
- Manually set mode via `sudo x86_energy_perf_policy --all "balance-power"`
- Compare values

4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.**
Following Gist contains two files with tlp-stat output.
First one is the result of battery plugged in and TLP loading default config
Second one is after setting the sudo x86_energy_perf_policy --all "balance-power" manually.
https://gist.github.com/DiegoGuidaF/edc9f12afcca455b85ca4d6e1b852457

DiegoGuidaF, 2020-05-13:
Okey, have just seen that when EPP is working EPB is not set, I have read that is so since intel_pstate ignores the EPB value.
I have not been able to find much information regarding the meanings of the values seen here:
```
$ sudo x86_energy_perf_policy
cpu0: EPB 8
cpu0: HWP_REQ: min 0 max 255 des 0 epp 192 window 0x0 (0*10^0us) use_pkg 0
cpu0: HWP_CAP: low 1 eff 8 guar 18 high 46
```

Could anybody shed some light over their meaning?
Nevertheless, you can close the issue it seems since there doesn't seem to be any bug.

linrunner, 2020-05-13:
Hi, 

imho your understanding is wrong. This may be because the intel_pstate matter is quite complicated:

TLP does not employ `x86_energy_perf_policy` for your combination of kernel and CPU, but rather writes kernel settings directly. Look at your outputs:

> /sys/devices/system/cpu/cpu0/cpufreq/energy_performance_preference = balance_power [HWP.EPP]

Relevant configuration is:

> /etc/tlp.conf L0101: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"

The reason for the differing EPB values is: 

TLP is addressing the HWP.EPP feature of the CPU, not EPB as x86_energy_perf_policy does. The EPB values x86_energy_perf_policy shows seem to be sort of a kernel "reverse translation" from the HWP.EPP settings. Why they don't match is the secret of the kernel devs or the CPU designers I guess.

Btw the kernel doc says that EPB will be ignored when HWP.EPP is employed. The relevant [commit](https://github.com/linrunner/TLP/commit/81a99c0b531101c93f7953aedcf966bedca4e2de) has some more details and doc links.

Short story: TLP's setting gets applied correctly (see above output quotes for proof). The discrepancy is caused by x86_energy_perf_policy (and the kernel).

Have a nice evening!

linrunner, 2020-05-13:
As you desire :-)

closed 2020-05-13 by linrunner

linrunner, 2020-05-13:
ps. the [kernel commit](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=4beec1d7519691b4b6c6b764e75b4e694a09c5f7) has the following details about HWP.EPP:

> 0 performance
> 128 balance_performance (default)
> **192 balance_power**
> 255 power

which nicely matches  x86_energy_perf_policy's output:

> cpu0: HWP_REQ: min 0 max 255 des 0 **epp 192** window 0x0 (0*10^0us) use_pkg 0