linrunner/TLP #491: Parameters AHCI_RUNTIME_PM_ON_BAT/AC don't work

rocky7x, 2020-05-14:
**Describe the bug**

The parameters AHCI_RUNTIME_PM_ON_BAT and AHCI_RUNTIME_PM_ON_AC don't work

**Expected behavior**

It is expected to set proper value into control files (on or auto)

**To Reproduce**

Set the parameters in /etc/tlp.conf to on or auto and start tlp
tlp-stat -T with proper DEBUG options returns following:
set_ahci_runtime_pm(0).auto: disk=/sys/block/sda timeout=15000; rc=1
set_ahci_runtime_pm(0).auto: disk=/sys/block/sdb timeout=15000; rc=1

And the values in control files are not changed

**Additional context**

The values for AHCI powermanagement in powertop remain in BAD state. When I set it to GOOD state via powertop, the values in control files are switched correctly. The control files are:
/sys/bus/pci/devices/0000:00:1f.2/ata*/power/control

linrunner, 2020-05-14:
Hi, 

the requirement in the bug template is sincerely meant. I _always_ need full outputs, not just the part you think is relevant:

    tlp-stat -c -s -d

rocky7x, 2020-05-15:
Hi,

Sorry for not providing all the info. I have attached the full outputs of both the command you wrote and of tlp-stat -T. In case anything else is needed, please let me know.
[stat1.txt](https://github.com/linrunner/TLP/files/4632578/stat1.txt)
[stat2.txt](https://github.com/linrunner/TLP/files/4632579/stat2.txt)

linrunner, 2020-05-15:
Both of your SATA disks lack the autosuspend_delay_ms sysfile:

>  Runtime PM: control = on, autosuspend_delay_ms = (not available)

As a safety measure, TLP won't activate Runtime PM for the disks and all AHCI hosts to prevent system lockup and data loss. Works as designed.

rocky7x, 2020-05-15:
OK, I understand. Just a noob question: is this something that I can set via some parameter or is it a HW thing, thus I can do nothing about it?

linrunner, 2020-05-15:
Don't mess with it. You may re-check with a newer kernel.

rocky7x, 2020-05-15:
OK, understood :) Thanks a lot for your time and effort.

closed 2020-05-15 by rocky7x

linrunner, 2020-05-15:
ps. better to attach long outputs via https://gist.github.com/ rather than as file attachments. Keeps them searchable.