linrunner/TLP #492: Governor and Clock Speed Changed Automatically

UniverseObserver, 2020-05-16:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs (see below)

**Describe the bug**

CPU performance and power saving config changes automatically, when using Chrome or Firefox to play a youtube video, launch gmail web client, or any operation that requires preformance. For all cores, `cpufreq/scaling_governor` changes from `powersave` to `performance` and `cpufreq/scaling_min_freq`, `cpufreq/scaling_max_freq` are set to 4.2Ghz, which is the maximum possible clock of my device. 
Unless `sudo tlp start` is executed again, the settings remains at 4.2Ghz and `performance`. 

**Expected behavior**

The governor and clock speed settings shall not be changed. 

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?
both. 

2. Actions to reproduce the behaviour
(see description)

3. Shell commands entered and their output
cpu clock got from `lscpu | grep "CPU MHz"`

4. **Full output of `tlp-stat` 
before the problem occure:
https://gist.github.com/UniverseObserver/6be7b44017a58baf4a62d3252b330e75
after the problem occure
https://gist.github.com/UniverseObserver/08e2745c81b30c42ce46ad26e27fa895

**Additional context**

My device is ThinkPad X1 carbon 2018 i7-8650U with Ubuntu 20.04 installed.

UniverseObserver, 2020-05-17:
Currently I wrote a script that runs `sudo tlp start` every 1 second, which seems to be a workaround. However, why are those settings changed? How can I prevent that from happening?

linrunner, 2020-05-17:
Hi,

you're barking up the wrong tree. TLP doesn't make any dynamic or adaptive changes to the settings. The cause must be something else.

closed 2020-06-07 by linrunner