linrunner/TLP #493: When on charger (AC mode) max frequency set to turbo boost freq

rocky7x, 2020-05-20:
**Describe the bug**

When I plug in the battery charger, the cpu frequency is set to maximum turbo (3,4 GHz) and stays there, even though the p-state governor is set to powersave on AC. Very soon the laptop becomes very hot. This doesn't happen when on battery, it scales down correctly. When I turn off the boost (turbo) for AC profile in tlp, the frequency goes up to 2,4 GHz, which is more acceptable, but it also never scales down to lower values. I think that the problem is that the scaling_max_freq is set to 3,4 GHz instead of 2,4 GHz.
[tlp-stat.txt](https://github.com/linrunner/TLP/files/4657113/tlp-stat.txt)

**Expected behavior**

Even when on charger, the cpu should scale down the frequency, when not under load.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Problem occurs on AC
2. Plug in the charger, tlp switches to AC mode

linrunner, 2020-05-20:
Hi, 

you have not configured TLP to directly affect the frequencies. Your settings only enable turbo boost on AC:

> /etc/tlp.conf L0071: CPU_BOOST_ON_AC="1"
> /sys/devices/system/cpu/intel_pstate/no_turbo          =   0

Thus the behaviour you observe, in particular the missing down scaling on AC, is something the kernel decides on it's own, maybe in collaboration with the ACPI BIOS.

I'd try

    CPU_ENERGY_PERF_POLICY_ON_AC="balance_power"

linrunner, 2020-05-20:
ps. please use https://gist.github.com/ for your outputs, not file attachments.

rocky7x, 2020-05-20:
Hi,

Yes I haven't indeed, I rely on the automatic downscaling, which when I'm on charger doesn't work at all. When I turn off boost, it keeps it at 2,4GHz at all the times. Even if I set power for CPU energy policy, it doesn't downscale. I thought that with the powersave governor it would downscale even on charger. Any idea?

linrunner, 2020-05-20:
What you just try when you run out of ideas: latest kernen, latest BIOS, ...

Good luck.

closed 2020-05-20 by rocky7x