linrunner/TLP #494: RFE: add a 'lower power adapter' mode.

marcmerlin, 2020-05-28:
Currently, there is a battery mode and AC mode.
On bigger laptops like thinkpad P73, the power adapter is as big as 230W (!)
(this is without counting broken design where the laptop actively tries to reject any power supply smaller than 170W even if it only needs 40W at the moment, details and hardware workaround here: http://marc.merlins.org/perso/electronics/post_2020-05-04_Hacking-Power-Supplies-and-Battery-Pack-To-Get-Around-ThinkPad-P73-Broken-Power-Supply-Design.html )

There is a need where you get external power, but you want to limit use over the charging plug:
- because you are using an external battery pack
- because you're plugged into an airplane with limited wattage
- similarly you're using a car 12V adapter and many cars limit that power to 100W
- or even you're using a 90W adapter with a laptop that could use 150W or more if it runs at full power and tries to charge the battery.

So, it would be great to have a power mode for 'limited AC draw' to handle any of the cases above. It would be hard to auto detect via udev, but you could just run the command from the command line after plugging in the power supply.

linrunner, 2020-05-28:
Hi,

thanks for your suggestion.

I have no plans to introduce a third power profile besides AC an BAT. 

It would require a third line for the majority of settings and in addition, would clutter up the documentation and require really extensive code changes. 

By contrast I estimate the number of potential users to be very small, but the impact would affect the whole user base.

Apart from that I have my doubts if an effective solution is possible solely based on kernel settings without supporting changes in Lenovo's firmware. If you look at how Lenovo has (finally) messed up the [Intel DPTF topic](https://forums.lenovo.com/t5/Other-Linux-Discussions/X1C6-T480s-low-cTDP-and-trip-temperature-in-Linux/m-p/4028489?page=38#5059290), there's little hope for that.

closed 2020-05-28 by linrunner

marcmerlin, 2020-05-28:
no problem, I'll just hack up something on my side, thanks for tlp

marcmerlin, 2020-05-28:
If anyone finds this bug and needs to do the same thing than me, here's what I did
1) 2 config files with a small diff to force battery mode even when on limited AC and mostly turn off battery charging
2) ln -snf correctconfig tlp.conf; tlp start

hope this helps someone
```
sauron:/etc# diff -u tlp.conf.sauron.lowpower tlp.conf.sauron.power
--- tlp.conf.sauron.lowpower	2020-05-28 12:50:29.131632625 -0700
+++ tlp.conf.sauron.power	2020-05-28 12:50:10.884698696 -0700
@@ -33,13 +33,13 @@
 # Concerns some desktop and embedded hardware only.
 # Default: <none>
 
-TLP_DEFAULT_MODE=BAT
+#TLP_DEFAULT_MODE=AC
 
 # Operation mode select: 0=depend on power source, 1=always use TLP_DEFAULT_MODE
 # Note: use in conjunction with TLP_DEFAULT_MODE=BAT for BAT settings on AC.
 # Default: 0
 
-TLP_PERSISTENT_DEFAULT=1
+#TLP_PERSISTENT_DEFAULT=0
 
 # Power supply class to ignore when determining operation mode: AC, BAT.
 # Note: try on laptops where operation mode AC/BAT is incorrectly detected.
@@ -438,7 +438,7 @@
 # Main / Internal battery (values in %)
 # Default: <none>
 
-START_CHARGE_THRESH_BAT0=15
+START_CHARGE_THRESH_BAT0=75
 STOP_CHARGE_THRESH_BAT0=80
 
 # Ultrabay / Slice / Replaceable battery (values in %)
```

linrunner, 2020-05-29:
A simpler and (imho) more robust approach (no config change necessary) would be to use manual mode:

    tlp bat

This will persist until the next reboot or

    tlp start

The charge thresholds may be changed with

    tlp setcharge 15 80

marcmerlin, 2020-05-29:
@linrunner  I thought about this, but there are 2 issues:
1) if the laptop is drawing just a bit too much, it will occasionally quick out the power supply and reconnect to it as it recovers, causing power plug/unplug events. This would revert to high power mode until I type 'tlp bat' every time
2) more of an issue, if the laptop is fully running and tries to charge batteries, it can use over 100W which will instantly reject the power supply that can't keep up, faster than I can type 'tlp bat'
For one, battery charging cannot be on for any amount of time, of that will cause up to an extra 100W of load as it's a big battery and a fast charger. but even without the battery charging, the laptop will use triple the amount of power if I don't throttle all the chips down.

So basically the 2 config file and symlink trick is the only thing I could get to work.

linrunner, 2020-05-29:
Which TLP version do you use? 

Manual mode will survive power unplug/plug events and even a suspend/resume cycle since TLP 1.2. 

The same applies to the thresholds unless you configured `RESTORE_THRESHOLDS_ON_BAT=1`.

marcmerlin, 2020-05-31:
I have 1.3.1-2, but I don't think I was clear, let me explain
1) I'm on batteries, laptop is happily running with 15W
2) I plug in a 90W power supply, tlp switches to power mode, spins up all the CPUs, puts the batteries to charging, laptop tries to draw 150W (yes, it does that), power supply fails, dips voltage, laptop goes back on batteries
3) and it bounces back and forth at 1Hz or more, quicker than I can type 'tlp bat' when on AC mode

the only way around this is to have another power mode that does not spin up everything and take so much power. This is why I made 2 config files with 2 different settings for AC.

linrunner, 2020-06-03:
I think you'd best try what I suggested, Step by step:

1. You're on battery, TLP is in auto mode, BAT profile is active
2. You select manual mode with

       sudo tlp bat

3. You plugin the 90W power supply
4. Because TLP is in manual mode, BAT profile sticks (until reboot or tlp start`)

https://linrunner.de/tlp/usage/tlp.html#battery-mode