linrunner/TLP #496: Mode not optimized on AC adapter

Ricky-Tigg, 2020-06-02:
Component: tlp.noarch  1.3.1-1.fc32 @fedora

OSses Windows and Linux Fedora are instaled into an ASUS-branded laptop. Use-case is keeping my AC adapter connected while using my laptop. 

- On windows' side
 [ASUS Battery Health Charging](https://www.asus.com/us/support/FAQ/1032726/) software is installed, active and operating as intended. Option applied, _Maximum Lifespan Mode_ here illustrated, acts as follows: _Stops charging when power is above 60 % and resumes charging when power is below 58 %_, thus **optimized**.

[image]

- On Linux's side 
TLP is installed and active. The battery is often in a state of high-power, 98.4375–100 % for extended length of time which causes a reduction in battery life, thus mode applied by default expected to be **optimized** as stated in [TLP documentation](https://linrunner.de/tlp/index.html), **is not**. 

[tlp-stat.txt](https://github.com/linrunner/TLP/files/4716407/tlp-stat.txt)

linrunner, 2020-06-02:
The statement in the documentation is about battery life in the sense of maximizing battery runtime, not about minimizing battery wear.

closed 2020-06-02 by linrunner

Ricky-Tigg, 2020-06-02:
Documentation may partially be subject to interpretation. Your  interpretation must to be correct, mine wrong then. Are those sentences an incorrect interpretations as well of your previous comment; _maximizing battery runtime_  **does apply** to Lenovo-Thinkpad models as well as others laptop models, while,_minimizing battery wear_ **does apply** only to Lenovo-Thinkpad models, thus **not to others laptop model**s,

linrunner, 2020-06-02:
The term on the frontpage is about optimizing runtime only, not about minimizing wear (not even for a specific brand).

Btw: https://dictionary.cambridge.org/dictionary/english/battery-life

Ricky-Tigg, 2020-06-03:
It is never easy when someone ends here for the wrong reason, to forgot what mothers said to their t school-aged children in your country or the one you moved to. _If you keep not knowing your language you will end as a developer. Is that what you want?_

That _man_ page, is that your writing too?
```
setcharge [START_CHARGE STOP_CHARGE] [BAT0|BAT1] (ThinkPads only)
Set charge thresholds of main (BAT0) or auxiliary (BAT1) battery temporarily.
```
You refered  in your own English as "wear"  what is there as "_setcharge_", meant then as temporal wear. You attempted to deny it. That's indeed beyond my understanding.

linrunner, 2020-06-03:
Your criticism lacks any constructive element and you obviously lack technical background which would be a prerequisite for understanding the documentation. Your provocative tone of asking is just as unhelpful. I'm out of here.