linrunner/TLP #498: amdgpu and mobility integrated graphics

userofryzen, 2020-06-04:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs (see below)

**Describe the bug**

change amd graphics settings in tlp for AC and battery doesn't work for amd integrated graphics
Mine laptop has ryzen 7 3700U

**Expected behavior**

change
/etc/tlp.d/graficos.conf L0043: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
/etc/tlp.d/graficos.conf L0044: RADEON_DPM_PERF_LEVEL_ON_BAT="low"

would change the output of:

tlp bat
TLP started in battery mode (manual).
 cat /sys/class/drm/card0/device/power_dpm_force_performance_level 
auto

**To Reproduce**

config the desire perf dpm level in tlp and see that is not working.

1. Does the problem occur on battery or AC or both? both
2. Actions to reproduce the behaviour is answered already
3. Shell commands entered and their output
tlp bat
TLP started in battery mode (manual).
 cat /sys/class/drm/card0/device/power_dpm_force_performance_level 
auto
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.**
https://gist.github.com/userofryzen/831354948c4d2c25bdef2ca1e79e64ad
https://gist.github.com/userofryzen/9a239ad3f39afc08e43e9a99424c4af6
**Additional context**

Add any other context about the problem here.

linrunner, 2020-06-05:
Hi, 

the RADEON_* settings [were designed](https://linrunner.de/tlp/settings/graphics.html#radeon-power-profile-on-ac-bat) for the *radeon* driver: 

> Supported by the radeon driver only

Your GPU uses *amdgpu* which surely uses different sysfiles for power saving (if at all).

As a first measure I suggest to remove *amdgpu* from the runtime pm blacklist

     RUNTIME_PM_DRIVER_BLACKLIST="mei_me nouveau nvidia pcieport radeon"

to enable runtime pm which is off atm:

> /sys/bus/pci/devices/0000:05:00.0/power/**control = on**   (0x030000, VGA compatible controller, **amdgpu**)

userofryzen, 2020-06-05:
the blacklist is the default ones  I think. Why is it blacklisted ?
Shlould I unblacklist a integrated amd graphic card from PM?
IF yes  I will do  it but I though that PM management can for example power on the graphics by error. Can you tell me some scenario to tests the power usage and this option?

Are there any plans to support the amdgpu ones?

userofryzen, 2020-06-05:
and it gives me another question, the mei_me and pcieport.. are blacklisted too. What's your recommend about it?

linrunner, 2020-06-05:
Sorry, but you're asking way to many questions (no offense meant). This issue is about amdgpu, please keep focus.

I don't know what happens on _your_ hardware. Hardware support in TLP does not just appear out of nowhere, but depends on people (owning that hardware) _contributing_ details, experiences or pull requests (preferred).

I suggest you just try to enable runtime pm for amdgpu by removing it from the blacklist as shown above and see what happens.

Report back power consumption describing _your_ scenarios. Use powertop for measurement.

closed 2020-08-13 by linrunner

userofryzen, 2020-08-13:
Perfect. Thanks. I have been tested the pm on amdgpu and it works well so no problem here either.

linrunner, 2020-08-19:
I did merged amdgpu and radeon code. Could you please test the new code?

Note that configuration for amdgpu is now done with RADEON_DPM_PERF_LEVEL_ON_AC/BAT too.

linrunner, 2020-08-19:
ps. please show the output of

    tlp-stat -s -g

userofryzen, 2020-08-20:
Leave me a few weeks to get the laptop again because I am without access to it right now. At this way I can wait until manjaro updates tlp package on its repositories

userofryzen, 2020-08-31:
Tried and all was working as expected I think.

linrunner, 2020-08-31:
Thanks. Could you show the output of 

    tlp-stat -s -c -g

please?

userofryzen, 2020-10-22:
I have just seen it now. So I can not show the output now I think, there is no need, no?

linrunner, 2020-10-23:
Your output is still welcome.

nicman23, 2020-11-17:
```
--- TLP 1.4.0-alpha.0 --------------------------------------------

+++ System Info
System         = Hewlett-Packard A1029D1102 HP EliteBook 8570w
BIOS           = 68IAV Ver. F.64
OS Release     = Arch Linux
Kernel         = 5.9.8-arch1-1 #1 SMP PREEMPT Tue, 10 Nov 2020 22:44:11 +0000 x86_64
/proc/cmdline  = initrd=/initramfs-linux.img root=UUID=c82afd21-4dd4-4857-9b38-017f41c0ff18 rootflags=subvol=root rw loglevel=3 quiet rd.systemd.show_status=false udev.log_priority=3 mitigations=off radeon.si_support=0 amdgpu.si_support=1 radeon.cik_support=0 amdgpu.cik_support=1 sysrq_always_enabled=1 amdgpu.ppfeaturemask=0xffffffff
Init system    = systemd 
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = not installed
Last run       = 03:37:37 PM,     11 sec(s) ago
Mode           = battery
Power source   = battery

Warning: systemd-rfkill.service is not masked -- invoke "systemctl mask systemd-rfkill.service" to correct this!
Warning: systemd-rfkill.socket is not masked -- invoke "systemctl mask systemd-rfkill.socket" to correct this!

+++ AMD Radeon Graphics
/sys/class/drm/card0/device/driver           = amdgpu
/sys/class/drm/card0/device/power_dpm_force_performance_level     = low

```

working as intended - ignore the rfkill  thing

linrunner, 2020-11-17:
Thanks. The column formatting needs some care i guess.

linrunner, 2020-11-17:
Formatting done.

PR #520 (2020-08-13): add support for amdgpu
Fixes #498

Tested on Lenovo Thinkpad T14 Gen 1 AMD version