linrunner/TLP #499: RUNTIME_PM_BLACKLIST doesn't works on ThunderBolt3 dock

plumlis, 2020-06-09:
**Describe the bug**

My usb soundcard have some poping noise when using on thunderbolt3 dock.
And I found there is a device called 00:16.0 is set to battery saved.
I changed it to "bad" in Powertop so problem sloved

**Expected behavior**

I add this devices( 00:16.0 ) to RUNTIME_PM_BLACKLIST and it seems doesn't work on Thunderblot3 dock device.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. add a thunderbolt3 dock connected device to RUNTIME_PM_BLACKLIST
2. connect laptop to thunderbolt3 dock 
3. check powetop
4. found this device still in battery save mode

here is my tlp-stat

--- TLP 1.3.1 --------------------------------------------

+++ Configured Settings:
/etc/tlp.conf L0026: TLP_ENABLE="1"
defaults.conf L0005: TLP_PERSISTENT_DEFAULT="0"
defaults.conf L0006: DISK_IDLE_SECS_ON_AC="0"
defaults.conf L0007: DISK_IDLE_SECS_ON_BAT="2"
defaults.conf L0008: MAX_LOST_WORK_SECS_ON_AC="15"
defaults.conf L0009: MAX_LOST_WORK_SECS_ON_BAT="60"
/etc/tlp.conf L0100: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
/etc/tlp.conf L0101: CPU_ENERGY_PERF_POLICY_ON_BAT="default"
/etc/tlp.conf L0128: SCHED_POWERSAVE_ON_AC="0"
/etc/tlp.conf L0129: SCHED_POWERSAVE_ON_BAT="1"
defaults.conf L0014: NMI_WATCHDOG="0"
defaults.conf L0015: DISK_DEVICES="nvme0n1 sda"
defaults.conf L0016: DISK_APM_LEVEL_ON_AC="254 254"
defaults.conf L0017: DISK_APM_LEVEL_ON_BAT="128 128"
/etc/tlp.conf L0188: DISK_IOSCHED="bfq bfq"
/etc/tlp.conf L0198: SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
/etc/tlp.conf L0199: SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
/etc/tlp.conf L0219: AHCI_RUNTIME_PM_TIMEOUT="15"
/etc/tlp.conf L0226: PCIE_ASPM_ON_AC="performance"
/etc/tlp.conf L0227: PCIE_ASPM_ON_BAT="powersave"
defaults.conf L0024: RADEON_POWER_PROFILE_ON_AC="default"
defaults.conf L0025: RADEON_POWER_PROFILE_ON_BAT="default"
defaults.conf L0026: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
defaults.conf L0027: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
/etc/tlp.conf L0264: WIFI_PWR_ON_AC="off"
/etc/tlp.conf L0265: WIFI_PWR_ON_BAT="on"
/etc/tlp.conf L0270: WOL_DISABLE="Y"
/etc/tlp.conf L0276: SOUND_POWER_SAVE_ON_AC="1"
/etc/tlp.conf L0277: SOUND_POWER_SAVE_ON_BAT="1"
/etc/tlp.conf L0283: SOUND_POWER_SAVE_CONTROLLER="Y"
defaults.conf L0034: BAY_POWEROFF_ON_AC="0"
defaults.conf L0035: BAY_POWEROFF_ON_BAT="0"
defaults.conf L0036: BAY_DEVICE="sr0"
/etc/tlp.conf L0302: RUNTIME_PM_ON_AC="on"
/etc/tlp.conf L0303: RUNTIME_PM_ON_BAT="auto"
defaults.conf L0039: RUNTIME_PM_DRIVER_BLACKLIST="amdgpu mei_me nouveau nvidia pcieport radeon"
/etc/tlp.conf L0323: USB_AUTOSUSPEND="1"
defaults.conf L0041: USB_BLACKLIST_BTUSB="0"
defaults.conf L0042: USB_BLACKLIST_PHONE="0"
defaults.conf L0043: USB_BLACKLIST_PRINTER="1"
defaults.conf L0044: USB_BLACKLIST_WWAN="0"
defaults.conf L0045: USB_AUTOSUSPEND_DISABLE_ON_SHUTDOWN="0"
defaults.conf L0046: RESTORE_DEVICE_STATE_ON_STARTUP="0"
/etc/tlp.conf L0437: RESTORE_THRESHOLDS_ON_BAT="1"
/etc/tlp.conf L0442: NATACPI_ENABLE="1"
/etc/tlp.conf L0443: TPACPI_ENABLE="1"
[... 198 lines cut ...]
Bus 001 Device 018 ID 1199:9079 control = auto, autosuspend_delay_ms = 2000 -- Sierra Wireless, Inc. Sierra Wireless EM7455 Qualcomm Snapdragon X7 LTE-A (cdc_mbim)
Bus 001 Device 002 ID 04f2:b5cf control = auto, autosuspend_delay_ms = 2000 -- Chicony Electronics Co., Ltd Integrated IR Camera (uvcvideo)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 2.0 root hub (hub)

+++ Battery Features: Charge Thresholds and Recalibrate
natacpi    = active (data, thresholds)
tpacpi-bat = active (recalibrate)
tp-smapi   = inactive (ThinkPad not supported)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SMP
/sys/class/power_supply/BAT0/model_name                     = 01AV430
/sys/class/power_supply/BAT0/cycle_count                    =    235
/sys/class/power_supply/BAT0/energy_full_design             =  57020 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  54170 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  53080 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Unknown (threshold may prevent charging)

/sys/class/power_supply/BAT0/charge_start_threshold         =     96 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold          =    100 [%]
tpacpi-bat.BAT0.forceDischarge                              =      0

Charge                                                      =   98.0 [%]
Capacity                                                    =   95.0 [%]

linrunner, 2020-06-09:
The actual expected result when blacklisting a device isn't that the device (magically) works, but rather that power saving gets disabled, which is obviously the case (control=on means disabled, control=auto means enabled):

> /sys/bus/pci/devices/0000:00:16.0/power/**control = on** (0x078000, Communication controller, mei_me)

However, the whole blacklisting is redundant, because TLP will set control=on for all PCIe devices on AC power anyway, it's the default config:

> /etc/tlp.conf L0302: RUNTIME_PM_ON_AC="on"

What i do not understand at all is what the intel management engine (driver mei_me) has to do with the problem. I suggest you try to blacklist your USB sound card(s) instead:

> Bus 005 Device 003 ID 0b05:183c control = on, autosuspend_delay_ms = 2000 -- ASUSTek Computer, Inc. Xonar U7 MKII (snd-usb-audio, usbhid)
> Bus 003 Device 002 ID 17ef:306a control = on, autosuspend_delay_ms = 2000 -- Lenovo ThinkPad Thunderbolt 3 Dock USB Audio (snd-usb-audio, usbhid)

**Please be so kind and post long outputs via https://gist.github.com/ (as requested in the bug report template). Thanks for your understanding**

linrunner, 2020-06-12:
OK, the difference is as expected after setting the device to "bad" manually:

> echo 'auto' > '/sys/bus/pci/devices/0000:00:16.0/power/control';

The question is why the blacklisting fails for both driver and device.

Please provide a trace for boot an AC power – via https://gist.github.com/ please.

https://linrunner.de/tlp/support/index.html#trace-mode

plumlis, 2020-06-13:
Thanks for your reply
Here is my logfile

https://gist.github.com/plumlis/8c4fb5197e38607542b8f5eb7e4e549e

linrunner, 2020-06-14:
So for _your_ test case blacklisting works as designed. The device address based rule acts and no change is applied:

AC > BAT
> 21:11:38 plum-laptop tlp[13593]: set_runtime_pm(1).**bloack_address**: /sys/bus/pci/devices/0000:00:16.0 [0x078000]

BAT > AC
> 13 21:11:57 plum-laptop tlp[14175]: set_runtime_pm(0).**bloack_address**: /sys/bus/pci/devices/0000:00:16.0 [0x078000]

Compare with the other log entries with 'rc=0', where TLP actually writes to control (successful).

linrunner, 2020-06-14:
But i asked you for the trace of a different test case i.e. booting on AC power. Can you provide that trace too?

plumlis, 2020-06-14:
Thank you for your reply.

Here are more logfiles

Boot on AC Power with Thunderblot3 Dock connected ( Blacklist on)

https://gist.github.com/plumlis/8e25a20777cfa02c467d76d68f03158f

Boot on AC Power with Thunderblot3 Dock connected ( Remove Blacklist )
https://gist.github.com/plumlis/9a814c44f95ccbcd3ce9949b0dc90e9f

Both of them don't change the status of device 0000:00:16.0, Still "Good" in Powertop.

linrunner, 2020-06-14:
Both of your boot test cases hit a blacklist:

1. [Blacklist on] Device in RUNTIME_PM_DRIVER
> 14 21:26:35 plum-laptop tlp[1871]: set_runtime_pm(0).**bloack_address**: /sys/bus/pci/devices/0000:00:16.0 [0x078000]

2. [Remove Blacklist] But device driver _mei_me_ still in RUNTIME_PM_DRIVER_BLACKLIST
> 14 21:34:50 plum-laptop tlp[1366]: set_runtime_pm(0).**black_driver**: /sys/bus/pci/devices/0000:00:16.0 [0x078000]

Conclusion: _auto_ is in fact the kernel driver default and blacklisting just preserves it (works as designed).

Next step: remove _mei_me_ from  RUNTIME_PM_DRIVER_BLACKLIST, reboot and provide trace.

plumlis, 2020-06-15:
Actually I haven't enable RUNTIME_PM_DRIVER_BLACKLIST before
So I made these log

With default RUNTIME_PM_DRIVER_BLACKLIST enabled ,"mei_me" included.
https://gist.github.com/plumlis/570bc3a167cc5edcd3a4bf7f6d04a7ec

RUNTIME_PM_DRIVER_BLACKLIST enabled, without "mei_me"
https://gist.github.com/plumlis/bcc91b06910d7ecea7e6e68c2c1c34f2

Finally, With 

RUNTIME_PM_DRIVER_BLACKLIST="amdgpu nouveau nvidia pcieport radeon"

and devices/0000:00:16.0 is set to "Bad" in Powertop.

linrunner, 2020-06-15:
Very well. If you consider this solved now, feel free to close.

> Actually I haven't enable RUNTIME_PM_DRIVER_BLACKLIST before

Of course, but it's enabled by default and _mei_me_ was added as a workaround for shutdown hangs. The line must be uncommented and changed to deviate from the intrinsic default. I know it's a bit unintuitive ...

plumlis, 2020-06-16:
Thank you for your help.
Issue sloved.

closed 2020-06-16 by plumlis