linrunner/TLP #500: tlp recalibrate: /usr/sbin/tlp: 2666: printf: %]: invalid directive

linewrap, 2020-06-11:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs (see below)

ThinkPad X220 Tablet running ubuntu 18.04LTE+KDE/Plasma on kernel 4-15-0-106-generic.
TLP is confused whether battery is discharging or not. #tlp-stat -b reports battery as Discharging, and so does KDE/Plasma.

Typically, tlp displays capacity/percent left/time left

Steps to reproduce the unexpected behavior:
No particular steps -- I can't get it to work despite reboots or using different battery.

1. Does the problem occur on battery or AC or both? 
AC + recalibrate mode

2. Actions to reproduce the behaviour
run # tlp recalibrate

3. Shell commands entered and their output
# tlp recalibrate

Currently discharging battery BAT0:
voltage            =  11565 [mV]
remaining capacity =      0 [mWh]
/usr/sbin/tlp: 2666: printf: %]: invalid directive
remaining percent  = not available [remaining time     = not discharging [min]
state              = Discharging
force discharge    = 1
Press Ctrl+C to cancel.

4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.**

--- TLP 1.1 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
TLP_DEFAULT_MODE=AC
TLP_PERSISTENT_DEFAULT=0
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
CPU_HWP_ON_AC=balance_performance
CPU_HWP_ON_BAT=balance_power
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=power
DISK_DEVICES="sda sdb"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
AHCI_RUNTIME_PM_TIMEOUT=15
PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave
RADEON_POWER_PROFILE_ON_AC=high
RADEON_POWER_PROFILE_ON_BAT=low
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=off
WIFI_PWR_ON_BAT=on
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_AC=0
[... 173 lines cut ...]
Bus 002 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)
Bus 001 Device 003 ID 04f2:b217 control = auto, autosuspend_delay_ms =  2000 -- Chicony Electronics Co., Ltd Lenovo Integrated Camera (0.3MP) (uvcvideo)
Bus 001 Device 002 ID 8087:0024 control = auto, autosuspend_delay_ms =     0 -- Intel Corp. Integrated Rate Matching Hub (hub)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =     0 -- Linux Foundation 2.0 root hub (hub)

+++ ThinkPad Battery Features
tp-smapi   = inactive (kernel module 'tp_smapi' not installed)
tpacpi-bat = active

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SANYO
/sys/class/power_supply/BAT0/model_name                     = 45N1177
/sys/class/power_supply/BAT0/cycle_count                    = (not supported)
/sys/class/power_supply/BAT0/charge_full_design             =   5800 [mAh]
/sys/class/power_supply/BAT0/charge_full                    =   5670 [mAh]
/sys/class/power_supply/BAT0/charge_now                     =   4398 [mAh]
/sys/class/power_supply/BAT0/current_now                    =    827 [mA]
/sys/class/power_supply/BAT0/status                         = Discharging

tpacpi-bat.BAT0.startThreshold                              =     96 [%]
tpacpi-bat.BAT0.stopThreshold                               =    100 [%]
tpacpi-bat.BAT0.forceDischarge                              =      1

Charge                                                      =   77.6 [%]
Capacity                                                    =   97.8 [%]

linrunner, 2020-06-11:
Please install the package **tp-smapi-dkms** and try again.

linewrap, 2020-06-11:
Installation of tp-smapi-dkms worked, thank you. I was under the impression that tp-smapi only worked on older thinkpads, not the X220.

linrunner, 2020-06-11:
I'm a bit irritated because i can't reproduce the problem on my X220. 

Could you do me a favour?

1. Install version 1.3.1 from the PPA: https://linrunner.de/tlp/installation/ubuntu.html
2. Uninstall **tp-smapi-dkms**
3. Reboot
4. Retry `tlp recalibrate`

linewrap, 2020-06-11:
OK so I removed the module but first restarted with TLP 1.1 -- it DID report with no errors.
I added your PPA with 1.3.1 --> rebooted -- it DID report with no errors.
Then I closed the lid of the laptop and logged back in. The error returned and now it does NOT report (1.3.1). Again, this is X220 Tablet, your acpi call may be different.

linewrap, 2020-06-11:
OK, I ran an apt update. apt was previously reporting (my impression) that I was using version 1.3.1 but I ran the update anyway and now I'm certain it's 1.3.1
I rebooted, logged back in, TLP worked, no errors. 
Rebooted again closed lid, logged back in, all normal, TLP 1.3.1 works, reports percentages.
Bottom line: 1.3.1 appears to be working WITHOUT tp-smapi-dkms installed.

linrunner, 2020-06-12:
>  Again, this is X220 Tablet, your acpi call may be different.

Nope. The ACPI calls are identical for all models, even the kernel implementation (natacpi) uses the same. Besides, the status output while discharging reads just the sysfs files shown there; only enabling force_discharge needs ACPI calls (via tpacpi-bat in your case).

> The error returned and now it _does NOT report_ (1.3.1).

I confess I don't get the meaning of that. 

But anyway I'm fine with your bottom line.

Bottom line: there's no point to fix this for 1.1.

closed 2020-06-12 by linrunner

linewrap, 2020-06-12:
Thanks for looking at this.
My apologies for not being entirely clear, I'm not a native EN speaker and
definitely a computer layman.

FYI: one more thing I wanted to add: I had to revert to TLP 1.1 (the
packaged version on my 18.04) because the time when I had it installed
overlaps with the time when my X220T stylus stopped working. Stylus didn't
work after reboots.
It worked the day before I installed TLP 1.3.1 and only started working
again upon uninstalling TLP 1.3.1 (and its dependencies, presumably). It
probably matters to very few people.
Thanks again.

linrunner, 2020-06-14:
> I'm not a native EN speaker

Neither am I ;-)

> It worked the day before I installed TLP 1.3.1 and only started working again upon uninstalling TLP 1.3.1 (and its dependencies, presumably).

Remaining at 1.1 is not a sustainable strategy. If I remember correctly the X220T's Wacom Multitouch is a USB device.

I suggest you try to isolate the cause with the help of the [Troubleshooting Guide](https://linrunner.de/tlp/support/troubleshooting.html). Feel free to open another issue for this.