linrunner/TLP #501: Power nap mode

dmirubtsov, 2020-06-12:
I'm return back from macOS to Linux laptop, and I have one feature which I missed. I want to implement mode that called on macOS as [Power Nap](https://support.apple.com/en-us/HT204032). 

Because power management on my laptop managaed by TLP, I think that this app should have this feature. In general, TLP should have one more mode -- is lid opened/closed.

In my case, I want this scenario:
When lid closed and AC is plugged, laptop should be in aggressive powersave mode, with minimal CPU frequency and disabled all of components as possible. In this mode it can do background tasks, like -- backup by network, Mail/IM notifications, updates, etc.

What do you think about this?

linrunner, 2020-06-14:
Hi, interesting idea.

if you find out the udev event for closing or opening the lid, i will think about it.

dmirubtsov, 2020-06-14:
Hi @linrunner, thanks for response

I've researched a bit, this event managed by `acpi`:

```
~ cat /proc/acpi/button/lid/LID0/state
state:      open
```

linrunner, 2020-06-14:
ACPI event is unfortunate because the whole /etc/acpi/event stuff is seriously deprecated imho. No udev event?

dmirubtsov, 2020-06-14:
@linrunner no,

systemd uses acpi for `HandleLidSwitch`, `HandlePowerKey` and other events like these, so I think that it's acceptable

linrunner, 2020-06-14:
That is exactly where the problem lies: systemd (logind) consumes those events but doesn't expose an interface to hook them.

konradmb, 2020-07-27:
There is a D-Bus property for lid in logind:
```
busctl get-property org.freedesktop.login1 /org/freedesktop/login1 org.freedesktop.login1.Manager LidClosed
```

linrunner, 2025-07-17:
@linrunner this could possibly be mapped via the planned [TLP-PD daemon](https://github.com/tbodt/TLP-PD)

closed 2025-10-31 by linrunner