linrunner/TLP #503: issue with commenting after a value

thiswillbeyourgithub, 2020-07-02:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs (see below)

**Describe the bug**

Apparently, commenting in the middle of a line can result in the line being ignored.

Example : `CPU_MAX_PERF_ON_AC=100 # some remarks by me`

This line was not working while the remark was in, the line was just ignored and tlp defaulted to using the BAT max perf value.

**Expected behavior**

Comments should be ignored even when inline, or if ignored : they should be mentionned in `sudo service tlp status` and `journalctl` etc IMO.

**To Reproduce**

1. turn `CPU_MAX_PERF_ON_AC=100` into : 
`CPU_MAX_PERF_ON_AC=10`
2. `sudo service tlp restart`
3. now turn the same line into `CPU_MAX_PERF_ON_AC=100 #whatever`
4. `sudo service tlp restart`
5. `sudo tlp-stat -p` => shows that the line was not taken into account

Current version : 1.3.0

linrunner, 2020-07-05:
That's correct, the parser won't accept comments after parameters and will ignore those lines. Works as designed.

The [documentation](https://linrunner.de/tlp/settings/introduction.html#config-files) says: 

> Files consist of PARAMETER=”value” entries, comments characterized by # in the 1st column and empty lines

thiswillbeyourgithub, 2020-07-05:
Thanks.

But if lines are ignored, should it really be silent? I would have expected to see it in `journalctl -u tlp`

closed 2022-01-19 by linrunner