linrunner/TLP #505: Automatically change mode to auto on power cord connection

claike, 2020-07-04:
**Is your feature request related to a problem? Please describe.**

Sometimes when my laptop is working off battery I need to set it to AC mode in order to get higher performance, but occasionally I forget to switch it back to auto, and because of that the system is running in full performance mode the whole time draing battery very fast.

**Describe the solution you'd like**

I'd like to have a config option that will allow tlp to automatically switch to auto mode whenever the power cable is connected no matter if it was in manual mode

**Describe alternatives you've considered**

None, couldn't find anything like this in tlp config.

**Additional context**

OS: Debian Testing; TLP version: 1.3.1

linrunner, 2020-07-05:
Hi,

I can understand your use case very well, on the other hand "manual mode" implies that it does not change automatically. 

I let this open for comments.

phrxmd, 2020-11-06:
Maybe there could be a `--temporary` flag to the `tlp ac` and `tlp bat` commands, to indicate that manual mode is temporary and that upon the next power source change it should reenter auto mode.

mightyllama89, 2021-08-20:
I would suggest maybe a new "Performance" power mode. This mode could be used when you want everything cranked to 11. Even on AC I don't push everything to max because of the unnecessary heat, noise, contribution to rising sea levels, etc.. This mode could be started by something like `tlp perf' and only be active until the next power state change. I can discuss further if you are interested.

LinuxOnTheDesktop, 2022-03-05:
+1 for the `--temporary` flag idea.

hjdfdr44, 2022-06-01:
The driver isn't always yet inside the kernel, there are a few complex points that want to be remodeled, however, I don't know when I'll have time to try this to get a [laptop for software engineering](https://laptopsadviser.com/best-laptop-for-software-engineering-students/).

linrunner, 2025-07-17:
The introduction of power profiles `tlp performance|balanced|power-saver` will also cover this requirement: https://github.com/linrunner/TLP/issues/720#issuecomment-2815324866 -> I'll close here.

closed 2025-07-17 by linrunner