linrunner/TLP #506: offer a way to revert all settings to system default

nsklaus, 2020-07-13:
**Is your feature request related to a problem? Please describe.**
i installed tlp and tweaked many settings, now i have some problems on my system and i'm unsure what settings i changed is causing it. 

**Describe the solution you'd like**

tlp should save system default settings (as in: before tlp install) and offer a way to restore system default settings. because right now even if i disable tlp.service, all the modifications i made to system settings are still effective even if tlp itself is disabled. it would be good for when diagnosing problems, to be able to put tlp and all its settings tweaks out of the picture.
for example right now i'm having bluetooth connectivity problem and i suspect some of tlp  usbautosuspend settings to be the cause. i have to toggle each one of them, one by one, test and reboot also, and it is not very convenient that way. if i could tell tlp to temporarily restore system's default settings, and disable tlp completly that would make the diagnose steps easier.

**Additional context**
in this particular case, bluetooth works flawlessly from distro's livecd, but on my installed system (with tlp), bluetooth works 1 out of 10 reboots. module gets inserted fine, journalctl doesn't show much bluetooth problems at boot time, but my system reports: no adapter present in 1 ou of 10 boots/sessions. i suspect some of the tweaks i made through tlp might be the cause, like usb-auto-suspend, but it's a pain to have to test each one, reboot, test more, reboot ..etc ..
if tlp could make itself act like it is not installed (restoring system default settings) that would be nice.

linrunner, 2020-07-15:
I can understand that is time consuming and upsets you. 

The feature would mean to save every original kernel setting. The current design does not allow for this – and it would be a _real_ pain to implement.

Concerning your case: USB_ settings may all be reverted via config file and `tlp start` (i.e. without reboot). Runtime PM as well. If in doubt compare `tlp-stat -u` output

Btw: were did you get that nonsense about disabling tlp.service?  There is only one correct way: https://linrunner.de/tlp/faq/operation.html#how-do-i-stop-or-disable-tlp-completely

nsklaus, 2020-07-15:
thank you for your reply but you misunderstood most of the point i was making.

first, this not just about me and my personal needs. i was talking in a general sense for every tlp users out there not just me. 
then, the basic principle is: when your software do something, it should be able to also undo. think about it, if package managers only supported installing and not removing, for example. that would not be correct. one could argue that tlp users could go through all the options and manualy undo everything changes made through tlp but, well, still with my analogy to package managers that would mean every users would have to manualy hunt down all the installed files and remove them. it is possible yes, but not very practical or convenient.

then about your final question, i was just talking about stopping and disabling tlp as a service. which is highlighted in the doc you pointed me to here:
```
systemctl stop tlp.service
systemctl disable/mask tlp.service
```
that is what i was refering to. and my point was: even when you do that, all the changes in settings made through tlp are still effective, even when tlp as a service is down and disabled. for example i capped my cpu and gpu frequency to minimum in order to avoid unwanted heat. so when tlp service is down and disabled, my cpu and gpu still work at the values i have defined in tlp. so disabling tlp doesn't make the system to behave as it would normaly if tlp was never installed. which link back to my general point: that tlp forces the user to hunt down manualy all changes one by one, instead of giving him a simple way to quick toggle off all the change made through tlp. which could be useful in many cases. one of the case i presented as an example is when diagnosing problems on the system.

nsklaus, 2020-07-15:
and as for a suggestion to implement this:
at tlp install, gather all the vanilla values for each settings tlp is able to modify and keep a key/value array of that somewhere.
then if at some point user wishes to restore system to vanilla state, it would just be a matter of itterating through that array to apply all the values that were saved at install time. this would work without touching at tlp code at all. two scripts would be needed:
one for gathering vanilla values. (it would run once, at tlp install time)
and another one to restore them upon user request. (could be run every time a user request it)

linrunner, 2020-07-16:
> first, this not just about me and my personal needs. i was talking in a general sense for every tlp users out there not just me.

To be exact: this not about _every_ TLP user, but at most those observing issues with power saving and want to track them down. You propose to offer a way more comfortable approach to isolate the causing setting. 

> if package managers only supported installing and not removing,

I find the comparison inappropriate. The absence of an uninstall would be unacceptable for a package manager, while your feature is just "nice to have".

> there is no need to save "every" kernel settings, but rather just the ones tlp is able to tweak.

That alone would require a lot of effort.

> i was just talking about stopping and disabling tlp as a service. which is highlighted in the doc you pointed me to here:

Maybe you overlooked the

> _Wrong ways:_

above it?

nsklaus, 2020-07-16:
> I find the comparison inappropriate. The absence of an uninstall would be unacceptable for a package manager, while your feature is just "nice to have".

it's the same. you're just trying to find reasons to justify the lack of something basic that's missing. "do/undo" it's the most simple concept. what tlp do, is destructive modifications without a way to revert back, unless going through manualy one by one, having users to memorize what was the default values and try to replicate their way back to vanilla from memory. it's nice that tlp can help minimize power usage on laptop, but it's not nice that it cannot put everything back like it was. when you rent a flat, you are supposed to put it back like it was when you rented it. it's simple. same for package managers, same for everything.
tlp does half the job. next time i'll have to use it, i'll remember to note down all vanilla values myself maybe make a script to do what tlp should be able to do but doesn't.
it should be written as a warning before using this software: "the burden of storing vanilla values is on you. once you modify something there's no wayback to default provided".

linrunner, 2020-07-16:
"Kernel settings" are volatile. Their state is held in RAM during the runtime of the kernel. There is no persistence or storage possibility. Upon boot the kernel creates a "default state" and changes have to be applied by user space software. 

Every power management tool known to me works like this. None of them has a feature like you requested.

nsklaus, 2020-07-16:
that is too bad. i don't understand how people can make a switch to flip the light on and never suspected people would want to turn it off at some point. "do/undo" is a basic principle.
being able to restore something to vanilla state after tweaking seems obvious.
imagine someone messing up the settings and wanting to restart from scratch for example.

you were saying:
`Upon boot the kernel creates a "default state" and changes have to be applied by user space`
then why not grabbing these default when tlp starts, before actualy applying the changes, first grab and store defaults ?

and also, you were implying this would be tricky / a non-trivial thing to implement,
but, is it not just a matter of echo'ing a few variables and store them in a file somewhere ?
there's more or less 50 independent settings being tweaked by tlp.
so echo 50 times some variables, pipe output to a file, that should be barely measurable in human sense of time being spent. a few microseconds.
and once the file that would hold the default values is generated, then just test if file exists then if yes: skip the grabbing part jump straight to apply tweaks. so as long as that file would exists the "kernel defaults grabbing event" would not take place again. basicaly saying it would be done once. that wouldn't be something that would slow the tlp service up and down process.

closed 2020-10-25 by linrunner