linrunner/TLP #507: Thinkpad T480s Wont Charge Unless Fully Discharged Via "tlp recalibrate'

ratboy777, 2020-07-14:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs (see below)

**Describe the bug**

A clear and concise description of what the bug is.

**Expected behavior**

A clear and concise description of what you expected to happen.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?
2. Actions to reproduce the behaviour
3. Shell commands entered and their output
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.**

**Additional context**

Add any other context about the problem here.

ratboy777, 2020-07-14:
Doing some further investigating it now seems for whatever reason TLP will start charging when my laptop is at 93% or lower despite the threshold settings. fullcharge and setcharge still dont appear to be working.

linrunner, 2020-07-15:
Hi, 

with your current settings, i.e. no thresholds configured, TLP has no effect at all on charging. I'm afraid you'll have to look for the cause somewhere else.

ratboy777, 2020-07-15:
I suppose based on the config it looks like I had been using defaults the whole time, but I have tried to change them in the config file. I later set everything back to default since writing to the config file and rebooting didn't change the shown threshold values in tlp-stat -b. Also changing them with setcharge didn't work either. The only way to change them was to echo directly to /sys/class/power_supply/BAT0/charge_start_threshold. Anyway perhaps another program is interfering or there is an incompatibility in the 5.7.0 kernel with natacpi or something. Thanks for the fast response and for the amazing TLP program.

linrunner, 2020-07-15:
I don't know what you tried in the config file, but maybe you didn't uncomment the settings. "Didn't work" is not a useful description. 

**Nevertheless, your output clearly shows that currently thresholds are not set:**

> /sys/class/power_supply/BAT0/charge_start_threshold = 96 [%]
> /sys/class/power_supply/BAT0/charge_stop_threshold = 100 [%]

96/100 are the factory defaults.

If thresholds refuse to change or have no (or wrong) effect, then the EC firmware (ECP) is your prime suspect. Normally I would have suggested to update it, but your BIOS is already up to date (1.42) and should be accompanied by ECP 1.21.

- https://pcsupport.lenovo.com/us/en/products/laptops-and-netbooks/thinkpad-t-series-laptops/thinkpad-t480s-type-20l7-20l8/downloads/ds502226
- https://download.lenovo.com/pccbbs/mobiles/n22ul24w.txt

A somewhat similar case is [described the FAQ](https://linrunner.de/tlp/faq/battery.html#disabling-the-charge-thresholds-does-not-work)

Good luck.

ratboy777, 2020-07-15:
I did uncomment and change those values, but since I was still having problems I restored the original config file. When I would change the values in the config nothing would change in /sys/class/power_supply/BAT0/charge_stop_threshold or /sys/class/power_supply/BAT0/charge_start_threshold. Just to clarify. Following that guide in the FAQ is probably what got it to start charging again since that was one of the things I tried. I also upgraded the BIOS when troubleshooting this problem initially (from 1.3.4 to current iirc) but that seemed to have no effect. Anyway thank you again for helping to troubleshoot. I just thought you might be interested in knowing about this problem since it seems the firmware on the T480s has not played nicely with TLP in the past based on some reports I've seen.

closed 2020-07-15 by ratboy777