linrunner/TLP #508: ppa:linrunner/tlp missing Ubunto 20.04 package  => battery thresholds problem

atagunov, 2020-07-18:
https://launchpad.net/~linrunner/+archive/ubuntu/tlp/+packages
shows 1.3.1-1~bionic (Ubuntu 18.04 LTS) but nothing for focal (Ubuntu 20.04 LTS)

following https://linrunner.de/tlp/installation/ubuntu.html on Ubuntu 18.04 results in installation from PPA
following https://linrunner.de/tlp/installation/ubuntu.html on Ubuntu 20.04 results in installation from official Ubuntu repo

/etc/tlp.conf ends up quite different between Ubuntu 18.04 and 20.04
I'm attaching both under the names of tlp.conf.bionic.ppa and tlp.conf.focal.ubuntu-official respectively
The files I'm attaching have minor modifications from me; I'm sorry I cannot provide original ones; but these should be very close

the real issue is this:

- Thinkpad T520 with one original battery attached normally
- Ubuntu 20.04
- tlp/focal,focal,now 1.3.1-2 from official Ubuntu repo

executing `sudo tlp start` does not cause battery charge thresholds from /etc/tlp.conf to be applied. More specifically I:

sudo vi /etc/tlp.conf
# set START_CHARGE_THRESH_BAT0 to a new value, save file, exit vi
sudo tlp start
sudo tlp-stat -b
# expected behaviour: I see new charge threshold in /sys/class/power_supply/BAT0/charge_start_threshold
# actual behaviour: I see old charge threshold

Note: `sudo tlp setcharge .. .. BAT0` works as expected. Immediately after it `sudo tlp-stat -b` shows new values

the issue has been completely fixed when I copied over /etc/tlp.conf which remained on a different disk from the times when TLP was installed on the same T520 with Ubuntu 18.04.4

e.g. using /etc/tlp.conf installed from PPA TLP version 1.3.1-1 insetad of /etc/tlp.conf installed from official Ubuntu repo has fixed the issue. Now if I change battery charting start threshold in /etc/tlp.conf and do `sudo tlp start` and immediately follow it with `sudo tlp-stat -b` I see the new threshold applied

I would be curious to know which part of /etc/tlp.conf is causing this.. I'm attaching both version to this bug report, maybe it will help..

The issue seems two-fold:

- official Ubuntu repo is shipping "bad" /etc/tlp.conf
- PPA doesn't have a version for the latest LTS Ubuntu

atagunov, 2020-07-18:
[tlp.conf-variants.zip](https://github.com/linrunner/TLP/files/4942671/tlp.conf-variants.zip)
Here's how I installed TLP

```
sudo add-apt-repository ppa:linrunner/tlp
apt-get install tlp tp-smapi-dkms acpi-call-dkms smartmontools linux-tools-generic powertop
```

Here are details on istalled software:
```
$ cat /etc/issue
Ubuntu 20.04 LTS \n \l

$ apt list --installed | grep '\(tlp\|smapi\|acip\|smartmon\)'

...

smartmontools/focal,now 7.1-1build1 amd64 [installed]
tlp-rdw/focal,focal,now 1.3.1-2 all [installed]
tlp/focal,focal,now 1.3.1-2 all [installed]
tp-smapi-dkms/focal,focal,now 0.43-1 all [installed]
```

linrunner, 2020-07-19:
Hi, 

thanks for your report. 

It speeds up the analysis considerably if you provide the full outputs required by the bug report template and for the exact constellation where the error occurs. 

The crucial section of `diff tlp.conf.bionic.ppa tlp.conf.focal.ubuntu-official` shows:

    < START_CHARGE_THRESH_BAT0=65
    < STOP_CHARGE_THRESH_BAT0=85

    > START_CHARGE_THRESH_BAT0=50
    > #STOP_CHARGE_THRESH_BAT0=80

In **tlp.conf.focal.ubuntu-official** you configured only START_CHARGE_THRESH_BAT0. You must specify _both_ thresholds or version 1.3 will ignore your setting completely, e.g.

    START_CHARGE_THRESH_BAT0=50
    STOP_CHARGE_THRESH_BAT0=100

atagunov, 2020-07-19:
> you must specify both thresholds or version 1.3 will ignore your setting completely

Yes! Once I set both `tlp.conf.focal.ubuntu-official` started working as expected.

Behaviour is a little non-obvious though. I think users' natural intuition is they can set  `START_CHARGE_THRESH_BAT0` only and `STOP_CHARGE_THRESH_BAT0` will automatically go to 100.

Having this explained in comments in `tlp.conf` would be an improvement.

Changing behaviour of the app itself would be an even bigger improvement, but I'm not sure if there's anybody around ready to contribute effort to making changes and testing :)

atagunov, 2020-07-19:
Actually this bit of documentation makes it even worse: https://linrunner.de/tlp/faq/battery.html#faq-erratic-battery-behavior

Erratic battery behavior on ThinkPad T420(s)/T520/W520/X220 (and all later models)

> Symptom: some users report severely reduced battery capacity or sudden drops of the charge level from around 30% to zero when employing charge thresholds.
> Probable cause: conflict with dualmode battery firmware.
> Solution: remove battery thresholds completely or *use only the start threshold*; then recalibrate battery once.

I've seen this erratic behaviour on X220 with both original battery and a KingSener one. I'm very interested to find a solution.. I may solder an extension lead for battery and try to intercept SMBus communication between battery and EC to build a better understanding..

But as things stand documentation advises to *use only the start threshold*. This is clearly not a good advice :)

Ideally, ideally it would default the upper threshold to 100% :-D

linrunner, 2020-07-21:
I agree with you that TLP silently ignoring the incomplete setting does not meet user's expectations. I have fixed the documentation as an immediate measure.

tlp.conf and warning messages will follow.

With 1.3 there was a change in the plausibility checks which results in implicit defaults being no longer possible. 

I think a conflict with the dual mode battery firmware is the reason for the erratic behaviour: 
* https://forums.lenovo.com/t5/Windows-10/Power-Manager-for-Windows-10/m-p/2129075#M794
* https://support.lenovo.com/de/de/solutions/ht104070

atagunov, 2020-07-22:
> I think a conflict with the dual mode battery firmware is the reason for the erratic behaviour:
>     * https://forums.lenovo.com/t5/Windows-10/Power-Manager-for-Windows-10/m-p/2129075#M794
>     * https://support.lenovo.com/de/de/solutions/ht104070

[1]

I think the first of the two links could be added to https://linrunner.de/tlp/faq/battery.html#faq-erratic-battery-behavior
Sadly the 2nd link no longer works

[2]

It will likely be beneficial to adjust advice given in https://linrunner.de/tlp/faq/battery.html#faq-erratic-battery-behavior
That page effectively advises to use `START_CHARGE_THRESH_BAT0` but not `STOP_CHARGE_THRESH_BAT0`. This is exactly how I ended up with wrong config to start with..

[3]

Could be listed as its own issue under https://linrunner.de/tlp/faq/battery.html The symptoms are that charge thresholds are not updated on system restart or on `tlp start`

[4]

I'm sorry, I failed to resist the urge and poured my thoughts out here: https://github.com/linrunner/TLP/issues/509 Please feel free to close that one if you think I should experiment on my own first :)

linrunner, 2020-07-22:
(2) and (3): why don't you just look in the commit referenced above? It's already done.

atagunov, 2020-07-22:
New to github, sorry :)

linrunner, 2020-10-25:
FAQ is changed. A check for completeness (start *and* stop required) is on my list.

closed 2020-10-25 by linrunner