linrunner/TLP #509: tlp topup/storage/levelup/markup/remark

atagunov, 2020-07-22:
**Is your feature request related to a problem? Please describe.**

Have seen on Thinkpad X220 both with original and after-market battery:
https://linrunner.de/tlp/faq/battery.html#faq-erratic-battery-behavior

**Describe the solution you'd like**

https://forums.lenovo.com/t5/Windows-10/Power-Manager-for-Windows-10/m-p/2129075#M794

> 2015-07-28, 12:27 PM
> There is not any Power Manager for Windows 10
>The battery firmware itself will recognize the scenario where the battery is ALWAYS fully charged 100% (over a period of many weeks) and adjust the full charge capacity downwards in a way to maintain maximum battery healt

is one possible explanation. However I have been able to find virtually no other confirmation on the internet that Lenovo batteries detect "always plugged in" condition on their own and act accordingly.

Instead there are multiple statements saying that "battery conservation mode" can be enabled manually. [Example](https://support.lenovo.com/us/en/solutions/ht103159). In fact the 2015 statement that there is no "power manager for Windows 10" appears no longer correct either.

I actually have come to believe another theory behind my issue. Wikipedia [says](https://en.wikipedia.org/wiki/State_of_charge) that remaining charge in a battery can be estimated either on the basis of voltage - which drops down the more current is drawn at the moment - and on the basis of integrating the current. E.g. the controller can measure how many amp-hours have gone in and out. The controller is likely to be using a combination of the two methods. It seems the controller might be able to gauge when battery is fully charged on the basis of voltage. But it may be that as battery is depleted and partially recharged it may be measuring Amp-hours going in and out. And it may be that the controller eventually gets confused without a solid reference point. E.g. the controller thinks battery is 60% full when it's only 30% full.

It's true that if this theory was correct it would be a problem under Windows as well. But we don't know for sure the problem does not exist under Windows. In fact the most common reason people don't like after-market batteries is behaviour like this..

I believe we could try the following work-around for the issue:

- `tlp fullcharge BAT0`
- `tlp discharge BAT0` until we reach `STOP_CHARGE_THRESH_BAT0`

The hope is that once charged fully the battery controller will reset itself to a correct state. From that point it should arrive at a fairly correct charge by only discharging the battery.

It is true that I could implement this as a script and gain experience before suggesting so boldly as a head-line feature.. I may do this. Yet I felt an urge to suggest this.

I have even tried to come up with a good name for this command. If implemented it would complement the `fullcharge`, `discharge`, `calibrate` family. The options I could think of are: `topup/storage/levelup/markup/remark/recharge/refresh`. `storage` is suggested because this is the action you may wish to do before putting the battery for a long time into storage.

* Your use case(s)

I would like to run this command manually or on schedule once in a while to ensure that my battery is charged to a predictable level. The notebook is used plugged-in most of the time. But occasionally I take it out "in the field". I don't care for it to be 100% charged. But would like to have `STOP_CHARGE_THRESH_BAT0` of charge in it at least, not half of that.

* What you want to happen

I would like to see an alternative/experimental solution for https://linrunner.de/tlp/faq/battery.html#faq-erratic-battery-behavior

**Describe alternatives you've considered**

Implement this as a script, gain experience and then pursue as a `tlp` improvement :)

linrunner, 2020-07-22:
Sorry. I have absolutely no intention to implement any more workarounds for Lenovo's (or any other manufacturer's) quirky battery firmware. I've decided long ago that my time is too precious for that. 

Thanks for your understanding.

closed 2020-07-22 by linrunner

atagunov, 2020-07-22:
I don't believe it's fair to say this is a quirk though. Any battery controller that is asked to charge the battery precisely to 60% without being allowed to reach 100% or 0% first has got a tough job to do. We may be having this issue with Lenovo only precisely because these are the only controllers where we can set thresholds.

I do understand lack of time though and I may proceed to have more experimental data before communicating any further on the subject.

atagunov, 2020-07-22:
Looking at it from a different perspective, `TLP` now has commands to go to `0%` and to `100%` and it can facilitate going from `0%` to `X%` via `STOP..` threshold. But there is no command for `TLP` to natively go down from `100%` to `X%`. The toolkit seems a bit incomplete

linrunner, 2020-07-22:
> But there is no command for `TLP` to natively go down from `100%` to `X%`

There's a gross misconception in your use case: the battery does not wear out by remaining at 100%, but while _being charged_ to 100%. What you propose would be just artificial wear by discharging the battery while on AC. Better to discharge it by mobile use.

Good luck.

atagunov, 2020-07-22:
> the battery does not wear out by remaining at 100%, but while _being charged_ to 100%

There are claims Li-Ion live longest around 3.6-3.8V per cell: [link](https://batteryuniversity.com/learn/article/how_to_store_batteries), [link](https://forums.whirlpool.net.au/thread/91qvpn89#r52091825), [link](https://www.techspot.com/community/topics/minimize-battery-wear.196805/post-1348709). Trusting them is a diff question :)

> Storing at 4.2v/cell (fully charged) will increase degradation. Keeping it below ~3.0v/cell (discharged) will degrade it

How about this? `TLP` has commands to

- charge to `100%`
- discharge to `0%`

Wouldn't a 3rd one

- make charge exactly `X%`

make sense? It would charge or discharge as required