linrunner/TLP #511: TLP on Thinkpad 13

dhernandezdeckers, 2020-07-24:
I've been trying to get TLP to work on my Lenovo Thinkpad 13 notebook running on Ubuntu 18 without success. I've installed TLP 1.3.1, as well as  tp-smapi-dkms and acpi-call-dkms, and have secure boot disabled. However, it seems there is no kernel support for natacpi, according to the output of tlp-stat -b:

--- TLP 1.3.1 --------------------------------------------

+++ Battery Features: Charge Thresholds and Recalibrate
natacpi    = inactive (no kernel support)
tpacpi-bat = inactive (superseded by natacpi)
tp-smapi   = inactive (ThinkPad not supported)

+++ Battery Status: BAT1
/sys/class/power_supply/BAT1/manufacturer                   = SMP
/sys/class/power_supply/BAT1/model_name                     = LNV-01AV401
/sys/class/power_supply/BAT1/cycle_count                    = (not supported)
/sys/class/power_supply/BAT1/energy_full_design             =  42010 [mWh]
/sys/class/power_supply/BAT1/energy_full                    =  38890 [mWh]
/sys/class/power_supply/BAT1/energy_now                     =  19470 [mWh]
/sys/class/power_supply/BAT1/power_now                      =   3026 [mW]
/sys/class/power_supply/BAT1/status                         = Discharging

Charge                                                      =   50.1 [%]
Capacity                                                    =   92.6 [%]

My current kernel is Linux 4.15.0-111-generic

I wonder if I'm doing something wrong, or if this is not supported for this particular Thinkpad model.

linrunner, 2020-07-25:
Hi,

ThinkPad 13 is a difficult case, refer to #227 and https://github.com/teleshoes/tpacpi-bat/commit/179cb976b73e310a94b2da3f05c84eba409e4614 . In theory it should work with the tpacpi method, but I'm not 100% convinced that this really applies to all bios versions and the two different model generations.

It is perhaps easiest to install a newer kernel that can handle natacpi, see the [FAQ](https://linrunner.de/tlp/faq/battery.html#which-kernel-module-do-i-need-for-my-hardware). Ubuntu 18.04 HWE kernel (currently 5.3) will do.

What bothers me is that you don't provide the full outputs as required by the issue template. In your case

    tlp-stat -c -s -b 

is sufficient.

dhernandezdeckers, 2020-07-25:
Thanks for your reply, linrunner. Here's the output of tlp-stat -c -s -b (sorry for not providing it before):

--- TLP 1.3.1 --------------------------------------------                                                                                                                                                                                                                     
                                                                                                                                                                                                                                                                               
+++ Configured Settings:                                                                                                                                                                                                                                                       
/etc/tlp.conf L0009: TLP_ENABLE="1"                                                                                                                                                                                                                                            
/etc/tlp.conf L0017: TLP_PERSISTENT_DEFAULT="0"                                                                                                                                                                                                                                
/etc/tlp.conf L0021: DISK_IDLE_SECS_ON_AC="0"                                                                                                                                                                                                                                  
/etc/tlp.conf L0022: DISK_IDLE_SECS_ON_BAT="2"                                                                                                                                                                                                                                 
/etc/tlp.conf L0025: MAX_LOST_WORK_SECS_ON_AC="15"                                                                                                                                                                                                                             
/etc/tlp.conf L0026: MAX_LOST_WORK_SECS_ON_BAT="60"
defaults.conf L0010: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
defaults.conf L0011: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"
/etc/tlp.conf L0081: SCHED_POWERSAVE_ON_AC="0"
/etc/tlp.conf L0082: SCHED_POWERSAVE_ON_BAT="1"
/etc/tlp.conf L0086: NMI_WATCHDOG="0"
/etc/tlp.conf L0103: DISK_DEVICES="sda sdb"
/etc/tlp.conf L0109: DISK_APM_LEVEL_ON_AC="254 254"
/etc/tlp.conf L0110: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0018: DISK_IOSCHED="keep keep"
/etc/tlp.conf L0131: SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
/etc/tlp.conf L0132: SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
/etc/tlp.conf L0145: AHCI_RUNTIME_PM_TIMEOUT="15"
/etc/tlp.conf L0149: PCIE_ASPM_ON_AC="performance"
/etc/tlp.conf L0150: PCIE_ASPM_ON_BAT="powersave"
/etc/tlp.conf L0154: RADEON_POWER_PROFILE_ON_AC="high"
/etc/tlp.conf L0155: RADEON_POWER_PROFILE_ON_BAT="low"
/etc/tlp.conf L0162: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
/etc/tlp.conf L0163: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
/etc/tlp.conf L0166: WIFI_PWR_ON_AC="off"
/etc/tlp.conf L0167: WIFI_PWR_ON_BAT="on"
/etc/tlp.conf L0170: WOL_DISABLE="Y"
/etc/tlp.conf L0174: SOUND_POWER_SAVE_ON_AC="0"
/etc/tlp.conf L0175: SOUND_POWER_SAVE_ON_BAT="1"
/etc/tlp.conf L0178: SOUND_POWER_SAVE_CONTROLLER="Y"
/etc/tlp.conf L0184: BAY_POWEROFF_ON_AC="0"
/etc/tlp.conf L0185: BAY_POWEROFF_ON_BAT="0"
/etc/tlp.conf L0187: BAY_DEVICE="sr0"
/etc/tlp.conf L0190: RUNTIME_PM_ON_AC="on"
/etc/tlp.conf L0191: RUNTIME_PM_ON_BAT="auto"
defaults.conf L0039: RUNTIME_PM_DRIVER_BLACKLIST="amdgpu mei_me nouveau nvidia pcieport radeon"
/etc/tlp.conf L0205: USB_AUTOSUSPEND="1"
/etc/tlp.conf L0214: USB_BLACKLIST_BTUSB="0"
/etc/tlp.conf L0218: USB_BLACKLIST_PHONE="0"
/etc/tlp.conf L0222: USB_BLACKLIST_PRINTER="1"
/etc/tlp.conf L0226: USB_BLACKLIST_WWAN="1"
defaults.conf L0045: USB_AUTOSUSPEND_DISABLE_ON_SHUTDOWN="0"
/etc/tlp.conf L0241: RESTORE_DEVICE_STATE_ON_STARTUP="0"
defaults.conf L0047: RESTORE_THRESHOLDS_ON_BAT="0"
defaults.conf L0048: NATACPI_ENABLE="1"
defaults.conf L0049: TPACPI_ENABLE="1"
defaults.conf L0050: TPSMAPI_ENABLE="1"
/etc/tlp.conf L0013: TLP_DEFAULT_MODE="AC"
/etc/tlp.conf L0059: CPU_HWP_ON_AC="balance_performance"
/etc/tlp.conf L0060: CPU_HWP_ON_BAT="balance_power"
/etc/tlp.conf L0098: ENERGY_PERF_POLICY_ON_AC="performance"
/etc/tlp.conf L0099: ENERGY_PERF_POLICY_ON_BAT="power"
/etc/tlp.conf L0158: RADEON_DPM_STATE_ON_AC="performance"
/etc/tlp.conf L0159: RADEON_DPM_STATE_ON_BAT="battery"
/etc/tlp.conf L0273: START_CHARGE_THRESH_BAT0="75"
/etc/tlp.conf L0274: STOP_CHARGE_THRESH_BAT0="80"
/etc/tlp.conf L0276: START_CHARGE_THRESH_BAT1="75"
/etc/tlp.conf L0277: STOP_CHARGE_THRESH_BAT1="80"

+++ System Info
System         = LENOVO ThinkPad 13 20GJCTO1WW
BIOS           = R0CET37W (1.25 )
Release        = Ubuntu 18.04.4 LTS
Kernel         = 4.15.0-111-generic #112-Ubuntu SMP Thu Jul 9 20:32:34 UTC 2020 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-4.15.0-111-generic root=UUID=590b32db-72ef-475a-9814-ef7afc08f8a8 ro quiet splash vt.handoff=1
[... 2 lines cut ...]

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 11:16:04 AM,    310 sec(s) ago
[... cut ...]
/sys/class/power_supply/BAT1/energy_full_design             =  42010 [mWh]
/sys/class/power_supply/BAT1/energy_full                    =  38890 [mWh]
/sys/class/power_supply/BAT1/energy_now                     =  11910 [mWh]
/sys/class/power_supply/BAT1/power_now                      =   4443 [mW]
/sys/class/power_supply/BAT1/status                         = Discharging

Charge                                                      =   30.6 [%]
Capacity                                                    =   92.6 [%]

linrunner, 2020-07-25:
> LENOVO ThinkPad 13 

Yet another model string tpacpi-bat is not aware of :(.

What happens with the HWE  kernel?

dhernandezdeckers, 2020-07-25:
Good news! I Installed HWE kernel (5.4), and now it works. natacpi is active.

Thanks!

linrunner, 2020-07-26:
Wait, we are not done yet! The newer kernel provides only the thresholds. For `tlp discharge` to work, a change to tpacpi-bat is necessary.

Please edit the file **/usr/share/tlp/tpacpi-bat** by hand as shown in [this commit](https://github.com/linrunner/TLP/commit/a714590ea35ba888dcdae7468c756db447ae2d31).

Then show again the output of

     tlp-stat -c -s -b

dhernandezdeckers, 2020-07-26:
Ok, I've done this, but discharge does not seem to work. Here's the output of tlp-stat -c -s -b:

--- TLP 1.3.1 --------------------------------------------

+++ Configured Settings:
/etc/tlp.conf L0009: TLP_ENABLE="1"
/etc/tlp.conf L0017: TLP_PERSISTENT_DEFAULT="0"
/etc/tlp.conf L0021: DISK_IDLE_SECS_ON_AC="0"
/etc/tlp.conf L0022: DISK_IDLE_SECS_ON_BAT="2"
/etc/tlp.conf L0025: MAX_LOST_WORK_SECS_ON_AC="15"
/etc/tlp.conf L0026: MAX_LOST_WORK_SECS_ON_BAT="60"
defaults.conf L0010: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
defaults.conf L0011: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"
/etc/tlp.conf L0081: SCHED_POWERSAVE_ON_AC="0"
/etc/tlp.conf L0082: SCHED_POWERSAVE_ON_BAT="1"
/etc/tlp.conf L0086: NMI_WATCHDOG="0"
/etc/tlp.conf L0103: DISK_DEVICES="sda sdb"
/etc/tlp.conf L0109: DISK_APM_LEVEL_ON_AC="254 254"
/etc/tlp.conf L0110: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0018: DISK_IOSCHED="keep keep"
/etc/tlp.conf L0131: SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
/etc/tlp.conf L0132: SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
/etc/tlp.conf L0145: AHCI_RUNTIME_PM_TIMEOUT="15"
/etc/tlp.conf L0149: PCIE_ASPM_ON_AC="performance"
/etc/tlp.conf L0150: PCIE_ASPM_ON_BAT="powersave"
/etc/tlp.conf L0154: RADEON_POWER_PROFILE_ON_AC="high"
/etc/tlp.conf L0155: RADEON_POWER_PROFILE_ON_BAT="low"
/etc/tlp.conf L0162: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
/etc/tlp.conf L0163: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
/etc/tlp.conf L0166: WIFI_PWR_ON_AC="off"
/etc/tlp.conf L0167: WIFI_PWR_ON_BAT="on"
/etc/tlp.conf L0170: WOL_DISABLE="Y"
/etc/tlp.conf L0174: SOUND_POWER_SAVE_ON_AC="0"
/etc/tlp.conf L0175: SOUND_POWER_SAVE_ON_BAT="1"
/etc/tlp.conf L0178: SOUND_POWER_SAVE_CONTROLLER="Y"
/etc/tlp.conf L0184: BAY_POWEROFF_ON_AC="0"
/etc/tlp.conf L0185: BAY_POWEROFF_ON_BAT="0"
/etc/tlp.conf L0187: BAY_DEVICE="sr0"
/etc/tlp.conf L0190: RUNTIME_PM_ON_AC="on"
/etc/tlp.conf L0191: RUNTIME_PM_ON_BAT="auto"                                                                                                                                                                                                                                  
defaults.conf L0039: RUNTIME_PM_DRIVER_BLACKLIST="amdgpu mei_me nouveau nvidia pcieport radeon"                                                                                                                                                                                
/etc/tlp.conf L0205: USB_AUTOSUSPEND="1"                                                                                                                                                                                                                                       
/etc/tlp.conf L0214: USB_BLACKLIST_BTUSB="0"                                                                                                                                                                                                                                   
/etc/tlp.conf L0218: USB_BLACKLIST_PHONE="0"                                                                                                                                                                                                                                   
/etc/tlp.conf L0222: USB_BLACKLIST_PRINTER="1"                                                                                                                                                                                                                                 
/etc/tlp.conf L0226: USB_BLACKLIST_WWAN="1"                                                                                                                                                                                                                                    
defaults.conf L0045: USB_AUTOSUSPEND_DISABLE_ON_SHUTDOWN="0"                                                                                                                                                                                                                   
/etc/tlp.conf L0241: RESTORE_DEVICE_STATE_ON_STARTUP="0"                                                                                                                                                                                                                       
defaults.conf L0047: RESTORE_THRESHOLDS_ON_BAT="0"                                                                                                                                                                                                                             
defaults.conf L0048: NATACPI_ENABLE="1"                                                                                                                                                                                                                                        
defaults.conf L0049: TPACPI_ENABLE="1"                                                                                                                                                                                                                                         
defaults.conf L0050: TPSMAPI_ENABLE="1"                                                                                                                                                                                                                                        
[... cut ...]
/sys/class/power_supply/BAT1/power_now                      =      0 [mW]
/sys/class/power_supply/BAT1/status                         = Unknown (threshold may prevent charging)

/sys/class/power_supply/BAT1/charge_start_threshold         =     80 [%]
/sys/class/power_supply/BAT1/charge_stop_threshold          =     80 [%]
tpacpi-bat.BAT1.forceDischarge                              =      0

Charge                                                      =   78.8 [%]
Capacity                                                    =   92.6 [%]

linrunner, 2020-07-27:
"does not work"? Please show the output of

    sudo tlp discharge BAT1

dhernandezdeckers, 2020-07-27:
here's the output of sudo tlp discharge BAT1:

Currently discharging battery BAT1:
voltage            =  11943 [mV]
remaining capacity =  30580 [mWh]
remaining percent  =     78 [%]
remaining time     =    474 [min]
power              =   3869 [mW]
state              = Discharging
force discharge    = 2
Press Ctrl+C to cancel.

What I find strange is that in the tlp.conf file I've set START_CHARGE_THRESH_BAT0="75" and STOP_CHARGE_THRESH_BAT1="80", but in the output of tlp-stat -c -s -b both thresholds are at 80%.

linrunner, 2020-07-27:
I don't get why do you think it "doesn't work":

> state = Discharging

is proof enough that it's actually discharging.

> force discharge = 2

The value 2 is unusual, should be 1. But as long as it discharges, I wouldn't worry.

> What I find strange is that in the tlp.conf file I've set START_CHARGE_THRESH_BAT0="75" and STOP_CHARGE_THRESH_BAT1="80", but in the output of tlp-stat -c -s -b both thresholds are at 80%.

Did you check the [FAQ](https://linrunner.de/tlp/faq/battery.html#charge-thresholds-shown-by-tlp-stat-b-do-not-correspond-to-the-configured-ones)?

dhernandezdeckers, 2020-07-27:
Ok, that means I'm all set. Thank you!

closed 2020-07-29 by dhernandezdeckers