linrunner/TLP #512: TLP freezes Ubuntu 20.04 as soon as installed if it is on battery mode.

aman-roy, 2020-07-25:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs (see below)

**Describe the bug**

After installing tlp in Ubuntu 20.04 on my Asus vivobook s14 with Ryzen 5 4500u processor, it freezes if the laptop is on battery. As soon as the power source gets connected, it starts working fine.

**Expected behavior**

It should not freeze on battery mode.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?
battery
2. Actions to reproduce the behavior
install tlp
3. Shell commands entered and their output
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.**
can't get that, because it freezes as soon as I install it. Even after restarting, it freezes on log in screen and doesn't work until power supply is provided.

**Additional context**

Add any other context about the problem here.

linrunner, 2020-07-25:
Hi,

you have to isolate the causing setting by following the [Troubleshooting Guide](https://linrunner.de/tlp/support/troubleshooting.html). Difficult and tedious because of the freezes, but it can only done on your hardware.

aman-roy, 2020-07-27:
It is solved now. We can close it now.

linrunner, 2020-07-27:
To help others running into the same issue, could you please post the solution – e.g. the changed setting(s) or other cause?

aman-roy, 2020-07-29:
Sure.

In Ubuntu 20.04, don't just install using apt.
```
sudo apt install tlp
```

Instead, you should first add the TLP-PPA and then install it. Like this - 
```
sudo add-apt-repository ppa:linrunner/tlp
sudo apt update
sudo apt install tlp tlp-rdw
```

linrunner, 2020-07-29:
This is generally advisable. However, for 20.04 the TLP version in the PPA and the Ubuntu repos is the same (version 1.3.1).

closed 2020-10-25 by linrunner