linrunner/TLP #513: TLP uses non-standard sysfs attributes to control battery charge thresholds

pobrn, 2020-07-26:
**Describe the bug**

[This commit](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=813cab8f3994250e136819ae48fbd1c95d980466) specifies that the preferred names of the sysfs attributes that control charging thresholds are `charge_control_{start, end}_threshold`.

TLP uses `charge_{start, stop}_threshold`. (Or at least `tlp-stat -b` does not display anything about charge limits when the standard attributes exist.) These two attributes are not documented anywhere, and only used in the `thinkpad_acpi` kernel driver.

**Expected behavior**

TLP uses `/sys/class/power_suppy/*/charge_control_{start, end}_threshold`.

linrunner, 2020-07-27:
I have put the topic on hold until the corresponding patches for thinkpad_acpi are applied.

pobrn, 2020-08-04:
[Merged.](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?id=09a0bd07764359650d41dbcb723f81321e523217)

linrunner, 2020-08-05:
Nice :)

linrunner, 2021-03-25:
Now implemented with the upcoming plugin architecture --> TLP 1.4.

https://github.com/linrunner/TLP/commit/ea6769e25b4420f53f48fb565a2fdaf69afc0912

closed 2021-03-25 by linrunner