linrunner/TLP #514: Allow specifying just one of the battery charge thresholds

pobrn, 2020-07-26:
**Describe the solution you'd like**

Allowing just one of the battery charge thresholds to be specified would make TLP's battery features compatible with more laptops. For example, the `asus-wmi` kernel driver provides only the `charge_control_end_threshold` attribute, which means that TLP completely ignores it (disregarding the issue laid out in #513).

Are there any technical obstacles why this limitation exists?

(I apologize if this has been asked/answered before.)

linrunner, 2020-07-27:
10 years experience with ThinkPads shows that charge thresholds are a very support intensive topic. 

This is due to many technical peculiarities or firmware bugs of the different model series and on the other hand due to users who have difficulties in understanding the concept.

If contributor(s) possessing the necessary range of hardware – let's say for the ASUS brand – step up to:

* provide well tested pull requests (that don't break TP functionality); and
* shoulder mainentance and support of the code here in the issue tracker

then I'll consider merging. 

And yes, the hurdle is deliberately high. I don't have time to do this myself.

pobrn, 2020-07-27:
I apologize because I have not been 100% explicit in my original text. What I'd like to see is the ability to set only one of the charge limits *if only one of them is available*.

Furthermore, in my understanding the semantics of these attributes are documented, and it is the responsibility of the driver to handle hardware/firmware peculiarities if it chooses to provide this interface.

linrunner, 2020-07-27:
> and it is the responsibility of the driver to handle hardware/firmware peculiarities if it chooses to provide this interface.

In theory yes, in practice the user space tool has to deal with them. Believe me, I'm doing this for 10 years now.

pobrn, 2020-07-27:
I see. So it isn't probable that this will be implemented for all drivers that support these attributes, correct? The reason why I would like to see this feature is that I'm writing a platform driver for my current laptop, and it only supports the "stop threshold", and I would like to see TLP support for it.

linrunner, 2020-07-27:
If you write the driver yourself, you can also provide detailed information for whitelisting models and I suppose you would help with debugging and support here. That's a different situation. Go ahead with your driver, please.

ps. may I ask which laptop? If you don't want to say this in public, drop me an email.

pobrn, 2020-07-27:
It's an [Intel LAPQC71X](https://www.intel.com/content/dam/support/us/en/documents/laptops/whitebook/QC71_PROD_SPEC.pdf) based laptop. This design is sold under different names: XMG Fusion 15, Eluktronics MAG 15, Aftershock Vapor 15 - the ones I know of. It's probably not a mainstream laptop, less so among Linux users, but hey, at least it'll work for me.

linrunner, 2020-07-27:
That'll be difficult to whitelist properly. Let's see.

pobrn, 2020-07-27:
`/sys/devices/virtual/dmi/id/board_name` provides - as far as I see - a reliable way to identify the board.

linrunner, 2020-10-25:
What's the state of your driver?

pobrn, 2020-10-25:
The battery charge threshold controlling parts are and have been done for some time as you can see [in the repository](https://github.com/pobrn/qc71_laptop). However, there are some things which need to be sorted out before I attempt to submit it.

linrunner, 2020-10-25:
I see. I have not yet come to implement because it requires some basic changes. But it is still on my list.

linrunner, 2021-08-04:
@pobrn : with TLP 1.4 all the infrastructure to support new drivers is in place. What's your status?

pobrn, 2021-08-04:
The driver is not yet in the kernel, there are some problematic points which need to be reworked, but I don't know when I'll have time to do that. As I've mentioned in October, the functionality is there and working.

linrunner, 2021-12-07:
@pobrn : Any news on this?