linrunner/TLP #515: Missing USB tethering disable/enable device functionality

jacob-emery, 2020-07-27:
**Is your feature request related to a problem? Please describe.**

I set DEVICES_TO_DISABLE_ON_LAN_CONNECT="wifi" and I expected it to work also when USB tethering from my Android device. However, the wifi stays on.
Since on the Network section of Gnome Settings there is a distinction between Ethernet and USB Ethernet I guess this the reason why it does not work.
(My USB Ethernet interface is called enp0s20... and not something like eth1.)

**Describe the solution you'd like**

I would like to have the ability to automatically disable the wifi when USB tethering.
I would suggest to implement this under DEVICES_TO_DISABLE_ON_LAN_CONNECT or to add an additional option e.g. DEVICES_TO_DISABLE_ON_USB_LAN_CONNECT

**Describe alternatives you've considered**

Not aware of any alternatives.

pobrn, 2020-07-27:
FYI, `enp0s20` is a completely normal name for an ethernet interface nowadays. It means ethernet device in PCI bus 0, slot 20 (afaik).

jacob-emery, 2020-07-27:
Its complete name is enp0s20f0u4u1u1

jacob-emery, 2020-07-27:
Do you know then why the wifi does not deactivate automatically when i plug in my phone and activate USB tethering?

linrunner, 2020-07-28:
Hi, 

imho this works already. I just cross checked with my Android phone. It is also known to work for USB ethernet adapters in  general.

The feature depends on NetworkManager's classification of the USB connection. May I see the output of

    tlp-stat -c -s -r -e -u
    nmcli d

please (with the phone connected and tethering enabled of course).

jacob-emery, 2020-07-28:
I really thought at first it was a bg or a missing feature with TLP, but after you said it works for you I realized it must be something with my configuration. I had installed TLP a long time ago and I was sure that I had done it properly, but my configuration was not finished. I did not follow [these steps (Radio Device Wizard)](https://wiki.archlinux.org/index.php/TLP#Radio_Device_Wizard_(tlp-rdw)) present on the Arch Wiki's TLP page. Now everything works as it should =)

closed 2020-07-28 by jacob-emery