linrunner/TLP #516: change to ac or bat not detected, original modeoperation is persistent since boot or " tlp start"

userofryzen, 2020-08-01:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs (see below)

**Describe the bug**
https://gist.github.com/userofryzen/213acd3efbf9f4f95d82c3d12af7c1f9(don't mind the persistent configuration, this is a workaround to run on battery mode always because if I power tlp on ac and disconnect after tlp doesn't detect it)
 
the important part is:
+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 04:56:42,   1749 sec(s) ago
Mode           = battery
Power source   = AC
This happens when I power on with battery but after that I connect the ac source
OR

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 04:56:42,   1749 sec(s) ago
Mode           = AC
Power source   = battery
This happens when I power on with AC but after that I disconnect the ac source

If I run "sudo tlp start" then TLP resets itself and detect if it is on battery or ac and change the mode of operation, but this is not automatic!

**Expected behavior**

TLP detects the change on power supply and change the operation mode automatically..

**To Reproduce**

having the laptop reported on gist. Install any distribution with TLP or installing it after.. Power on the PC and connecting or disconnecting the AC power. The pc would not recognicze it.

1. Does the problem occur on battery or AC or both? BOTH
2. Actions to reproduce the behaviour
3. Shell commands entered and their output
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.**

**Additional context**

output of tlp-stat --pev
--- TLP 1.3.1 --------------------------------------------

+++ Monitor power supply events -- cancel with ^C

monitor will print the received events for:
UDEV - the event which udev sends out after rule processing

there is no udev sends detected from TLP at least. 

output of tlp-stat --psup

--- TLP 1.3.1 --------------------------------------------

+++ Power supply diagnostic
/sys/class/power_supply/ADP0/type:Mains
/sys/class/power_supply/ADP0/online:1
/sys/class/power_supply/ADP0/device/path:\_SB_.ADP0
/sys/class/power_supply/BAT0/type:Battery
/sys/class/power_supply/BAT0/present:1
/sys/class/power_supply/BAT0/status:Unknown
/sys/class/power_supply/BAT0/device/path:\_SB_.PCI0.LPCB.EC0_.BAT0

linrunner, 2020-08-01:
Without a udev event there is no possibility for TLP to recognize the change of power source to BAT. This is a BIOS or kernel issue and as such off topic here. Good luck.

closed 2020-08-01 by linrunner

userofryzen, 2020-08-01:
> Without a udev event there is no possibility for TLP to recognize the change of power source to BAT. This is a BIOS or kernel issue and as such off topic here. Good luck.

But the DE shows when the pc is charging and when not so.. o don't know because it's something weird that when I run tlp-stat, tlp shows the correct power source but assign bad mode operation..
Like ..
Power source=battery
Mode : AC 
Tlp detects the power supply but there is no change in mode operation. Why? 
And.. actually I have seen that my pc has a weird name to name AC .. it's up. It's adp0 or something like this

linrunner, 2020-08-05:
The reason is already given in my above post. TLP works [event based](https://linrunner.de/tlp/introduction.html#event-driven-architecture) whereas the DE continously polls the power status i guess. 

The missing events are a kernel or BIOS issue. No workaround possible in TLP.