linrunner/TLP #518: USB Autosuspend Blacklist Settings Not Applied at Boot on Ubuntu 20.04

pointful, 2020-08-01:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs (see below)

**Describe the bug**

Trying to disable autosuspend on bluetooth USB device, but setting is not being applied at boot on Ubuntu 20.04. Have enable both general bluetooth blacklist as well as specific device to blacklist.

Settings are not applied at boot, and are not applied when calling "tlp init start" directly from the command line, but they are applied when calling "tlp start" directly from the command line. See full details below with a full tlp-stat between each state.

$ sudo tlp-stat
--- TLP 1.3.1 --------------------------------------------

+++ Configured Settings:
defaults.conf L0004: TLP_ENABLE="1"
defaults.conf L0005: TLP_PERSISTENT_DEFAULT="0"
defaults.conf L0006: DISK_IDLE_SECS_ON_AC="0"
defaults.conf L0007: DISK_IDLE_SECS_ON_BAT="2"
defaults.conf L0008: MAX_LOST_WORK_SECS_ON_AC="15"
defaults.conf L0009: MAX_LOST_WORK_SECS_ON_BAT="60"
defaults.conf L0010: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
defaults.conf L0011: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"
defaults.conf L0012: SCHED_POWERSAVE_ON_AC="0"
defaults.conf L0013: SCHED_POWERSAVE_ON_BAT="1"
defaults.conf L0014: NMI_WATCHDOG="0"
defaults.conf L0015: DISK_DEVICES="nvme0n1 sda"
defaults.conf L0016: DISK_APM_LEVEL_ON_AC="254 254"
defaults.conf L0017: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0018: DISK_IOSCHED="keep keep"
defaults.conf L0019: SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
defaults.conf L0020: SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
defaults.conf L0021: AHCI_RUNTIME_PM_TIMEOUT="15"
defaults.conf L0022: PCIE_ASPM_ON_AC="default"
defaults.conf L0023: PCIE_ASPM_ON_BAT="default"
defaults.conf L0024: RADEON_POWER_PROFILE_ON_AC="default"
defaults.conf L0025: RADEON_POWER_PROFILE_ON_BAT="default"
defaults.conf L0026: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
defaults.conf L0027: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
defaults.conf L0028: WIFI_PWR_ON_AC="off"
defaults.conf L0029: WIFI_PWR_ON_BAT="on"
defaults.conf L0030: WOL_DISABLE="Y"
defaults.conf L0031: SOUND_POWER_SAVE_ON_AC="0"
defaults.conf L0032: SOUND_POWER_SAVE_ON_BAT="1"
defaults.conf L0033: SOUND_POWER_SAVE_CONTROLLER="Y"
defaults.conf L0034: BAY_POWEROFF_ON_AC="0"
defaults.conf L0035: BAY_POWEROFF_ON_BAT="0"
defaults.conf L0036: BAY_DEVICE="sr0"
defaults.conf L0037: RUNTIME_PM_ON_AC="on"
defaults.conf L0038: RUNTIME_PM_ON_BAT="auto"
defaults.conf L0039: RUNTIME_PM_DRIVER_BLACKLIST="amdgpu mei_me nouveau nvidia pcieport radeon"
/etc/tlp.conf L0323: USB_AUTOSUSPEND="1"
/etc/tlp.conf L0337: USB_BLACKLIST_BTUSB="1"
/etc/tlp.conf L0343: USB_BLACKLIST_PHONE="1"
defaults.conf L0043: USB_BLACKLIST_PRINTER="1"
/etc/tlp.conf L0355: USB_BLACKLIST_WWAN="1"
defaults.conf L0045: USB_AUTOSUSPEND_DISABLE_ON_SHUTDOWN="0"
defaults.conf L0046: RESTORE_DEVICE_STATE_ON_STARTUP="0"
defaults.conf L0047: RESTORE_THRESHOLDS_ON_BAT="0"
defaults.conf L0048: NATACPI_ENABLE="1"
defaults.conf L0049: TPACPI_ENABLE="1"
defaults.conf L0050: TPSMAPI_ENABLE="1"
/etc/tlp.conf L0331: USB_BLACKLIST="8087:0029"
/etc/tlp.conf L0426: START_CHARGE_THRESH_BAT0="75"
/etc/tlp.conf L0427: STOP_CHARGE_THRESH_BAT0="80"

+++ System Info
System         = LENOVO ThinkPad P1 Gen 2 20QUS46300
BIOS           = N2OET44W (1.31 )
Release        = Ubuntu 20.04.1 LTS
Kernel         = 5.4.0-42-generic #46-Ubuntu SMP Fri Jul 10 00:24:02 UTC 2020 x86_64
/proc/cmdline  = BOOT_IMAGE=/vmlinuz-5.4.0-42-generic root=/dev/mapper/ubuntu--vg-root ro quiet splash vt.handoff=7
[... 857 lines cut ...]

+++ Battery Features: Charge Thresholds and Recalibrate
natacpi    = active (data, thresholds)
tpacpi-bat = inactive (kernel module 'acpi_call' not installed)
tp-smapi   = inactive (ThinkPad not supported)

+++ ThinkPad Battery Status: BAT0
/sys/class/power_supply/BAT0/manufacturer                   = Celxpert
/sys/class/power_supply/BAT0/model_name                     = 5B10W13953
/sys/class/power_supply/BAT0/cycle_count                    =      1
/sys/class/power_supply/BAT0/energy_full_design             =  80400 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  78900 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  78900 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Full

/sys/class/power_supply/BAT0/charge_start_threshold         =     75 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold          =     80 [%]

Charge                                                      =  100.0 [%]
Capacity                                                    =   98.1 [%]

+++ Recommendations
* Install acpi_call kernel module for ThinkPad battery recalibration
* Install smartmontools for disk drive health info

linrunner, 2020-08-05:
Hi,

the initialization of all USB devices on boot (and on plugin) is done via udev events. Therefore `tlp init start` has no effect, whereas `tlp start` initializes all USB devices again in a batch.

I can only speculate that in early boot the kernel driver somehow blocks write access to **power/control**. 

As a workaround you may try to use USB batch mode by adding

    X_TLP_USB_MODE=1

to your config file.

pointful, 2020-08-05:
Excellent! That workaround worked. Thanks!

closed 2020-08-08 by linrunner