linrunner/TLP #519: [T450 on PopOS 20.04] External battery threshold_start_charge reported by TLP is wrong.

AkechiShiro, 2020-08-09:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs (see below)

**Describe the bug**
The reported value of the start charge threshold seems to be wrong, I've had activated the battery threshold feature on my Thinkpad T450, had a bit of an episode with weird batteries erratic behaviors (about 3 batteries). After all that I disabled the feature, did a downgrade of the EC firmware (currently 1.02 JBHT15WW) to be sure it's not lying about the threshold feature being disabled as I've read something about it happening to some models in the FAQ.
I've recently bought a new external battery that had the feature threshold disabled (I mainly checked using Lenovo Vantage on Windows 10).

Current my tlp-stat -b output:
```--- TLP 1.3.1 --------------------------------------------

+++ Battery Features: Charge Thresholds and Recalibrate
natacpi    = active (data, thresholds)
tpacpi-bat = active (recalibrate)
tp-smapi   = inactive (ThinkPad not supported)

+++ ThinkPad Battery Status: BAT1 (Ultrabay / Slice / Replaceable)
/sys/class/power_supply/BAT1/manufacturer                   = SANYO
/sys/class/power_supply/BAT1/model_name                     = 45N1125
/sys/class/power_supply/BAT1/cycle_count                    = (not supported)
/sys/class/power_supply/BAT1/energy_full_design             =  72150 [mWh]
/sys/class/power_supply/BAT1/energy_full                    =  57130 [mWh]
/sys/class/power_supply/BAT1/energy_now                     =  57130 [mWh]
/sys/class/power_supply/BAT1/power_now                      =      0 [mW]
/sys/class/power_supply/BAT1/status                         = Full

/sys/class/power_supply/BAT1/charge_start_threshold         =     96 [%]
/sys/class/power_supply/BAT1/charge_stop_threshold          =    100 [%]
tpacpi-bat.BAT1.forceDischarge                              =      0

Charge                                                      =  100.0 [%]
Capacity                                                    =   79.2 [%]
```

I just moved the computer from one place to another and the battery was previously full, it lost 1% and reported being in the "unknown" state when I plugged back the AC, I've thus suspected some threshold feature is blocking the charge despite one being not activated at first.
So I unplug the AC and wait for the battery to be at 95.9% then plug back. The battery is still not charging, and the state is still "unknown (threshold may be active)".
Now, if I reboot on Windows 10 and check the threshold start charge, it is **NOT** set at 96% but at 95% (I'm being a bit picky but what would've happened if the difference was very big).
So I would like to know where does TLP gets the threshold charge values from, and in case it's coming from the EC firmware registers then shouldn't there be only one number correctly reported by both Lenovo Vantage and TLP ?
To confirm this was the issue I waited that the battery got to 94.9% and plugged the AC and it thus began charging confirming that there is an issue with the values being read by TLP.

**Expected behavior**

Correct threshold_start_charge percentage reported just as Lenovo Vantage (on Windows 10) or a possible explanation for the unexpected behavior.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?
Only on AC.
2. Actions to reproduce the behaviour
Read tlp-stat -b output (most importantly the threshold_start_charge value) and compare it with Lenovo Vantage threshold_start_charge reported values.
3. Shell commands entered and their output
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.**
[Full output of tlp-stat.](https://gist.github.com/AkechiShiro/2bfa92237e9cafd1023679ae4e0246f4)

**Additional context**

Another issue I've seen is that despite disabling the battery threshold on Lenovo Vantage on Windows.
When I unplug physically the battery from the computer and re-plug it, the charge threshold feature comes back on but TLP does report the same numbers no matter what.
My BIOS is updated to the latest one from Lenovo the 1.37(JBET73WW).
I can also provide the ACPI tables for my computer if there is any need for them.

I did and do not have any internal battery in the computer right now, so I don't know if this issue affects only external batteries or internal batteries too.

linrunner, 2020-08-09:
Hi, interesting observation. 

> So I would like to know where does TLP gets the threshold charge values from

Part of the answer is in plain sight in your outputs:

    /sys/class/power_supply/BAT1/charge_start_threshold         =     96 [%]
    /sys/class/power_supply/BAT1/charge_stop_threshold          =    100 [%]

That's the kernel interface invented with 4.17 (not part of TLP!), which uses ACPI calls ([see thinkpad_acpi driver code](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/platform/x86/thinkpad_acpi.c?id=HEAD#n9316)) to get and set thresholds. These are the same calls implemented in [tpacpi-bat](https://github.com/teleshoes/tpacpi-bat), which TLP uses for older kernels and force_discharge. It can be assumed that the BIOS passes these calls to the EC firmware, which controls the entire charging process. 

This interface is the only way under Linux to read and write the charge thresholds for ThinkPads T420/X220 and newer. Afaik there is no published documentation from Lenovo about it.

Short answer: you either have to take TLP's current implementation as is or leave it.

closed 2020-08-09 by linrunner

AkechiShiro, 2020-08-10:
Well, thanks for the explanation (I saw that there were files provided by the system but didn't what it was exactly), I wasn't sure if the issue was really with TLP or not, it may be with the kernel interface. I'm aware that Lenovo doesn't provide any documentation about it (and that's a real pain), do you know how the charge threshold gets enforced ? It is done by the EC ? 
Because after having done a few SMBus capture (of batteries on my Thinkpad) and reading them (using SBS).
I found nothing really relevant to the threshold being set or not in the battery status information or other things that I could sniff using an logical analyzer (one thing was that the battery still does get authenticated even tho on my model there is supposedly no whitelist for batteries, there is for LCD screens tho but that's another matter). 
So it seems the settings is at **"higher level"**. 
I was wondering if you could also enlighten me about this or if I should try to extract my own EC firmware from an update and find the answer by doing some reverse but since I'm not confident in my skills yet, I didn't want go down that path but if need be, I'll then probably go for it.

pobrn, 2020-08-10:
Allegedly, @smclt30p is in possession of [ACPI firmware documentation](https://github.com/linrunner/TLP/issues/321#issuecomment-364574576) from Lenovo.

By the way, can you directly read those files (`/sys/class/power_supply/BAT1/charge_{start,stop}_threshold`) using `cat` for example? What do you get?

AkechiShiro, 2020-08-11:
I get these values using `cat`.
```
0 # for charge_start_threshold
100 # for charge_stop_threshold
```
Do you @pobrn know if @smclt30p is legally bound by anything making him unable to share them online or are they already online, I'd like to take a look if possible ?

smclt30p, 2020-08-11:
Technically speaking you don't need the docs. Its a simple Excel spreadsheet with ACPI method names and parameters. Nothing more. It does not explain how the internal ACPI firmware works.

You can look up these paremeters in the source code.

AkechiShiro, 2020-08-11:
I see so it's somewhat like a lesser complete specification. Thanks you for the precision.
But these ACPI method names follow somewhat the ACPI specification no ? (or maybe some of them do and some of them don't, like there are some additional ones made by Lenovo that are only documented in your excel spreadsheet, I mean I don't really know, just wondering)

smclt30p, 2020-08-11:
It's really a mix of everything, because the battery control driver is very old (think 10-15+ years) so I assume many people worked on it over its lifetime, thus the different approach to some stuff.

The driver is known to work very well on up-to xx30 series ThinkPad laptops.
I don't know about the xx40 series of laptops. I think the xx50 series and onwards has some issues.

AkechiShiro, 2020-08-11:
I see so it gets a bit messy for kinda of the xx50 series I suppose (do you know how is it for the newest Lenovo Thinkpads for instance the x395 or the T14, T15? Have they anything much better in terms of battery control driver and ACPI firmware ? I suppose that their EC firmware is I guess less error prone since they're kinda of the latest laptops being sold currently).
Also the source code you mentioned I should have a look at, where should I be able to find it (any ideas) ?

pobrn, 2020-08-11:
Those are the "default" values that you get as far as I know. It seems the battery charge limits are not set. TLP thinks the default for the start threshold is 96%, maybe Lenovo changed it? And it's now 95%? Do the charging threshold work by the way?

AkechiShiro, 2020-08-12:
The charge threshold at least for the **start** percentage works, I haven't tested modifying the start to something lower than 95% or changing the **stop** percentage to something lower as well. 
On Windows it seems that the default start threshold is indeed 95%. For TLP it seems it is 96%.
Even if it seems the battery charge limits are not set. It feels like they are. If I were to reboot into Windows and check in Lenovo Vantage I would find that the charge threshold is set.
Every time I physically unplug the battery and replug it into the computer this settings re-enables itself automatically (even after I have downgraded my EC firmware thinking the setting was bugged).
(PS: I tested it the charge threshold only on Windows, not on Linux but I suppose it should have the same behavior since this feature is OS independent)