linrunner/TLP #521: Default governor for intel_pstate driver changed in kernel 5.7

micha030201, 2020-08-17:
The TLP config file, as well as the corresponding bit in the docs (https://github.com/linrunner/tlp-doc/blob/988e228b3e385a622ac0f0cb255b8d41ce2b6f93/settings/processor.rst) both claim that `powersave` is the default kernel governor when the `intel_pstate` driver is used:

https://github.com/linrunner/TLP/blob/bd068be60fe286e60b009a22e9cfaf4a43c9ed06/tlp.conf#L62-L74

However, as of Linux 5.7, this is no longer the case: https://www.phoronix.com/scan.php?page=news_item&px=Linux-5.7-Schedutil-P-State

The default governor now seems to be `schedutil`, with `intel_pstate` running in passive mode (https://www.kernel.org/doc/html/v5.8/admin-guide/pm/intel_pstate.html#passive-mode -- note that these docs are also outdated, apparently), as is the case on my fresh Arch install:

```
[micha@micha-laptop-hp1 ~]$ cat /sys/devices/system/cpu/intel_pstate/status 
passive
[micha@micha-laptop-hp1 ~]$ cat /sys/devices/system/cpu/cpu*/cpufreq/scaling_governor
schedutil
schedutil
schedutil
schedutil
```

Note that the available governors also seem to have changed, with `powersave` getting ditched. Here is my `tlp-stat -p` output:

```
--- TLP 1.3.1 --------------------------------------------

+++ Processor
CPU model      = Intel(R) Pentium(R) Silver N5000 CPU @ 1.10GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_cpufreq
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = schedutil
/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors = performance schedutil 
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   800000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  2700000 [kHz]

[... repeated 4 times for 4 cores ...]

/sys/devices/system/cpu/intel_pstate/min_perf_pct      =  29 [%]
/sys/devices/system/cpu/intel_pstate/max_perf_pct      = 100 [%]
/sys/devices/system/cpu/intel_pstate/no_turbo          =   0
/sys/devices/system/cpu/intel_pstate/turbo_pct         =  81 [%]
/sys/devices/system/cpu/intel_pstate/num_pstates       =  20

Intel EPB: unsupported CPU.

/sys/module/workqueue/parameters/power_efficient       = Y
/proc/sys/kernel/nmi_watchdog                          = 0
```

I haven't been able to make sure these are all stock kernel changes, and not something introduced by Arch maintainers, but I believe that to be the case, as I doubt they would have played with these settings. I have not been able to find up-to-date documentation about this, as both the kernel documentation and Arch wiki appear to be outdated.

This is, of course, purely a documentation issue -- TLP doesn't change the governor by default; but it may cause confusion for users: "here it says `powersave` is the default and what I should be using, but `schedutil` is instead enabled on my system for some reason," or something like that.

I'm not making this a pull request because I'm not 100% on how it should be worded, whether there should be a note about it being changed in 5.7, etc.

linrunner, 2020-08-17:
Hi, thanks for the headup. It gets even more complicated when reading the kernel docs:

> Passive Mode
> This is the default operation mode of intel_pstate f**or processors without hardware-managed P-states** (HWP) support.

So it depends on the CPU. For example my i7-8550U with 5.4 uses active mode.

I'll have to think that through, probably scrap the recommendation too ...

Btw: many users do not even realize that TLP does not set the governor by default :(.

micha030201, 2020-08-17:
> So it depends on the CPU. For example my i7-8550U with 5.4 uses active mode.

Ah, that makes sense. Hardware-managed P-states support is what Intel calls "Intel® Speed Shift Technology" on their website, apparently. I take back the bit about kernel docs being outdated 😅

Indeed, reading the notes for the [commit](https://lore.kernel.org/lkml/CAJZ5v0ji9p4_whgcJbh6mm8cdYpruHEzOsTqje7JedD45wH5Dg@mail.gmail.com/) linked in the article, it would seem only the default governor for `intel_pstate`'s passive mode got changed:
```
 - Choose schedutil as the preferred scaling governor by default on
   ARM big.LITTLE systems and on x86 systems using the intel_pstate
   driver in the passive mode (Linus Walleij, Rafael Wysocki).
```

So the default for processors with HWP support is probably still `powersave`.

Thank you for caring so much about documentation, by the way, it's really appreciated :)

linrunner, 2020-10-25:
Done for tlp.conf and docs, see commits.

closed 2020-10-25 by linrunner

linrunner, 2020-12-22:
Added some more documentation.

micha030201, 2020-12-22:
Thank you so much for all the care you're putting into this!