linrunner/TLP #524: no turbo boost in battery mode

fidelisakilan, 2020-08-26:
I have this bug where the turbo boost is not enabled in battery mode but in AC mode it is working properly. 
I have made in the conf file to allow turbo boost in both modes.
CPU: i5 8250u @1.6 ghz and boost clock 3.4 ghz.

AC mode:
`--- TLP 1.3.1 --------------------------------------------

+++ Processor
CPU model      = Intel(R) Core(TM) i5-8250U CPU @ 1.60GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors = performance powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   400000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  3400000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/energy_performance_preference = balance_performance [HWP.EPP]
/sys/devices/system/cpu/cpu0/cpufreq/energy_performance_available_preferences = default performance balance_performance balance_power power 

/sys/devices/system/cpu/cpu1/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu1/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu1/cpufreq/scaling_available_governors = performance powersave
/sys/devices/system/cpu/cpu1/cpufreq/scaling_min_freq  =   400000 [kHz]
/sys/devices/system/cpu/cpu1/cpufreq/scaling_max_freq  =  3400000 [kHz]
/sys/devices/system/cpu/cpu1/cpufreq/energy_performance_preference = balance_performance [HWP.EPP]
/sys/devices/system/cpu/cpu1/cpufreq/energy_performance_available_preferences = default performance balance_performance balance_power power 

/sys/devices/system/cpu/cpu2/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu2/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu2/cpufreq/scaling_available_governors = performance powersave
/sys/devices/system/cpu/cpu2/cpufreq/scaling_min_freq  =   400000 [kHz]
/sys/devices/system/cpu/cpu2/cpufreq/scaling_max_freq  =  3400000 [kHz]
/sys/devices/system/cpu/cpu2/cpufreq/energy_performance_preference = balance_performance [HWP.EPP]
/sys/devices/system/cpu/cpu2/cpufreq/energy_performance_available_preferences = default performance balance_performance balance_power power 

/sys/devices/system/cpu/cpu3/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu3/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu3/cpufreq/scaling_available_governors = performance powersave
/sys/devices/system/cpu/cpu3/cpufreq/scaling_min_freq  =   400000 [kHz]
/sys/devices/system/cpu/cpu3/cpufreq/scaling_max_freq  =  3400000 [kHz]
/sys/devices/system/cpu/cpu3/cpufreq/energy_performance_preference = balance_performance [HWP.EPP]
/sys/devices/system/cpu/cpu3/cpufreq/energy_performance_available_preferences = default performance balance_performance balance_power power 

/sys/devices/system/cpu/cpu4/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu4/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu4/cpufreq/scaling_available_governors = performance powersave
/sys/devices/system/cpu/cpu4/cpufreq/scaling_min_freq  =   400000 [kHz]
/sys/devices/system/cpu/cpu4/cpufreq/scaling_max_freq  =  3400000 [kHz]
/sys/devices/system/cpu/cpu4/cpufreq/energy_performance_preference = balance_performance [HWP.EPP]
/sys/devices/system/cpu/cpu4/cpufreq/energy_performance_available_preferences = default performance balance_performance balance_power power 

/sys/devices/system/cpu/cpu5/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu5/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu5/cpufreq/scaling_available_governors = performance powersave
/sys/devices/system/cpu/cpu5/cpufreq/scaling_min_freq  =   400000 [kHz]
/sys/devices/system/cpu/cpu5/cpufreq/scaling_max_freq  =  3400000 [kHz]
/sys/devices/system/cpu/cpu5/cpufreq/energy_performance_preference = balance_performance [HWP.EPP]
/sys/devices/system/cpu/cpu5/cpufreq/energy_performance_available_preferences = default performance balance_performance balance_power power 

/sys/devices/system/cpu/cpu6/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu6/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu6/cpufreq/scaling_available_governors = performance powersave
/sys/devices/system/cpu/cpu6/cpufreq/scaling_min_freq  =   400000 [kHz]
/sys/devices/system/cpu/cpu6/cpufreq/scaling_max_freq  =  3400000 [kHz]
/sys/devices/system/cpu/cpu6/cpufreq/energy_performance_preference = balance_performance [HWP.EPP]
/sys/devices/system/cpu/cpu6/cpufreq/energy_performance_available_preferences = default performance balance_performance balance_power power 

/sys/devices/system/cpu/cpu7/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu7/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu7/cpufreq/scaling_available_governors = performance powersave
/sys/devices/system/cpu/cpu7/cpufreq/scaling_min_freq  =   400000 [kHz]
[... 72 lines cut ...]
/sys/devices/system/cpu/cpu6/cpufreq/energy_performance_preference = balance_performance [HWP.EPP]
/sys/devices/system/cpu/cpu6/cpufreq/energy_performance_available_preferences = default performance balance_performance balance_power power 

/sys/devices/system/cpu/cpu7/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu7/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu7/cpufreq/scaling_available_governors = performance powersave
/sys/devices/system/cpu/cpu7/cpufreq/scaling_min_freq  =   400000 [kHz]
/sys/devices/system/cpu/cpu7/cpufreq/scaling_max_freq  =  1600000 [kHz]
/sys/devices/system/cpu/cpu7/cpufreq/energy_performance_preference = balance_performance [HWP.EPP]
/sys/devices/system/cpu/cpu7/cpufreq/energy_performance_available_preferences = default performance balance_performance balance_power power 

/sys/devices/system/cpu/intel_pstate/min_perf_pct      =  11 [%]
/sys/devices/system/cpu/intel_pstate/max_perf_pct      = 100 [%]
/sys/devices/system/cpu/intel_pstate/no_turbo          =   1
/sys/devices/system/cpu/intel_pstate/turbo_pct         =  59 [%]
/sys/devices/system/cpu/intel_pstate/num_pstates       =  31

/sys/module/workqueue/parameters/power_efficient       = Y
/proc/sys/kernel/nmi_watchdog                          = 0

`

I am not sure whether it is a tlp issue or some other kernel issue
I have got info in reddit that keeping in energy_performance_preference = balance_performance for both modes 
will solve it but even after changing it my pc is stuck below 1.6ghz

linrunner, 2020-08-26:
Hi, without complete outputs as described in the issue template, your issue cannot be processed.

fidelisakilan, 2020-09-07:
https://gist.github.com/fidelisakilan/3dfaed1aed3c4d8e120fa542ac5e576f

fidelisakilan, 2020-09-07:
**Describe the bug**
The bug i am experiencing is that in battery mode my cpu clocks are not above 1.6ghz that is my base clock and turbo boost is unavailable but in AC mode turbo boost is working properly.

**Expected behavior**
I would like it to turbo boost in battery aswell because in high load my pc is lagging
**To Reproduce**

Steps to reproduce the unexpected behavior:
1.The problem occurs only in BATTERY mode
2.CPU clocks are limited to 1.6 ghz even when running heavy task

fidelisakilan, 2020-09-07:
With Charging
[image]

Without Charging:
[image]

linrunner, 2020-09-07:
You haven't done the corresponding configuration. So why do you complain?

To allow turbo mode on AC and BAT you need to configure:

    CPU_BOOST_ON_AC=1
    CPU_BOOST_ON_BAT=0

For your reference: https://linrunner.de/tlp/settings/processor.html#cpu-boost-on-ac-bat

closed 2020-09-07 by linrunner

fidelisakilan, 2020-09-07:
Even after enabling it, the problem persists
https://gist.github.com/fidelisakilan/a143b162c41bcad319a094b99d3b4be4

linrunner, 2020-09-07:
Please enable trace mode by configuring:

     TLP_DEBUG="pm run"

Then on battery power invoke the command

    sudo tlp start

and provide the output of

    sudo tlp-stat -T

fidelisakilan, 2020-09-07:
`-- Logs begin at Wed 2020-08-05 09:06:58 IST, end at Mon 2020-09-07 12:02:29 IST. --
Sep 07 08:27:58 akilan-Aspire-A515-51 tlp[1872]: Applying power save settings...done.
Sep 07 08:27:58 akilan-Aspire-A515-51 tlp[1872]: Setting battery charge thresholds...done.
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: +++ start (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: clear_manual_mode
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: power_source=bat
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: manual_mode=none
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: power_mode=bat
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_laptopmode(1): 2; rc=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_dirty_parms(1): 6000; ec=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_cpu_scaling_governor(1).not_configured
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_cpu_scaling_min_max_freq(1).not_configured
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_intel_cpu_perf_policy(1).hwp_epp: balance_performance; ec=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_intel_cpu_perf_pct(1).min.not_configured
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_intel_cpu_perf_pct(1).max.not_configured
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_cpu_boost_all(1).intel_pstate.cpu_not_supported
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_sched_powersave(1).not_available
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_nmi_watchdog: 0; rc=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_phc_controls.not_configured
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_sata_link_power(1).med_power_with_dipm: /sys/class/scsi_host/host0; rc=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_sata_link_power(1).med_power_with_dipm: /sys/class/scsi_host/host1; rc=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_sata_link_power(1).med_power_with_dipm: /sys/class/scsi_host/host2; rc=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_ahci_runtime_pm(1).not_configured
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_pcie_aspm(1).disabled_by_kernel
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: check_intel_gpu.present: parm=/sys/module/i915/parameters; drm=/sys/class/drm/card0; dbg=/sys/kernel/debug/dri/0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_intel_gpu_min_max_boost_freq(1).not_configured
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_radeon_profile(1).no_card
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_wifi_power_mode(1, wlp2s0).iw: on; rc=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: disable_wake_on_lan: enp1s0f1; rc=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_sound_power_mode(1).hda: 1; rc=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_sound_power_mode(1).hda_controller: Y; rc=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:00.0 [0x060000]; rc=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:02.0 [0x030000]; rc=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:14.0 [0x0c0330]; rc=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:15.0 [0x118000]; rc=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:15.1 [0x118000]; rc=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:16.0 [0x078000]; rc=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:17.0 [0x010601]; rc=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:1c.0 [0x060400]; rc=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:1c.5 [0x060400]; rc=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:1f.0 [0x060100]; rc=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:1f.2 [0x058000]; rc=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:1f.3 [0x040380]; rc=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:00:1f.4 [0x0c0500]; rc=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:01:00.0 [0xff0000]; rc=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:01:00.1 [0x020000]; rc=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: set_runtime_pm(1).auto: /sys/bus/pci/devices/0000:02:00.0 [0x028000]; rc=0
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: poweroff_drivebay(1).disabled
Sep 07 12:02:23 akilan-Aspire-A515-51 tlp[29938]: save_runconf.ok: /run/tlp/tlp-run.conf_tmprIAuXX -> /run/tlp/run.conf
`

linrunner, 2020-09-07:
> kilan-Aspire-A515-51 tlp[29938]: set_cpu_boost_all(1).intel_pstate.cpu_not_supported

Means that the kernel denies write access to **/sys/devices/system/cpu/intel_pstate/no_turbo**. This is either a kernel issue or the BIOS tells the kernel to not use turbo mode on BAT. 

The problem cannot be solved in TLP.