linrunner/TLP #526: TLP does not switch to AC mode on Dell XPS 15 9500

diesieben07, 2020-08-29:
On the new Dell XPS 15 9500 (2020 model) TLP does not switch to AC mode automatically.

**Expected behavior**

TLP should switch to AC mode automatically.

**To Reproduce**

Connect the charger, observe `tlp start` reporting "TLP started in battery mode (auto)".

1. Does the problem occur on battery or AC or both?
AC.
2. Actions to reproduce the behaviour
See above.
3. Shell commands entered and their output
`tlp-stat --psup` with charger connected: https://gist.github.com/diesieben07/dcf0f7b48c3803973c8586a26f50ce8b#file-psup-txt
`tlp-stat --pev` when disconnecting the charger: https://gist.github.com/diesieben07/dcf0f7b48c3803973c8586a26f50ce8b#file-pev-disconnect-txt
`tlp-stat --pev` when connecting the charger: https://gist.github.com/diesieben07/dcf0f7b48c3803973c8586a26f50ce8b#file-pev-connect-txt
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.**
`tlp-stat` with charger connected: https://gist.github.com/diesieben07/dcf0f7b48c3803973c8586a26f50ce8b#file-stat-txt

**Additional context**

Operating System is Linux Mint 20. The problem occurs with the default charger as well as a docking station (output above shows the normal charger).
The OS itself (i.e. the battery icon in the status bar) detects the AC status fine, it shows the battery as charging.

linrunner, 2020-08-29:
LOL. This is exactly the behaviour that Dell intended and configured in their factory image:

    /etc/tlp.d/50-estar-default.conf L0038: TLP_PERSISTENT_DEFAULT="1"
    /etc/tlp.d/50-estar-default.conf L0032: TLP_DEFAULT_MODE="BAT"

Dell does this to achieve the Energy Star 7 certification.

If you really want to change it, you may modify **50-estar-default.conf** directly or create another file to override:

**99-dell-override.conf**

    TLP_PERSISTENT_DEFAULT="0"

diesieben07, 2020-08-29:
Aha! Thank you for that information. I am guessing by "factory image" you mean the linux image they have for the XPS 13?
Interesting choice on their part to just restrict the system's performance by default...

I wonder though how this configuration (decided on by Dell?) ended up in the default TLP installation. I didn't install any Dell factory image, this is just a fresh Linux Mint installation from USB, which installed TLP from the ppa.

Your solution worked perfectly.

linrunner, 2020-08-29:
> I wonder though how this configuration (decided on by Dell?) ended up in the default TLP installation.

Wrong. It's not in the default TLP installation package. There's a Dell specific package in Canonicals OEM repo containing that file. Guess who installed it?

diesieben07, 2020-08-29:
Aha! So it's Linux Mint's fault then? I'm just getting schooled today aren't I? :smile:

closed 2020-10-25 by linrunner