linrunner/TLP #527: Battery status shown as "charging" when charge threshold reached

ivdimitro, 2020-09-03:
Just to clarify - I am not sure if this bug or feature request. 

**Describe the bug**

When battery charge threshold value is set the charging of the battery stops as expected. The issue is that the battery widget of Plasma shows the battery as "charging" indefinitely. 

**Expected behavior**

The battery is considered as charged when the threshold value is reached. 

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Configure charging thresholds for the battery. 
2. Discharge the battery to the min level
3. Charge the battery. 

**Additional context**

>  $ sudo tlp-stat
> [sudo] password for idimitro: 
> --- TLP 1.3.1 --------------------------------------------
> 
> >+++ Configured Settings:
> defaults.conf L0004: TLP_ENABLE="1"
> defaults.conf L0005: TLP_PERSISTENT_DEFAULT="0"
> defaults.conf L0006: DISK_IDLE_SECS_ON_AC="0"
> defaults.conf L0007: DISK_IDLE_SECS_ON_BAT="2"
> defaults.conf L0008: MAX_LOST_WORK_SECS_ON_AC="15"
> defaults.conf L0009: MAX_LOST_WORK_SECS_ON_BAT="60"
> defaults.conf L0010: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
> defaults.conf L0011: CPU_ENERGY_PERF_POLICY_ON_BAT="balance_power"
> defaults.conf L0012: SCHED_POWERSAVE_ON_AC="0"
> defaults.conf L0013: SCHED_POWERSAVE_ON_BAT="1"
> defaults.conf L0014: NMI_WATCHDOG="0"
> defaults.conf L0015: DISK_DEVICES="nvme0n1 sda"
> defaults.conf L0016: DISK_APM_LEVEL_ON_AC="254 254"
> defaults.conf L0017: DISK_APM_LEVEL_ON_BAT="128 128"
> defaults.conf L0018: DISK_IOSCHED="keep keep"
> defaults.conf L0019: SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
> defaults.conf L0020: SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
> defaults.conf L0021: AHCI_RUNTIME_PM_TIMEOUT="15"
> defaults.conf L0022: PCIE_ASPM_ON_AC="default"
> defaults.conf L0023: PCIE_ASPM_ON_BAT="default"
> defaults.conf L0024: RADEON_POWER_PROFILE_ON_AC="default"
> defaults.conf L0025: RADEON_POWER_PROFILE_ON_BAT="default"
> defaults.conf L0026: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
> defaults.conf L0027: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
> defaults.conf L0028: WIFI_PWR_ON_AC="off"
> /etc/tlp.conf L0265: WIFI_PWR_ON_BAT="off"
> defaults.conf L0030: WOL_DISABLE="Y"
> defaults.conf L0031: SOUND_POWER_SAVE_ON_AC="0"
> /etc/tlp.conf L0277: SOUND_POWER_SAVE_ON_BAT="0"
> /etc/tlp.conf L0283: SOUND_POWER_SAVE_CONTROLLER="N"
> defaults.conf L0034: BAY_POWEROFF_ON_AC="0"
> defaults.conf L0035: BAY_POWEROFF_ON_BAT="0"
> defaults.conf L0036: BAY_DEVICE="sr0"
> defaults.conf L0037: RUNTIME_PM_ON_AC="on"
> /etc/tlp.conf L0303: RUNTIME_PM_ON_BAT="on"
> defaults.conf L0039: RUNTIME_PM_DRIVER_BLACKLIST="amdgpu mei_me nouveau nvidia pcieport radeon"
> /etc/tlp.d/10-corpoBuntu.conf L0002: USB_AUTOSUSPEND="0"
> /etc/tlp.conf L0336: USB_BLACKLIST_BTUSB="1"
> /etc/tlp.conf L0342: USB_BLACKLIST_PHONE="1"
> defaults.conf L0043: USB_BLACKLIST_PRINTER="1"
> defaults.conf L0044: USB_BLACKLIST_WWAN="0"
> defaults.conf L0045: USB_AUTOSUSPEND_DISABLE_ON_SHUTDOWN="0"
> defaults.conf L0046: RESTORE_DEVICE_STATE_ON_STARTUP="0"
> /etc/tlp.d/10-corpoBuntu.conf L0019: RESTORE_THRESHOLDS_ON_BAT="1"
> defaults.conf L0048: NATACPI_ENABLE="1"
[... 246 lines cut ...]
> tpacpi-bat.BAT0.forceDischarge                              =      0
> 
> Charge                                                      =   56.3 [%]
> Capacity                                                    =   88.7 [%]
> 
> +++ ThinkPad Battery Status: BAT1 (Ultrabay / Slice / Replaceable)
> /sys/class/power_supply/BAT1/manufacturer                   = LGC
> /sys/class/power_supply/BAT1/model_name                     = 01AV490
> /sys/class/power_supply/BAT1/cycle_count                    =     43
> /sys/class/power_supply/BAT1/energy_full_design             =  23940 [mWh]
> /sys/class/power_supply/BAT1/energy_full                    =  21950 [mWh]
> /sys/class/power_supply/BAT1/energy_now                     =  10750 [mWh]
> /sys/class/power_supply/BAT1/power_now                      =      0 [mW]
> /sys/class/power_supply/BAT1/status                         = Unknown (threshold may prevent charging)
> 
> /sys/class/power_supply/BAT1/charge_start_threshold         =     40 [%]
> /sys/class/power_supply/BAT1/charge_stop_threshold          =     50 [%]
> tpacpi-bat.BAT1.forceDischarge                              =      0
> 
> Charge                                                      =   49.0 [%]
> Capacity                                                    =   91.7 [%]
> 
> +++ Charge total                                            =   52.6 [%]

[image]

linrunner, 2020-09-03:
>  The issue is that the **battery widget of Plasma** shows the battery as "charging" indefinitely.

> /sys/class/power_supply/BAT0/status = Unknown (**threshold may prevent charging**)
> /sys/class/power_supply/BAT1/status = Unknown (**threshold may prevent charging**)

Hi, 

in fact tlp-stat shows the status you expect (the kernel says "unknown" instead of "idle" for this), the Plasma widget doesn't show the status you expect. The [FAQ](https://linrunner.de/tlp/faq/battery.html#why-does-the-panel-applet-show-the-battery-state-charging-despite-charge-thresholds-are-effective) tells the details.

So why did you file your issue _here_? 

Good luck.

closed 2020-09-03 by linrunner

ivdimitro, 2020-09-15:
Hi, I opened the case because my expectation for the functionality was that - when threshold reached battery marked as "charged". 
Thanks for the clarification. I will check with the KDE widget.