linrunner/TLP #528: [ASUS laptops] Implementation of charge end threshold

linrunner, 2020-09-03:
Hi, 

for a proper implementation in TLP i need answers to the following questions: 

1. What are the valid values / the range for **charge_control_end_threshold**?
2. May i assume that the threshold works exactly the same way – values, range, charge behaviour – for all ASUS laptops that expose `/sys/class/power_supply/BAT0/charge_control_end_threshold`? 
2a. If not, how do I distinguish the variants e.g. in **/sys/class/dmi/id/***
3. How do i check that **asus_wmi** is loaded – does **/sys/module/asus_wmi** exist?
4. Is the battery always named **BATx** or else?

Reference: https://gist.github.com/klausenbusk/643f15320ae8997427155c38be13e445

linrunner, 2020-09-03:
Answer from @klausenbusk

> What are the valid values / the range for **charge_control_end_threshold**?

0-100, but there is [one report](https://gist.github.com/klausenbusk/643f15320ae8997427155c38be13e445#gistcomment-3183736) of it only supporting `60, 80 and 100`.

> May i assume that the threshold works exactly the same way – values, range, charge behaviour – for all ASUS laptops that expose `/sys/class/power_supply/BAT0/charge_control_end_threshold`?

I don't know, but probably (if you ignore that one report).

> How do i check that **asus_wmi** is loaded – does **/sys/module/asus_wmi** exist?

`/sys/module/asus_wmi` does indeed exists.

> Is the battery always named **BATx** or else?

[The code](https://github.com/torvalds/linux/blob/master/drivers/platform/x86/asus-wmi.c#L443-L445) is currently hard-coded to only create `charge_control_end_threshold` if the battery name is `BAT0`, `BAT1` or `BATT`.

linrunner, 2020-09-03:
> Answer from @klausenbusk
> 0-100

0 doesn't make sense as stop threshold, it would mean to never charge the battery. On ThinkPads 0 = factory default = 100.

Could you check what really happens when you write a 0?

linrunner, 2020-09-03:
Can you add the output of

    grep . /sys/class/dmi/id/*

for your laptop. **Don't** invoke as root!

klausenbusk, 2020-09-03:
> Could you check what really happens when you write a 0?

I do not have time to check right now, but I think it disable charging completely.

> Can you add the output of

```
/sys/class/dmi/id/bios_date:04/17/2019
/sys/class/dmi/id/bios_release:5.12
/sys/class/dmi/id/bios_vendor:American Megatrends Inc.
/sys/class/dmi/id/bios_version:UX430UNR.308
/sys/class/dmi/id/board_asset_tag:ATN12345678901234567
/sys/class/dmi/id/board_name:UX430UNR
/sys/class/dmi/id/board_vendor:ASUSTeK COMPUTER INC.
/sys/class/dmi/id/board_version:1.0       
/sys/class/dmi/id/chassis_asset_tag:No Asset Tag
/sys/class/dmi/id/chassis_type:10
/sys/class/dmi/id/chassis_vendor:ASUSTeK COMPUTER INC.
/sys/class/dmi/id/chassis_version:1.0       
/sys/class/dmi/id/modalias:dmi:bvnAmericanMegatrendsInc.:bvrUX430UNR.308:bd04/17/2019:br5.12:svnASUSTeKCOMPUTERINC.:pnUX430UNR:pvr1.0:rvnASUSTeKCOMPUTERINC.:rnUX430UNR:rvr1.0:cvnASUSTeKCOMPUTERINC.:ct10:cvr1.0:
/sys/class/dmi/id/product_family:ZenBook
/sys/class/dmi/id/product_name:UX430UNR
/sys/class/dmi/id/product_version:1.0       
/sys/class/dmi/id/sys_vendor:ASUSTeK COMPUTER INC.
/sys/class/dmi/id/uevent:MODALIAS=dmi:bvnAmericanMegatrendsInc.:bvrUX430UNR.308:bd04/17/2019:br5.12:svnASUSTeKCOMPUTERINC.:pnUX430UNR:pvr1.0:rvnASUSTeKCOMPUTERINC.:rnUX430UNR:rvr1.0:cvnASUSTeKCOMPUTERINC.:ct10:cvr1.0:
```

linrunner, 2020-09-04:
Thanks.

> > Could you check what really happens when you write a 0?

> I do not have time to check right now, but I think it disable charging completely.

Take your time. An exact answer is important to me.

IgorFurmanov, 2021-01-13:
Hi.
Recently I created the #548 ticket but it was closed as duplicate of this one.

Let me provide some information (hope it will speed-up the development).
Results of "grep . /sys/class/dmi/id/*" command: https://gist.github.com/IgorFurmanov/63f4878352c7ebe0b7c2f98f608792e7

Also I tried to solve the issue by using following script:
https://gist.github.com/IgorFurmanov/b2ebf41434a7f38002f6ed86bb5621d5

I setup cron to thigger this script every 5 minutes and it set values correctly to the 'charge_control_end_threshold' file.

But there still an issue.
When I set 'charge_control_end_threshold' to value below current charge level then it's ignored. And charging continued up to 100%.

So, answer to your question "Could you check what really happens when you write a 0?" is:
charging just ignores this value as invalid and fallback to 100% value.
Maybe it has been improved in new cores (I'm using 5.8 now)

linrunner, 2021-01-14:
> When I set 'charge_control_end_threshold' to value below current charge level then it's ignored.

What happens when you discharge the battery below the the previously written `charge_control_end_threshold`? Does the threshold then take effect?

klausenbusk, 2021-01-14:
>  When I set 'charge_control_end_threshold' to value below current charge level then it's ignored. And charging continued up to 100%.

I don't see that behavior on my system (ASUS Zenbook UX430UNR). I think some of the behavior is firmware dependent and is different between laptop models.

linrunner, 2021-04-20:
The wait is over: the stop charge threshold is implemented for 1.4. 

Please participate in the alpha test. All necessary information is [here](https://gist.github.com/linrunner/2ead4b591eed33055cf86a38ccc73949).

Have fun.

123unix, 2021-06-11:
> Answer from @klausenbusk
> 
> > What are the valid values / the range for **charge_control_end_threshold**?
> 
> 0-100, but there is [one report](https://gist.github.com/klausenbusk/643f15320ae8997427155c38be13e445#gistcomment-3183736) of it only supporting `60, 80 and 100`.

I confirm this limited list of valid values.

Initially when testing my new laptop I was rushing things out and was setting thresholds just a couple of percentage points ahead of the battery's current capacity. Unsurprisingly those happened to fall outside this small set, which frustrated me a lot into believing my new laptop doesn't support the feature.

Now seeing this post I decided to give it one last shot and just put 60 to    charge_control_end_threshold and patiently waited for the charge to climb there from 40-something - and voila it stopped there!

linrunner, 2021-06-11:
Thanks for reporting. For the record (and so that i understand it correctly), i would like to have some outputs:

    sudo tlp setcharge 0 65
    sudo tlp-stat -s -b --cdiff

123unix, 2021-06-11:
# tlp setcharge 0 65
Error: battery charge thresholds not available.
# tlp-stat -s -b --cdiff
Usage: tlp-stat [ -b | --battery   | -c | --config    |
                  -d | --disk      | -e | --pcie      |
                  -g | --graphics  | -p | --processor |
                  -r | --rfkill    | -s | --system    |
                  -t | --temp      | -u | --usb       |
                  -w | --warn      | -v | --verbose   |
                  -P | --pev       |    | --psup      |
                  -T | --trace ]
# tlp-stat -s -b 
--- TLP 1.3.1 --------------------------------------------

+++ System Info
System         = ASUSTeK COMPUTER INC. 1.0 ZenBook UX425JA_UX425JA
BIOS           = UX425JA.301
Release        = Ubuntu 20.10
Kernel         = 5.8.0-53-generic #60-Ubuntu SMP Thu May 6 07:46:32 UTC 2021 x86_64
/proc/cmdline  = BOOT_IMAGE=/BOOT/ubuntu_7d6gbz@/vmlinuz-5.8.0-53-generic root=ZFS=rpool/ROOT/ubuntu_7d6gbz ro quiet splash vt.color=0xF0 vt.cur_default=112 vt.handoff=1
Init system    = systemd v246 (246.6-1ubuntu1)
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = unknown
Mode           = unknown
Power source   = AC

+++ Battery Features: Charge Thresholds and Recalibrate
natacpi    = inactive (laptop not supported)
tpacpi-bat = inactive (laptop not supported)
tp-smapi   = inactive (laptop not supported)

+++ Battery Status: BAT0
/sys/class/power_supply/BAT0/manufacturer                   = ASUSTeK
/sys/class/power_supply/BAT0/model_name                     = UX425
/sys/class/power_supply/BAT0/cycle_count                    =     46
/sys/class/power_supply/BAT0/charge_full_design             =   4220 [mAh]
/sys/class/power_supply/BAT0/charge_full                    =   4172 [mAh]
/sys/class/power_supply/BAT0/charge_now                     =   2700 [mAh]
/sys/class/power_supply/BAT0/current_now                    =      0 [mA]
/sys/class/power_supply/BAT0/status                         = Unknown

Charge                                                      =   64.7 [%]
Capacity                                                    =   98.9 [%]

linrunner, 2021-06-11:
You probably missed my info about the 1.4 alpha test above.

Please install TLP 1.4 ([Download](https://download.linrunner.de/packages/)) and show again.

123unix, 2021-06-11:
> You probably missed my info about the 1.4 alpha test above.
> 
> Please install TLP 1.4 ([Download](https://download.linrunner.de/packages/)) and show again.
Sorry, no time at the moment to participate in non-repository software exploration...

linrunner, 2021-06-11:
Very unfortunate.

123unix, 2021-06-15:
Well, I just wanted to share the joy of discovering a workaround for a broken functionality.

linrunner, 2021-06-16:
If you are running Debian/Ubuntu or Arch: it takes no more than a minute to download the packages, install them and show the outputs. Give yourself a push -- and the joy of having improved test coverage and documentation.

mthw0, 2021-07-27:
I just tested this on an Asus TUF laptop and setting battery charge limit works even with custom values (tested with 65). I used tlp-git from AUR. The only issue I have is setting the limit to 65 resulted in battery charging stopping at 64%, but I also had this issue before.

linrunner, 2021-07-27:
@mthw0 : well, please post the output of 

    tlp setcharge 0 65
    tlp-stat -s -c -b

mthw0, 2021-07-27:
> Notice: some ASUS laptops silently ignore charge thresholds other than 40, 60 or 80. Please check if 65 works as expected.
Setting temporary charge threshold for BAT0:
  stop =  65

and 

https://pastebin.com/H4etdtFw

linrunner, 2021-07-27:
@mthw0 : thanks.

linrunner, 2021-09-24:
[TLP 1.4 is released](https://linrunner.de/tlp/news.html#tlp-1-4-released). Have fun and go on testing please!

klausenbusk, 2021-09-24:
@linrunner: I don't use TLP myself, but excellent work :)

closed 2021-11-10 by linrunner

IvnSoft, 2023-02-18:
Uh, sorry to comment on a closed issue, but wanted to let you know that asus_wmi has undergone changes, and it is doing weird things on some hardware.
On my Asus TUF A15 (ryzen 7) up until kernel 6.0 i could use any step on the charge_end_threshold. After the last update to 6.1.11, its only understanding values from 52 to 80. Any other value, just keeps charging until 100%

linrunner, 2023-02-27:
@IvnSoft in fact there have been no changes to the kernel in this regard recently. The function in question still checks for 0 <= stop_threshold <= 100:
* https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/log/drivers/platform/x86/asus-wmi.c
* https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/drivers/platform/x86/asus-wmi.c#n854

On the other hand, it is well known and mentioned in TLP's command output and docs that some ASUS laptop models ignore values other than 40, 60, 80. 
* https://linrunner.de/tlp/settings/bc-vendors.html#asus
* https://github.com/linrunner/TLP/commit/0b323ee32b39968d28a35e7e9d3c1cf966f3ad5f

Perhaps there are other variations of this scheme out there.