linrunner/TLP #529: Implement Intel power limit configuration options

validatedev, 2020-09-12:
**Is your feature request related to a problem? Please describe.**

There isn't any proper program which had implemented both AC and BAT profiles. I think setting the power limits for CPU like there https://github.com/georgewhewell/undervolt is a good extension for saving power / maintaining performance-power balance. balance-power EPP setting causing much slowness on my PC.

**Describe the solution you'd like**

Implement the configuration for Intel power limit options something like:

```
AC_POWER_LIMIT_LONG_ENABLED=X
AC_POWER_LIMIT_SHORT_ENABLED=X
AC_POWER_LIMIT_LONG_DURATION=X
AC_POWER_LIMIT_SHORT_DURATION=X
AC_POWER_LIMIT_LOCK=X

AC_POWER_LIMIT_LONG_ENABLED=X
AC_POWER_LIMIT_SHORT_ENABLED=X
AC_POWER_LIMIT_LONG_DURATION=X
AC_POWER_LIMIT_SHORT_DURATION=X
AC_POWER_LIMIT_LOCK=X

```

**Describe alternatives you've considered**

There's the program called https://github.com/georgewhewell/undervolt, it works but lacks of options like BAT and AC profile seperate.

**Additional context**

http://manpages.ubuntu.com/manpages/cosmic/man1/rapl-set.1.html
https://01.org/blogs/2014/running-average-power-limit-%E2%80%93-rapl

**Note**

If you're interested, I can implement and PR, but right now I haven't much time to do it.

linrunner, 2020-09-13:
Hi,

there are laptops that need to refresh the setting periodically – ThinkPads for example. So do you think this works without a daemon? Because TLP does not have one ...

When i look at the impenetrable documentation about the configuration of the thermald, i also doubt that it is possible to keep the configuration so simple that the majority of users can understand it. Otherwise such a feature becomes a nightmare in support, which i don't want to do to myself.

validatedev, 2020-09-14:
Hi, I'm very busy recently. I'll read the related documentations and get back there, don't close the issue please.

linrunner, 2020-09-15:
No problem.

validatedev, 2020-11-09:
Sorry about that, I really haven’t had any time to look at it. Thus, closing the issue until I have a time to do something about it. Sorry again.

closed 2020-11-09 by validatedev