linrunner/TLP #531: ThinkPad E14 hangs with TLP

mmiat, 2020-09-15:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs (see below)

**Describe the bug**

When I install tlp my ThinkPad E14 hangs
I can't run tlp-stat when in battery mode

**Expected behavior**

the it runs and doesn't hang notebook

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both? only on battery
2. Actions to reproduce the behaviour
3. Shell commands entered and their output: apt install tlp, sudo service tlp start
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.**

**Additional context**

Linux Mint 20, fresh install
I've also tried as suggested here: https://github.com/linrunner/TLP/issues/186

linrunner, 2020-09-15:
Hi,

since the actual hardware is needed to find the cause, only you can do it.

The general procedure is laid out [here](https://linrunner.de/tlp/support/troubleshooting.html).

Focus on settings where the default for AC and BAT is different. Good luck.

mmiat, 2020-09-15:
I'll try
for now I've tried:
- to remove only tlp-rdw
- attach AC after freeze
with no success

linrunner, 2020-09-16:
It is very tedious, i know. Unfortunately, there is no other way.

linrunner, 2020-11-09:
Feel free to reopen when you have results.

closed 2020-11-09 by linrunner

mmiat, 2020-11-09:
solved with last kernel (5.8+)

linrunner, 2020-11-09:
Thanks for your feedback.