linrunner/TLP #532: USB_BLACKLIST does not work, USB device is still put to auto power management

marcmerlin, 2020-09-20:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs (see below)

**Describe the bug**

tlp 1.3.1-2 on debian, when I unplug power, it puts most devices to auto sleep, as expected.
One of them causes problems, so I tried to exclude it

sauron:~# lsusb | grep -i yubi
Bus 001 Device 002: ID 1050:0200 Yubico.com Gnubby U2F

sauron:~# grep ^USB_ /etc/tlp.conf
USB_AUTOSUSPEND=1
USB_BLACKLIST="1050:0200"
USB_BLACKLIST_BTUSB=0
USB_BLACKLIST_PHONE=0
USB_BLACKLIST_WWAN=0

and yet it gets suspended anyway. I have to manually auto suspend after unplugging power:
sauron:~# echo on > "$(grep Yubico /sys/bus/usb/devices/*/manufacturer | sed 's/manufacturer.*//')power/control"

before and after I run my command
sauron:~# diff -u /tmp/tlp-stat.bat /tmp/tlp-stat.bat2
-Bus 001 Device 002 ID 1050:0200 control = auto, autosuspend_delay_ms = 1000 -- Yubico.com Gnubby U2F (usbhid)
+Bus 001 Device 002 ID 1050:0200 control = on,   autosuspend_delay_ms = 1000 -- Yubico.com Gnubby U2F (usbhid)

[tlp-stat.bat.txt](https://github.com/linrunner/TLP/files/5250987/tlp-stat.bat.txt)
[tlp-stat.bat2.txt](https://github.com/linrunner/TLP/files/5250988/tlp-stat.bat2.txt)
[tlp-stat.power.txt](https://github.com/linrunner/TLP/files/5250989/tlp-stat.power.txt)

linrunner, 2020-09-20:
Hi, 

a Yubikey is an usbhid device which gets excluded from autosuspend by default. So there should be no need for USB_BLACKLIST. Btw, my Yubikeys work wonderfully, even if I whitelist them. From what you write in your other issue i'd suspect your quirky hardware/drivers is causing this, but lets see.

Please enable trace mode, run the command:

     sudo tlp usb

and show the trace output via https://gist.github.com/ .

ps. i would like to point out that gist output is *required*, file attachments are not desired.

marcmerlin, 2020-09-20:
If I unplug power, tlp runs, and powertop, shows
"   Good          Autosuspend for USB device Yubico Gnubby (gnubby1) [Yubico]"
(although good, is actually bad in my case, due to the kernel bug I seem to have)

I then ran tlp usb like you suggested, and it changed to:
   Bad           Autosuspend for USB device Yubico Gnubby (gnubby1) [Yubico]

Ok, so that seems to show that tlp is doing the right thing, but that maybe something else on my system (not tlp) is putting the yubikey in autosuspend mode?

I did the test again with running "tlp start"  after the system went on batteries
TLP started in battery mode (auto).
   Bad           Autosuspend for USB device Yubico Gnubby (gnubby1) [Yubico]

That really seems to confirm that tlp is doing the right thing, but something else on my system is putting the device to suspend after tlp. I spent more time debugging and found an old leftover acpid script that did that, written may years back, before I had tlp and was still using laptop-mode.

My apologies for missing that and taking your time, thanks for giving me the 'tlp usb' hint that allowed me to find the problem, and I'll resolve this bug.
(also, now I see why you prefer gist to attachment, fully noted, thanks)

closed 2020-09-20 by marcmerlin

linrunner, 2020-09-20:
> Found an old leftover acpid script that did that

No problem. This shows the disadvantage of hand carved scripts compared to a tool installed as a package. When i started TLP development 10 years ago, all people employed such scripts and i was tired to help them in the german Thinkpad forum.