linrunner/TLP #533: Thinkpad P73: PME enabled/disabled loop with USB-C/thunberbolt + nvidia/nouveau + Cannon Lake PCH USB 3.1 xHCI Host Controller

marcmerlin, 2020-09-20:
I spent an hour debugging this, on my thinkpad, this is not a TLP problem, but TLP triggered it, so I thought I'd file this issue to describe the issue and fix/workaround I applied.

Kernel is 5.8. Things have changed with PME and/or nouveau since 5.5 and I have issues at boot with PME and hangs 2-3mn at each boot. More details are here if anyone needs them
https://www.mail-archive.com/nouveau@lists.freedesktop.org/msg36134.html

Back to TLP, every time I unplug power, I get kernel spam every second, and if I'm unlucky I also get in some state where 40W is burned on I'm not sure what (goes away after a reboot).
Looks like this:
```
xhci_hcd 0000:00:14.0: PME# enabled
xhci_hcd 0000:00:14.0: power state changed by ACPI to D3hot
xhci_hcd 0000:00:14.0: power state changed by ACPI to D0
xhci_hcd 0000:00:14.0: PME# disabled
xhci_hcd 0000:00:14.0: enabling bus mastering
xhci_hcd 0000:00:14.0: saving config space at offset 0x0 (reading 0xa36d8086)
xhci_hcd 0000:00:14.0: saving config space at offset 0x4 (reading 0x2900402)
xhci_hcd 0000:00:14.0: saving config space at offset 0x8 (reading 0xc033010)
xhci_hcd 0000:00:14.0: saving config space at offset 0xc (reading 0x800000)
xhci_hcd 0000:00:14.0: saving config space at offset 0x10 (reading 0xce420004)
xhci_hcd 0000:00:14.0: saving config space at offset 0x14 (reading 0x0)
xhci_hcd 0000:00:14.0: saving config space at offset 0x18 (reading 0x0)
xhci_hcd 0000:00:14.0: saving config space at offset 0x1c (reading 0x0)
xhci_hcd 0000:00:14.0: saving config space at offset 0x20 (reading 0x0)
xhci_hcd 0000:00:14.0: saving config space at offset 0x24 (reading 0x0)
xhci_hcd 0000:00:14.0: saving config space at offset 0x28 (reading 0x0)
xhci_hcd 0000:00:14.0: saving config space at offset 0x2c (reading 0x229b17aa)
xhci_hcd 0000:00:14.0: saving config space at offset 0x30 (reading 0x0)
xhci_hcd 0000:00:14.0: saving config space at offset 0x34 (reading 0x70)
xhci_hcd 0000:00:14.0: saving config space at offset 0x38 (reading 0x0)
xhci_hcd 0000:00:14.0: saving config space at offset 0x3c (reading 0x1ff)
xhci_hcd 0000:00:14.0: PME# enabled
xhci_hcd 0000:00:14.0: power state changed by ACPI to D3hot
xhci_hcd 0000:00:14.0: power state changed by ACPI to D0
xhci_hcd 0000:00:14.0: PME# disabled
xhci_hcd 0000:00:14.0: enabling bus mastering
```

And it loops forever. I found 2 fixes:
1) disable autosuspend on this
#    Bad           Runtime PM for PCI Device Intel Corporation Cannon Lake PCH USB 3.1 xHCI Host Controller

2) better, disable autosuspend on my USB key plugged in a USB-C/thunderbolt port. I've seen issues between thunderbolt and nouveau on that system. I don't know exactly what they are.
# Bad           Autosuspend for USB device Yubico Gnubby (gnubby1) [Yubico]                                       

Now, I remove autosuspend on the USB device every time I unplug power, and the kernel log loop is gone:
[out.txt](https://github.com/linrunner/TLP/files/5251001/out.txt)

Now, there is currently a separate issue that USB blacklisting doesn't work for me (cf https://github.com/linrunner/TLP/issues/532 ) but I'm just doing this by hand for now;
echo on > "$(grep Yubico /sys/bus/usb/devices/*/manufacturer | sed 's/manufacturer.*//')power/control"

Feel free to close this issue, if there is nothing actionable to add in the documentation/troubleshooting docs, if this is really a weird problem that just impacts thinkpads P53 and P73, and probably only if you use thunderbolt, but either way it'll get indexed by google and help the next person looking for this, if any.

lspci if that helps:
```
00:00.0 Host bridge: Intel Corporation Device 3e20 (rev 0d)
00:01.0 PCI bridge: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor PCIe Controller (x16) (rev 0d)
00:02.0 VGA compatible controller: Intel Corporation UHD Graphics 630 (Mobile) (rev 02)
00:04.0 Signal processing controller: Intel Corporation Xeon E3-1200 v5/E3-1500 v5/6th Gen Core Processor Thermal Subsystem (rev 0d)
00:08.0 System peripheral: Intel Corporation Xeon E3-1200 v5/v6 / E3-1500 v5 / 6th/7th/8th Gen Core Processor Gaussian Mixture Model
00:12.0 Signal processing controller: Intel Corporation Cannon Lake PCH Thermal Controller (rev 10)
00:14.0 USB controller: Intel Corporation Cannon Lake PCH USB 3.1 xHCI Host Controller (rev 10)
00:14.2 RAM memory: Intel Corporation Cannon Lake PCH Shared SRAM (rev 10)
00:15.0 Serial bus controller [0c80]: Intel Corporation Cannon Lake PCH Serial IO I2C Controller #0 (rev 10)
00:15.1 Serial bus controller [0c80]: Intel Corporation Cannon Lake PCH Serial IO I2C Controller #1 (rev 10)
00:16.0 Communication controller: Intel Corporation Cannon Lake PCH HECI Controller (rev 10)
00:17.0 SATA controller: Intel Corporation Cannon Lake Mobile PCH SATA AHCI Controller (rev 10)
00:1b.0 PCI bridge: Intel Corporation Cannon Lake PCH PCI Express Root Port #17 (rev f0)
00:1c.0 PCI bridge: Intel Corporation Cannon Lake PCH PCI Express Root Port #1 (rev f0)
00:1c.5 PCI bridge: Intel Corporation Cannon Lake PCH PCI Express Root Port #6 (rev f0)
00:1c.7 PCI bridge: Intel Corporation Cannon Lake PCH PCI Express Root Port #8 (rev f0)
00:1e.0 Communication controller: Intel Corporation Cannon Lake PCH Serial IO UART Host Controller (rev 10)
00:1f.0 ISA bridge: Intel Corporation Cannon Lake LPC Controller (rev 10)
00:1f.3 Audio device: Intel Corporation Cannon Lake PCH cAVS (rev 10)
00:1f.4 SMBus: Intel Corporation Cannon Lake PCH SMBus Controller (rev 10)
00:1f.5 Serial bus controller [0c80]: Intel Corporation Cannon Lake PCH SPI Controller (rev 10)
00:1f.6 Ethernet controller: Intel Corporation Ethernet Connection (7) I219-LM (rev 10)
01:00.0 VGA compatible controller: NVIDIA Corporation TU104GLM [Quadro RTX 4000 Mobile / Max-Q] (rev a1)
01:00.1 Audio device: NVIDIA Corporation TU104 HD Audio Controller (rev a1)
01:00.2 USB controller: NVIDIA Corporation TU104 USB 3.1 Host Controller (rev a1)
01:00.3 Serial bus controller [0c80]: NVIDIA Corporation TU104 USB Type-C UCSI Controller (rev a1)
02:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller SM981/PM981/PM983
04:00.0 PCI bridge: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge 4C 2018] (rev 06)
05:00.0 PCI bridge: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge 4C 2018] (rev 06)
05:01.0 PCI bridge: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge 4C 2018] (rev 06)
05:02.0 PCI bridge: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge 4C 2018] (rev 06)
05:04.0 PCI bridge: Intel Corporation JHL7540 Thunderbolt 3 Bridge [Titan Ridge 4C 2018] (rev 06)
06:00.0 System peripheral: Intel Corporation JHL7540 Thunderbolt 3 NHI [Titan Ridge 4C 2018] (rev 06)
2c:00.0 USB controller: Intel Corporation JHL7540 Thunderbolt 3 USB Controller [Titan Ridge 4C 2018] (rev 06)
52:00.0 Network controller: Intel Corporation Wi-Fi 6 AX200 (rev 1a)
54:00.0 Unassigned class [ff00]: Realtek Semiconductor Co., Ltd. RTS525A PCI Express Card Reader (rev 01)
```

marcmerlin, 2020-09-20:
tlp-stat for that laptop is already in https://github.com/linrunner/TLP/issues/532  so I won't upload it a second time to this bug.

linrunner, 2020-09-20:
Why are you opening an issue here when you already know it's a kernel bug? 

Btw: your descriptions are extremely difficult to penetrate.

closed 2020-09-20 by linrunner

marcmerlin, 2020-09-20:
Why? I wrote
"Feel free to close this issue, if there is nothing actionable to add in the documentation/troubleshooting docs (...) but either way it'll get indexed by google and help the next person looking for this,"
My apologies, if this was not acceptable. I was only trying to help the next person who might hit this issue or an issue like this.

linrunner, 2020-09-20:
I like to add workarounds to the FAQ, but for this you should please write which device you added to RUNTIME_PM_BLACKLIST.

> Why?
A kernel bug report could bring a causal solution. Alternatively you could report in the lenovo forum. meanwhile there is support for Linux.

marcmerlin, 2020-09-20:
> ou should please write which device you added to RUNTIME_PM_BLACKLIST.

00:14.0 USB controller: Intel Corporation Cannon Lake PCH USB 3.1 xHCI Host Controller (rev 10)
is the device I could have put in PM_BLACKLIST, but I found that making sure the yubikey in the thunderbolt port, didn't get PM was enough to avoid the issue, so I made sure that one wouldn't go to sleep instead.

> Alternatively you could write a report in the Lenovo forum / there is support for Linux.

Indeed. I have 2 threads about this, on linux-kernel, linux-pci and another one on the nouveau list.
Honestly I don't know the linux kernel internals to say what's at fault here, but so few people seem to be using linux on laptops nowadays (and that's coming from someone who's been running linux on laptops since 1996), that the reports are rare, and the fixes are rare too.

As for getting support from lenovo, they don't really care about linux (I did try though). Hell, lenovo does not even allow you to turn off the nvidia chip in the bios, and is now forcing you to get over-powered power supplies (170W or more) for the laptop to even accept the power supply. Took me days of hardware hacking to get around that crap ( http://marc.merlins.org/perso/linux/post_2020-05-04_Hacking-Power-Supplies-and-Battery-Pack-To-Get-Around-ThinkPad-P73-Broken-Power-Supply-Design.html ).  Tlp is actually credited in that blog post, its functionality was instrumental to allowing me to use smaller power supplies with the laptop, so thank you again for your work.

Back to all this, I don't think any action is expected from you, I just really wanted to do my part in trying to help use linux on laptops better, and having this be indexed and searchable on the web, should help.

linrunner, 2020-09-21:
Well. Let's just leave it at that.