linrunner/TLP #534: [Maybe a Bug] When the laptop is on battery power, pressing the arrow keys in some programs will make a very loud Beep sound

whriedplanck, 2020-09-20:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)
[x] I've attached all required `tlp-stat` outputs (see below)

**Describe the bug**

When the laptop is on battery power, pressing the arrow keys/Backspace key in some programs will make a very loud Beep sound. This can be annoying sometimes, because the beep sound cannot be eliminated even if it is muted.
I found [a possibly related page in the FAQ](https://linrunner.de/tlp/settings/audio.html?highlight=sound), but it doesn't seem to solve this issue.

**Expected behavior**

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?
    - Only on battery.
2. Actions to reproduce the behaviour
    These actions are also what I did when I found this issue.
    - Open a program with input boxes (like Poedit)
    - Enter some long text, then long press the arrow keys to move the cursor to the end, or long press the backspace key to completely clear the text
    - If don’t let go, you can hear a loud beep from the computer
3. Shell commands entered and their output
    - `tlp-stat`
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.**
    - `tlp-stat`'s output: https://fars.ee/F4u2
   (Sorry, I cann't use the gist at the moment for some reasons, so I pasted the output on a pastebin site.）

**Additional context**

So far, only programs written with GTK+ have been found to have this issue (Qt programs do not currently exist). I found this issue when I used the Qalculate! and Poedit.

linrunner, 2020-09-20:
I suggest you [disable](https://linrunner.de/tlp/faq/operation.html#how-do-i-stop-or-disable-tlp-completely) or uninstall TLP, reboot and check if the problem persists.

Also refer to the [Troubleshooting Guide](https://linrunner.de/tlp/support/troubleshooting.html).

whriedplanck, 2020-09-20:
> I suggest you [disable](https://linrunner.de/tlp/faq/operation.html#how-do-i-stop-or-disable-tlp-completely) or uninstall TLP, reboot and check if the problem persists.
> 
> Also refer to the [Troubleshooting Guide](https://linrunner.de/tlp/support/troubleshooting.html).

I have uninstalled/disabled TLP. I'm pretty sure this happened after TLP was enabled.

linrunner, 2020-09-20:
Please specify: is the symptom gone after uninstalling?

whriedplanck, 2020-09-21:
> Please specify: is the symptom gone after uninstalling?

Sure. Sorry for my unclear expression.

linrunner, 2020-09-21:
Then you should try to find the causing setting with the help of the [Troubleshooting Guide](https://linrunner.de/tlp/support/troubleshooting.html).

whriedplanck, 2020-09-26:
> Then you should try to find the causing setting with the help of the [Troubleshooting Guide](https://linrunner.de/tlp/support/troubleshooting.html).

OK, I'll give it a try again.

linrunner, 2020-09-26:
I know it is tedious, but there is no alternative

whriedplanck, 2020-09-26:
> I know it is tedious, but there is no alternative

I know, actually I don't mind. I may just too busy to locate the issue ...... If I find something new, I will talk about it here ~ : )

whriedplanck, 2020-10-18:
Recently I found that this issue not only occurrs when I use TLP, but also when I use other power management tools (such as powertop). It seems that these tools, including TLP, have modified a certain setting to cause this issue. But I still don't have enough time to locate the specific setting ......

linrunner, 2020-10-23:
In the time you have spent with other tools, you could have followed the instructions ...

linrunner, 2020-11-09:
Feel free to reopen when you have results.

closed 2020-11-09 by linrunner

whriedplanck, 2021-01-06:
After a long period of debugging, this issue looks more like a `hardware quirk` as you said. Once `TLP` or `powertop` has been installed, this issue will happen randomly...... : / whether they are enabled or not. Even more strange is that this issue will only happen in GTK+ programs ...... I'm so sorry to bother you. Anyway, thanks for helping!

linrunner, 2021-01-06:
No problem. Good luck.

whriedplanck, 2021-01-06:
Thanks! It may be a bit late, but I still wish you a happy new year! :D

linrunner, 2021-01-06:
Thank you, likewise.

WhiredPlanck, 2021-06-15:
Hi, I finally found out why it is happening. It’s said that in a text box of a GTK application, when the cursor arrives at the beginning/end of the text and if you keep pressing arrow key, the GTK application will try to trigger the FD.o sound effect. However, if you doesn’t install the `sound-theme-freedesktop` package, this will  the buzzer sound very loud.

In some distributions, this feature is turned off by default while some distributions, like Arch Linux in my case, turn it on by default. I've turned it off manually in my DE settings. And now everything is fine.

Sorry for disturbing, it’s not the fault of `TLP`.

linrunner, 2021-06-15:
Thanks for your final feedback and for describing the solution.

WhiredPlanck, 2021-06-15:
> Thanks for your final feedback and for describing the solution.

No problem. This is what I'm supposed to do.

Actually I should also add that at the time I reported this issue, the KDE in Arch Linux was upgraded to turn on this sound effect by default. And at the same time, I install `TLP` and `powertop`, so I had mistakenly thought that the latter two were producing this problem until I found out the real cause.

Again, I apologize for my misunderstanding, and thanks for your help! Have a good day！