linrunner/TLP #537: TLP doesnt switch AC/BAT Mode

Bambus89, 2020-10-05:
Hello there,

TLP doesn't change the Mode on (un)plugging. It recognices the correct Power Source but did not change the right Mode.

I'm on openSuSe Tumbleweed and TLP 1.3.1

--- TLP 1.3.1 --------------------------------------------

+++ System Info
System         = ASUSTeK COMPUTER INC. 1.0        UX310UAR
BIOS           = UX310UAR.311
Release        = "openSUSE Tumbleweed"
Kernel         = 5.8.10-1-default #1 SMP Mon Sep 21 11:00:19 UTC 2020 (af3e800) x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-5.8.10-1-default root=/dev/mapper/system-root splash=silent quiet nowatchdog root_trim=yes mitigations=auto
Init system    = systemd 
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 09:43:03,     37 sec(s) ago
Mode           = battery
Power source   = AC

My Config Same results with no config (all default) and this one below:

# ------------------------------------------------------------------------------
# /etc/tlp.conf - TLP user configuration
# See full explanation: https://linrunner.de/en/tlp/docs/tlp-configuration.html
#
# New configuration scheme (TLP 1.3). Settings are read in the following order:

# 1. Intrinsic defaults
# 2. /etc/tlp.d/*.conf - Drop-in customization snippets
# 3. /etc/tlp.conf     - User configuration (this file)
#
# Notes:
# - In case of identical parameters, the last occurence has precedence
# - This also means, parameters enabled here will override anything else
# - IMPORTANT: all parameters here are disabled, remove the leading '#'
#   to enable them; shown values may be suggestions rather than defaults
# - Default *: intrinsic default that is effective when the parameter is missing
#     or disabled by a leading '#'; use PARAM="" to disable an intrinsic default
# - Default <none>: do nothing or use kernel/hardware defaults

# ------------------------------------------------------------------------------
# tlp - Parameters for power saving

# Set to 0 to disable, 1 to enable TLP.
# Default: 1

TLP_ENABLE=1

# Operation mode when no power supply can be detected: AC, BAT.
# Concerns some desktop and embedded hardware only.
# Default: <none>

TLP_DEFAULT_MODE=BAT

# Operation mode select: 0=depend on power source, 1=always use TLP_DEFAULT_MODE
# Note: use in conjunction with TLP_DEFAULT_MODE=BAT for BAT settings on AC.
# Default: 0

TLP_PERSISTENT_DEFAULT=0

# Power supply class to ignore when determining operation mode: AC, BAT.
# Note: try on laptops where operation mode AC/BAT is incorrectly detected.
# Default: <none>

#TLP_PS_IGNORE=BAT
[... 404 lines cut ...]
# Possible devices: bluetooth, wifi, wwan.
# Separate multiple radio devices with spaces.
# Default: <none> (for all parameters below)

# Radio devices to disable on connect.

#DEVICES_TO_DISABLE_ON_LAN_CONNECT="wifi wwan"
#DEVICES_TO_DISABLE_ON_WIFI_CONNECT="wwan"
#DEVICES_TO_DISABLE_ON_WWAN_CONNECT="wifi"

# Radio devices to enable on disconnect.

#DEVICES_TO_ENABLE_ON_LAN_DISCONNECT="wifi wwan"
#DEVICES_TO_ENABLE_ON_WIFI_DISCONNECT=""
#DEVICES_TO_ENABLE_ON_WWAN_DISCONNECT=""

# Radio devices to enable/disable when docked.

#DEVICES_TO_ENABLE_ON_DOCK=""
#DEVICES_TO_DISABLE_ON_DOCK=""

# Radio devices to enable/disable when undocked.

#DEVICES_TO_ENABLE_ON_UNDOCK="wifi"
#DEVICES_TO_DISABLE_ON_UNDOCK=""

linrunner, 2020-10-05:
I suggest you try the [workaround](https://linrunner.de/tlp/faq/operation.html#ac-or-bat-is-not-detected) from the FAQ.

Bambus89, 2020-10-05:
That doesn't change anything. As i wrote tlp reads the corret PowerSource plugged and unplugged. Bit It does not change the Mode equal to the current PowerSource

linrunner, 2020-10-05:
In your case it doesn't matter if tlp-stat shows the correct source, because the mode is changed by events only --> https://linrunner.de/tlp/developers/architecture.html

You'll have to look if udev power supply events are coming. Start the output with

    tlp-stat --pev

and show the event output for

1. AC unplugged
2. AC plugged

And while we are at it, also show

    tlp-stat -b

Bambus89, 2020-10-05:
First thank you for your help ;)

```
tlp-stat --pev
--- TLP 1.3.1 --------------------------------------------

+++ Monitor power supply events -- cancel with ^C

monitor will print the received events for:
UDEV - the event which udev sends out after rule processing

UDEV  [16806.231025] change   /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:19/PNP0C09:01/ACPI0003:00/power_supply/AC0 (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:19/PNP0C09:01/ACPI0003:00/power_supply/AC0
SUBSYSTEM=power_supply
POWER_SUPPLY_NAME=AC0
POWER_SUPPLY_TYPE=Mains
POWER_SUPPLY_ONLINE=1
SEQNUM=4506
USEC_INITIALIZED=16806230785

UDEV  [16806.243383] change   /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:19/PNP0C09:01/PNP0C0A:03/power_supply/BAT0 (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:19/PNP0C09:01/PNP0C0A:03/power_supply/BAT0
SUBSYSTEM=power_supply
POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_TYPE=Battery
POWER_SUPPLY_STATUS=Charging
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_TECHNOLOGY=Li-ion
POWER_SUPPLY_CYCLE_COUNT=283
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=11400000
POWER_SUPPLY_VOLTAGE_NOW=11400000
POWER_SUPPLY_POWER_NOW=5460000
POWER_SUPPLY_ENERGY_FULL_DESIGN=48336000
POWER_SUPPLY_ENERGY_FULL=40401000
POWER_SUPPLY_ENERGY_NOW=32296000
POWER_SUPPLY_CAPACITY=79
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_MODEL_NAME=ASUS Battery
POWER_SUPPLY_MANUFACTURER=ASUSTeK
POWER_SUPPLY_SERIAL_NUMBER= 
SEQNUM=4507
USEC_INITIALIZED=16806243145

UDEV  [16806.245474] change   /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:19/PNP0C09:01/ACPI0003:00/power_supply/AC0 (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:19/PNP0C09:01/ACPI0003:00/power_supply/AC0
SUBSYSTEM=power_supply
POWER_SUPPLY_NAME=AC0
POWER_SUPPLY_TYPE=Mains
POWER_SUPPLY_ONLINE=1
SEQNUM=4508
USEC_INITIALIZED=16806245369

UDEV  [16806.354038] change   /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:19/PNP0C09:01/PNP0C0A:03/power_supply/BAT0 (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:19/PNP0C09:01/PNP0C0A:03/power_supply/BAT0
SUBSYSTEM=power_supply
POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_TYPE=Battery
POWER_SUPPLY_STATUS=Charging
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_TECHNOLOGY=Li-ion
POWER_SUPPLY_CYCLE_COUNT=283
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=11400000
POWER_SUPPLY_VOLTAGE_NOW=11400000
POWER_SUPPLY_POWER_NOW=5460000
POWER_SUPPLY_ENERGY_FULL_DESIGN=48336000
POWER_SUPPLY_ENERGY_FULL=40401000
[... 61 lines cut ...]
Charge                                                      =   79.7 [%]
Capacity                                                    =   83.6 [%]

bambus:/home/christian # tlp-stat -b
--- TLP 1.3.1 --------------------------------------------

+++ Battery Features: Charge Thresholds and Recalibrate
natacpi    = inactive (laptop not supported)
tpacpi-bat = inactive (laptop not supported)
tp-smapi   = inactive (laptop not supported)

+++ Battery Status: BAT0
/sys/class/power_supply/BAT0/manufacturer                   = ASUSTeK
/sys/class/power_supply/BAT0/model_name                     = ASUS Battery
/sys/class/power_supply/BAT0/cycle_count                    =    283
/sys/class/power_supply/BAT0/energy_full_design             =  48336 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  40401 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  32182 [mWh]
/sys/class/power_supply/BAT0/power_now                      =   5027 [mW]
/sys/class/power_supply/BAT0/status                         = Charging

Charge                                                      =   79.7 [%]
Capacity                                                    =   83.6 [%]

```

linrunner, 2020-10-05:
Please enable trace mode by adding 

     TLP_DEBUG="ps run"

to your configuration. Then Unplug/Plug AC once again and show the trace output with:

     tlp-stat -T

Bambus89, 2020-10-05:
first unplugged, second plugged in...

--- TLP 1.3.1 --------------------------------------------

-- Logs begin at Sun 2020-10-04 14:34:05 CEST, end at Mon 2020-10-05 15:09:29 CEST. --
Oct 05 14:58:00 localhost tlp[1926]: Applying power save settings...done.
Oct 05 14:58:00 localhost tlp[1926]: Setting battery charge thresholds...done.
Oct 05 15:06:02 bambus.fritz.box tlp[5031]: get_sys_power_supply(AC0).ac_offline_remembered
Oct 05 15:06:02 bambus.fritz.box tlp[5031]: get_sys_power_supply(BAT0).bat_discharging: syspwr=1
Oct 05 15:06:02 bambus.fritz.box tlp[5031]: +++ start (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Oct 05 15:06:02 bambus.fritz.box tlp[5031]: power_source=bat
Oct 05 15:06:02 bambus.fritz.box tlp[5031]: manual_mode=none
Oct 05 15:06:02 bambus.fritz.box tlp[5031]: power_mode=bat
Oct 05 15:06:02 bambus.fritz.box tlp[5031]: compare_and_save_power_state(1).equal
Oct 05 15:06:02 bambus.fritz.box tlp[5031]: save_runconf.ok: /run/tlp/tlp-run.conf_tmpgKx3us -> /run/tlp/run.conf
bambus:/home/christian # tlp-stat -T
--- TLP 1.3.1 --------------------------------------------

-- Logs begin at Sun 2020-10-04 14:34:05 CEST, end at Mon 2020-10-05 15:10:17 CEST. --
Oct 05 14:58:00 localhost tlp[1926]: Applying power save settings...done.
Oct 05 14:58:00 localhost tlp[1926]: Setting battery charge thresholds...done.
Oct 05 15:06:02 bambus.fritz.box tlp[5031]: get_sys_power_supply(AC0).ac_offline_remembered
Oct 05 15:06:02 bambus.fritz.box tlp[5031]: get_sys_power_supply(BAT0).bat_discharging: syspwr=1
Oct 05 15:06:02 bambus.fritz.box tlp[5031]: +++ start (1.3.1) ++++++++++++++++++++++++++++++++++++++++
Oct 05 15:06:02 bambus.fritz.box tlp[5031]: power_source=bat
Oct 05 15:06:02 bambus.fritz.box tlp[5031]: manual_mode=none
Oct 05 15:06:02 bambus.fritz.box tlp[5031]: power_mode=bat
Oct 05 15:06:02 bambus.fritz.box tlp[5031]: compare_and_save_power_state(1).equal
Oct 05 15:06:02 bambus.fritz.box tlp[5031]: save_runconf.ok: /run/tlp/tlp-run.conf_tmpgKx3us -> /run/tlp/run.conf

linrunner, 2020-10-05:
That is not the expected output for udev power supply events. `+++` start" means manual invocation via `tlp start`. Power supply events would be `+++ auto`. 

Something seems to be wrong with your system. Please check if TLP's udev rules file is there:

    locate tlp.rules

Bambus89, 2020-10-05:
locate tlp.rules
/usr/libexec/udev/rules.d/85-tlp.rules

linrunner, 2020-10-05:
Stop it, i didn't mean invoking `tlp auto` or anything else manually. udevd should does it automatically when the power supply events occur. We have to check the logs.

Enable debug logs for udevd:

    udevadm control --log-priority=debug

Unplug/Plug AC and show log output:

    journalctl -u systemd-udevd | grep tlp.rules

Bambus89, 2020-10-05:
have done it. No Output

linrunner, 2020-10-05:
Another udevd rule check:

    udevadm test -a change /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:19/PNP0C09:01/ACPI0003:00/power_supply/AC0

    udevadm test -a change /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:19/PNP0C09:01/PNP0C0A:03/power_supply/BAT0

Bambus89, 2020-10-05:
udevadm test -a change /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:19/PNP0C09:01/ACPI0003:00/power_supply/AC0
This program is for debugging only, it does not run any program
specified by a RUN key. It may show incorrect results, because
some values may be different, or not available at a simulation run.

Load module index
Parsed configuration file /usr/lib/systemd/network/99-default.link
Created link configuration context.
Reading rules file: /usr/lib/udev/rules.d/00-dont-del-part-nodes.rules
Reading rules file: /usr/lib/udev/rules.d/01-md-raid-creating.rules
Reading rules file: /usr/lib/udev/rules.d/10-dm.rules
Reading rules file: /usr/lib/udev/rules.d/11-dm-lvm.rules
Reading rules file: /usr/lib/udev/rules.d/11-dm-mpath.rules
Reading rules file: /usr/lib/udev/rules.d/11-dm-parts.rules
Reading rules file: /usr/lib/udev/rules.d/13-dm-disk.rules
Reading rules file: /usr/lib/udev/rules.d/39-usbmuxd.rules
Reading rules file: /usr/lib/udev/rules.d/40-libgphoto2.rules
Reading rules file: /usr/lib/udev/rules.d/40-usb-blacklist.rules
Reading rules file: /usr/lib/udev/rules.d/40-usb_modeswitch.rules
Reading rules file: /usr/lib/udev/rules.d/42-hd-audio-pm.rules
Reading rules file: /usr/lib/udev/rules.d/42-usb-hid-pm.rules
Reading rules file: /usr/lib/udev/rules.d/50-iscsi-firmware-login.rules
Reading rules file: /usr/lib/udev/rules.d/50-udev-default.rules
Reading rules file: /usr/lib/udev/rules.d/54-before-scsi-sg3_id.rules
Reading rules file: /usr/lib/udev/rules.d/55-Argyll.rules
Reading rules file: /usr/lib/udev/rules.d/55-libsane.rules
Reading rules file: /usr/lib/udev/rules.d/55-scsi-sg3_id.rules
Reading rules file: /usr/lib/udev/rules.d/56-multipath.rules
Reading rules file: /usr/lib/udev/rules.d/56-sane-backends-autoconfig.rules
Reading rules file: /usr/lib/udev/rules.d/58-scsi-sg3_symlink.rules
Reading rules file: /usr/lib/udev/rules.d/59-fc-wwpn-id.rules
Reading rules file: /usr/lib/udev/rules.d/60-autosuspend.rules
Reading rules file: /usr/lib/udev/rules.d/60-block.rules
Reading rules file: /usr/lib/udev/rules.d/60-cdrom_id.rules
Reading rules file: /usr/lib/udev/rules.d/60-drm.rules
Reading rules file: /usr/lib/udev/rules.d/60-evdev.rules
Reading rules file: /usr/lib/udev/rules.d/60-fido-id.rules
Reading rules file: /usr/lib/udev/rules.d/60-input-id.rules
Reading rules file: /usr/lib/udev/rules.d/60-io-scheduler.rules
Reading rules file: /usr/lib/udev/rules.d/60-pcmcia.rules
Reading rules file: /usr/lib/udev/rules.d/60-persistent-alsa.rules
Reading rules file: /usr/lib/udev/rules.d/60-persistent-input.rules
Reading rules file: /usr/lib/udev/rules.d/60-persistent-storage-tape.rules
Reading rules file: /usr/lib/udev/rules.d/60-persistent-storage.rules
Reading rules file: /usr/lib/udev/rules.d/60-persistent-v4l.rules
Reading rules file: /usr/lib/udev/rules.d/60-scdaemon.rules
Reading rules file: /usr/lib/udev/rules.d/60-sensor.rules
Reading rules file: /usr/lib/udev/rules.d/60-serial.rules
Reading rules file: /usr/lib/udev/rules.d/60-vboxdrv.rules
Reading rules file: /usr/lib/udev/rules.d/61-gdm.rules
Reading rules file: /usr/lib/udev/rules.d/61-gnome-settings-daemon-rfkill.rules
Reading rules file: /usr/lib/udev/rules.d/61-persistent-storage-compat.rules
Reading rules file: /usr/lib/udev/rules.d/63-md-raid-arrays.rules
Reading rules file: /usr/lib/udev/rules.d/64-btrfs-dm.rules
Reading rules file: /usr/lib/udev/rules.d/64-btrfs.rules
Reading rules file: /usr/lib/udev/rules.d/64-md-raid-assembly.rules
Reading rules file: /usr/lib/udev/rules.d/65-libwacom.rules
Reading rules file: /usr/lib/udev/rules.d/65-wacom.rules
Reading rules file: /usr/lib/udev/rules.d/66-kpartx.rules
Reading rules file: /usr/lib/udev/rules.d/68-del-part-nodes.rules
Reading rules file: /usr/lib/udev/rules.d/69-cd-sensors.rules
Reading rules file: /usr/lib/udev/rules.d/69-dm-lvm-metad.rules
Reading rules file: /usr/lib/udev/rules.d/69-libmtp.rules
Reading rules file: /usr/lib/udev/rules.d/69-md-clustered-confirm-device.rules
Reading rules file: /usr/lib/udev/rules.d/69-xorg-vmmouse.rules
Reading rules file: /usr/lib/udev/rules.d/70-fido2.rules
Reading rules file: /usr/lib/udev/rules.d/70-joystick.rules
Reading rules file: /usr/lib/udev/rules.d/70-mouse.rules
Reading rules file: /usr/lib/udev/rules.d/70-power-switch.rules
Reading rules file: /usr/lib/udev/rules.d/70-printers.rules
[... 192 lines cut ...]
Reading rules file: /usr/lib/udev/rules.d/99-systemd.rules
Reading rules file: /usr/lib/udev/rules.d/99-wakeup-from-idle.rules
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0A08:00/device:19/PNP0C09:01/PNP0C0A:03/power_supply/BAT0
POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_TYPE=Battery
POWER_SUPPLY_STATUS=Discharging
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_TECHNOLOGY=Li-ion
POWER_SUPPLY_CYCLE_COUNT=284
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=11400000
POWER_SUPPLY_VOLTAGE_NOW=11400000
POWER_SUPPLY_POWER_NOW=8151000
POWER_SUPPLY_ENERGY_FULL_DESIGN=48336000
POWER_SUPPLY_ENERGY_FULL=40572000
POWER_SUPPLY_ENERGY_NOW=16245000
POWER_SUPPLY_CAPACITY=40
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_MODEL_NAME=ASUS Battery
POWER_SUPPLY_MANUFACTURER=ASUSTeK
POWER_SUPPLY_SERIAL_NUMBER= 
ACTION=change
SUBSYSTEM=power_supply
USEC_INITIALIZED=3299993679
Unload module index
Unloaded link configuration context.

linrunner, 2020-10-05:
Thanks. 

My conclusion: the package you installed (from the openSUSE repos i assume) is _severely broken_:

* The package installs the rules file at **/usr/libexec/udev/rules.d/85-tlp.rules**
* udev expects rules in **/usr/lib/udev/rules.d/** (see your last outputs)
* This way the rule never fires

I suggest you file a bug report against the openSUSE package (refer to this issue).

Bambus89, 2020-10-05:
Thank you for figuring this out ;) I file a bug on openSuSe

linrunner, 2020-10-05:
You're welcome. Please report back here when the package is fixed.

Bambus89, 2020-10-06:
Status Update:
I copied the udev rules to the correct path to fix the issue myself as a workaround.

I have reported this bug on opensuse bugzilla https://bugzilla.opensuse.org/show_bug.cgi?id=1177313.

The Bug and workaround was confirmed by Bugzilla.

Hoping they will fix this soon with a new release of the package.

Thank you again for the fast support ;)

Bambus89, 2020-10-14:
The bug is now solved by openSuSE and a new version is up in the tumbleweed repos.

linrunner, 2020-10-15:
Great. Thanks for the news.